springfall2008/batpred #260: Generalising the Inverter class to allow easy integration of other inverters

fboundy, 2023-11-05:
Having a separate class for the inverter seems a very good idea since it separates the core optimiser from the i/o however the current  Inverter class is quite specific to GivEnergy inverters. As I have worked through trying to add support for Solis inverters I have become concerned that the way I have modified the code to support this brand is becoming quite opaque and adding other brands would make it even more so.

To facilitate the integration of other brands I propose generalising and to some extent refactoring this class to make it easier to add new brands. This would also make it very clear exactly input parameters, input entities and output controls are required for an inverter brand.

I will get a PR together with a proposed revision. Initially I think this may just involve adding some more documentation to the existing class so that the parameters required are very explicit. It will also help me in debugging the Solis support mods I have made which aren’t quite there yet.

springfall2008, 2023-11-05:
Sounds like a good idea to me, I just lack the time to change too much so please propose things :)

fboundy, 2023-11-05:
First PR will just be to add some documentation to make sure my understanding of the current class is correct. Be good if you could make sure I haven’t missed anything

fboundy, 2023-11-09:
Closing this as it is addressed by #292 and similar

closed 2023-11-09 by fboundy