springfall2008/batpred #264: Repeated discharge slot alerts

gcoan, 2023-11-05:
I've turned off most of the notifications from predbat now, only set discharge notify is on.

But some of the alerts don't make sense as to why I'm getting them generated. 

From 18:20 to 18:55 I got an alert every 5 minutes (x2 inverters) telling me that the discharge time slot was being set.
I got repeated alerts that the slot was being set to 1850-1900 before from 1850 when I was then told that the slot is being set to 1845-1900

Why so many alerts that are all the same slot setting?
Personally I'd like the notifications simplified, I think I only really want to know when the batteries start and end force charging and force discharging. The slot changes are not really very useful /interesting 

[image]
[image]
[image]
[image]

gcoan, 2023-11-05:
Here's the logfile for this period
[appdaemon (4).log](https://github.com/springfall2008/batpred/files/13260577/appdaemon.4.log)

springfall2008, 2023-11-05:
I'm hoping this is fixed by this release: https://github.com/springfall2008/batpred/releases/tag/v7.11.7

gcoan, 2023-11-05:
That was quick!

I'll install it and let you know.  Thanks

gcoan, 2023-11-05:
The release note and the change you made to the sample dashboard are for the switch called set_state_notify but the actual new switch is set_status_notify 

Which is correct?

Cheers

springfall2008, 2023-11-05:
Ah crap it’s status, will fix the readme soon

gcoan, 2023-11-05:
Thanks for the  confirmation, I'll change my dashboard

gcoan, 2023-11-05:
I just noticed a series of rest errors in the predbat status so maybe that's why I kept on getting messages about the discharge slot being set every 5 minutes?
[image]

springfall2008, 2023-11-06:
I'm pretty sure this was fixed in the latest release, but your message still shows the old issue which is likely before the update?

gcoan, 2023-11-06:
I was on the prior release 11.6 when I noticed it and logged this issue.

Soon afterwards I saw the new release which I’ve installed.  Just going to see how it runs tonight at peak discharge time to confirm it’s fixed

Thanks

gcoan, 2023-11-06:
This is on 7.11.7.  I'll install 7.11.8 now 

Not got repeated discharge time slot alerts but am seeing multiple discharge rate alerts. Are these duplication detected as well?

[image]

springfall2008, 2023-12-29:
Believed to be fixed

closed 2023-12-29 by springfall2008