springfall2008/batpred #267: Charging plan not being created successfully

mhbriggs, 2023-11-05:
I initially reported this issue on the Facebook GivTCP group. I subsequently noticed that if I turn the Read Only switch to ON the charge plan looks correct. However, changing Read Only to OFF seems to cause this error where it appears that the battery would run out of charge. 

Extract from AppDaemon log below:

2023-11-05 20:39:10.887634 INFO pred_bat: Inverter 0 scheduled discharge enable is False
2023-11-05 20:39:10.892500 INFO pred_bat: Inverter 0 discharge windows currently [{'start': 0, 'end': 0, 'average': 0}, {'start': 1440, 'end': 1440, 'average': 0}]
2023-11-05 20:39:10.896018 INFO pred_bat: Found 1 inverters totals: min reserve 0.66 current reserve 0.66 soc_max 16.38 soc 10.32 charge rate 3.0 kw discharge rate 3.0 kw battery_rate_min 0.0 w ac limit 7.5 export limit 7.5 kw loss charge 3 % loss discharge 3 % inverter loss 4 %
2023-11-05 20:39:10.899974 INFO pred_bat: Base charge    window [ 05-11 23:32:00 - 06-11 05:28:00 @ 0.0p 14.74%, 06-11 23:32:00 - 07-11 05:28:00 @ 0.0p 14.74% ]
2023-11-05 20:39:10.903783 INFO pred_bat: Base discharge window [ 05-11 00:00:00 - 05-11 00:00:00 @ 0.0p 100.0%, 06-11 00:00:00 - 06-11 00:00:00 @ 0.0p 100.0% ]
2023-11-05 20:39:10.908047 INFO pred_bat: Will recompute the plan as it is invalid
2023-11-05 20:39:10.911954 INFO pred_bat: Best charge    window [ 05-11 23:30:00 - 06-11 05:30:00 @ 7.5p 14.74%, 06-11 23:30:00 - 07-11 05:30:00 @ 7.5p 14.74% ]
2023-11-05 20:39:10.915416 INFO pred_bat: Best discharge window [  ]
2023-11-05 20:39:10.969013 INFO pred_bat: predict base end_record 11-06 23:30:00 final soc 10.28 kwh metric 546.08 p metric_keep 0.0 min_soc 10.28 @ 11-06 23:25:00 kwh load 63.54 pv 14.67
2023-11-05 20:39:10.974393 INFO pred_bat:          [  20:35,   21:35,   22:35,   23:35,   00:35,   01:35,   02:35,   03:35,   04:35,   05:35,   06:35,   07:35,   08:35,   09:35,   10:35,   11:35,   12:35,   13:35,   14:35,   15:35,   16:35,   17:35,   18:35,   19:35,   20:35,   21:35,   22:35]
2023-11-05 20:39:10.979880 INFO pred_bat:     SOC: [  10.32,    9.78,    9.53,    8.74,   11.54,   14.33,   14.74,   14.74,   14.74,   14.72,    14.4,   13.95,   13.99,    14.5,   15.06,   16.03,   16.38,   16.38,   16.22,   15.77,   15.42,   14.93,   12.68,   11.84,   11.33,   10.91,   10.58]
2023-11-05 20:39:10.984889 INFO pred_bat:   STATE: [  g~be-,   g~be-,   g~be-,   g<bf+,   g<bf+,   g<bf+,   g<bf+,   g<be-,   g<be-,   g~be-,   g~be-,   g~be-,   g~be+,   g~be+,   g~be+,   g~be+,   g>be+,   g>be+,   g~be-,   g~be+,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-]
2023-11-05 20:39:10.990072 INFO pred_bat:    LOAD: [  14.66,   15.16,    15.4,   16.53,   24.23,   32.03,    34.1,    38.8,   46.62,   53.76,   54.06,   54.58,   54.98,   55.31,    55.9,   56.22,   56.58,   56.98,   57.62,   58.22,   58.55,   59.01,    61.3,   62.09,   62.56,   62.95,   63.26]
2023-11-05 20:39:10.995324 INFO pred_bat:      PV: [    7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,    7.59,    8.07,    8.93,   10.13,   11.47,   12.77,   13.81,   14.47,   14.66,   14.67,   14.67,   14.67,   14.67,   14.67,   14.67,   14.67]
2023-11-05 20:39:11.000486 INFO pred_bat:  IMPORT: [  11.47,   11.47,   11.47,   11.87,   22.57,   33.37,   35.88,   40.58,   48.39,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.72,   55.72,   55.72,   55.72,   55.72]
2023-11-05 20:39:11.005674 INFO pred_bat:  EXPORT: [    0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,    1.14,    1.75,    1.87,    1.87,    1.87,    1.87,    1.87,    1.87,    1.87,    1.87,    1.87]
2023-11-05 20:39:11.011008 INFO pred_bat:    CAR0: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    5.09,   14.16,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48]
2023-11-05 20:39:11.016274 INFO pred_bat:  METRIC: [ 156.24,  156.24,  156.24,  160.36,  292.93,  373.93,  392.78,  428.01,  486.63,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,  546.08,  546.08,  546.08,  546.08,  546.08]
2023-11-05 20:39:11.569671 INFO pred_bat: Battery has 48.0 hours left - now at 10.32
2023-11-05 20:39:11.662396 INFO pred_bat: predict base10 end_record 11-06 23:30:00 final soc 6.35 kwh metric 546.08 p metric_keep 0.0 min_soc 6.35 @ 11-06 23:25:00 kwh load 63.54 pv 9.39
2023-11-05 20:39:11.667962 INFO pred_bat:          [  20:35,   21:35,   22:35,   23:35,   00:35,   01:35,   02:35,   03:35,   04:35,   05:35,   06:35,   07:35,   08:35,   09:35,   10:35,   11:35,   12:35,   13:35,   14:35,   15:35,   16:35,   17:35,   18:35,   19:35,   20:35,   21:35,   22:35]
2023-11-05 20:39:11.673364 INFO pred_bat:     SOC: [  10.32,    9.78,    9.53,    8.74,   11.54,   14.33,   14.74,   14.74,   14.74,   14.72,    14.4,   13.87,   13.52,   13.38,   13.08,    13.1,   13.08,   12.95,   12.44,   11.84,   11.49,    11.0,    8.75,     7.9,     7.4,    6.98,    6.65]
2023-11-05 20:39:11.678987 INFO pred_bat:   STATE: [  g~be-,   g~be-,   g~be-,   g<bf+,   g<bf+,   g<bf+,   g<bf+,   g<be-,   g<be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be+,   g~be+,   g~be-,   g~be+,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-]
2023-11-05 20:39:11.684462 INFO pred_bat:    LOAD: [  14.66,   15.16,    15.4,   16.53,   24.23,   32.03,    34.1,    38.8,   46.62,   53.76,   54.06,   54.58,   54.98,   55.31,    55.9,   56.22,   56.58,   56.98,   57.62,   58.22,   58.55,   59.01,    61.3,   62.09,   62.56,   62.95,   63.26]
2023-11-05 20:39:11.689637 INFO pred_bat:      PV: [    7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,    7.52,     7.6,    7.82,    8.14,     8.5,    8.86,    9.16,    9.34,    9.38,    9.39,    9.39,    9.39,    9.39,    9.39,    9.39,    9.39]
2023-11-05 20:39:11.694862 INFO pred_bat:  IMPORT: [  11.47,   11.47,   11.47,   11.87,   22.57,   33.37,   35.88,   40.58,   48.39,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.52,   55.72,   55.72,   55.72,   55.72,   55.72]
2023-11-05 20:39:11.700124 INFO pred_bat:  EXPORT: [    0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6]
2023-11-05 20:39:11.705385 INFO pred_bat:    CAR0: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    5.09,   14.16,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48]
2023-11-05 20:39:11.710899 INFO pred_bat:  METRIC: [ 156.24,  156.24,  156.24,  160.36,  292.93,  373.93,  392.78,  428.01,  486.63,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,   540.1,  546.08,  546.08,  546.08,  546.08,  546.08]
2023-11-05 20:39:11.909710 INFO pred_bat: Calculate Best options: calculate_best_charge(True) calculate_best_discharge(True) calculate_discharge_first(True) calculate_discharge_oncharge(True) set_charge_freeze(True) set_discharge_freeze(True) set_discharge_freeze_only(False) set_discharge_during_charge(False) combine_charge_slots(True) combine_discharge_slots(False) best_soc_min(0.0 kWh) best_soc_max(0.0 kWh) best_soc_keep(4.0 kWh) inverter_loss(4 %) battery_loss(3 %) battery_loss_discharge (3 %) inverter_hybrid(True) metric_min_improvement(0.0 p) metric_min_improvement_discharge(0.1 p) metric_battery_cycle(0.0 p/kWh)
2023-11-05 20:39:11.913574 INFO pred_bat: Optimise all windows, total charge 1 discharge 1
2023-11-05 20:39:11.917259 INFO pred_bat: All prices [8, 7]
2023-11-05 20:39:11.959266 INFO pred_bat: Optimise all for buy/sell price band <= 8 metric 479.53 soc_min 10.28 windows [0] discharge off
2023-11-05 20:39:11.963776 INFO pred_bat: Optimise all charge found best buy/sell price band 8 best price threshold 8 at metric 479.531287766012 cost 558.98 limits [16.38, 16.38] discharge []
2023-11-05 20:39:12.005699 INFO pred_bat: Optimise all for buy/sell price band <= 7 metric 454.54 soc_min 5.83 windows [] discharge off
[... cut ...]
2023-11-05 20:40:03.577196 INFO pred_bat: Inverter 0 using Rest API http://homeassistant.local:6345
2023-11-05 20:40:04.859230 INFO pred_bat: Invertor time 2023-11-05 20:39:33+00:00 AppDeamon time 2023-11-05 20:40:00.012773+00:00 difference -0.47 minutes
2023-11-05 20:40:04.864665 INFO pred_bat: New Inverter 0 with soc_max 16.38 kWh nominal_capacity 16.38 kWh battery rate raw 3000.0 w charge rate 3.0 kw discharge rate 3.0 kw battery_rate_min 0.0 w ac limit 7.5 kw export limit 7.5 kw reserve 4.0 % current_reserve 4.0 %

springfall2008, 2023-11-05:
Do you have the log with read only off, this appears to be with it on?

mhbriggs, 2023-11-05:
I switched Read Only OFF again and the plan actually looks ok now! Here is the log:

2023-11-05 21:10:07.034820 INFO pred_bat:      PV: [    7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,    7.56,    7.71,     8.0,    8.35,    8.72,    9.05,    9.28,    9.37,    9.39,    9.39,    9.39,    9.39,    9.39,    9.39,    9.39,    9.39]
2023-11-05 21:10:07.039840 INFO pred_bat:  IMPORT: [   11.7,    11.7,    11.7,   17.42,   25.32,   30.82,   32.75,   40.54,   48.37,   50.94,   50.94,   50.94,   50.94,   50.94,   50.94,   50.94,   50.94,   50.94,   50.94,   50.94,   50.94,   51.02,   51.17,   51.17,   51.17,   51.17,   51.17]
2023-11-05 21:10:07.045305 INFO pred_bat:  EXPORT: [    0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6]
2023-11-05 21:10:07.050333 INFO pred_bat:    CAR0: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    1.44,   10.38,   19.46,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48]
2023-11-05 21:10:07.055387 INFO pred_bat:  METRIC: [ 157.99,  157.99,  157.99,  253.23,  312.49,  353.74,   368.2,   426.6,  485.32,   504.6,   504.6,   504.6,   504.6,   504.6,   504.6,   504.6,   504.6,   504.6,   504.6,   504.6,   504.6,  507.09,  511.57,  511.57,  511.57,  511.57,  511.57]
2023-11-05 21:10:07.292233 INFO pred_bat: predict best10 end_record 11-06 23:35:00 final soc 1.59 kwh metric 510.45 p metric_keep 74.97 min_soc 1.35 @ 11-06 23:25:00 kwh load 63.76 pv 9.39
2023-11-05 21:10:07.299388 INFO pred_bat:          [  21:10,   22:10,   23:10,   00:10,   01:10,   02:10,   03:10,   04:10,   05:10,   06:10,   07:10,   08:10,   09:10,   10:10,   11:10,   12:10,   13:10,   14:10,   15:10,   16:10,   17:10,   18:10,   19:10,   20:10,   21:10,   22:10,   23:10]
2023-11-05 21:10:07.306423 INFO pred_bat:     SOC: [   9.83,    9.41,    9.18,    9.63,    9.63,    9.63,    9.63,    9.63,    9.63,    9.47,    9.04,    8.65,    8.27,    8.18,    7.94,    8.05,    7.92,    7.55,     7.1,     6.7,     6.2,    5.06,    3.21,    2.61,    2.15,    1.78,    1.47]
2023-11-05 21:10:07.312832 INFO pred_bat:   STATE: [  g~be-,   g~be-,   g~be-,   g<bf+,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g~be-,   g~be-,   g~be-,   g~be+,   g~be-,   g~be+,   g~be-,   g~be-,   g~be+,   g~be+,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-]
2023-11-05 21:10:07.318640 INFO pred_bat:    LOAD: [   15.1,    15.5,   15.71,    20.5,    28.4,    33.9,   35.83,   43.62,   51.45,   54.17,   54.57,   54.99,   55.49,   55.84,   56.39,   56.64,   57.07,   57.63,   58.13,   58.52,   58.98,   60.13,   62.01,   62.56,   62.99,   63.34,   63.62]
2023-11-05 21:10:07.326572 INFO pred_bat:      PV: [    7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,     7.5,    7.56,    7.71,     8.0,    8.35,    8.72,    9.05,    9.28,    9.37,    9.39,    9.39,    9.39,    9.39,    9.39,    9.39,    9.39,    9.39]
2023-11-05 21:10:07.332605 INFO pred_bat:  IMPORT: [   11.7,    11.7,    11.7,    17.0,    24.9,    30.4,   32.33,   40.11,   47.94,   50.51,   50.51,   50.51,   50.51,   50.51,   50.51,   50.51,   50.51,   50.51,   50.51,   50.51,   50.51,   50.59,   50.74,   50.74,   50.74,   50.74,   50.74]
2023-11-05 21:10:07.339220 INFO pred_bat:  EXPORT: [    0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6,     0.6]
2023-11-05 21:10:07.345568 INFO pred_bat:    CAR0: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    1.44,   10.38,   19.46,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48,   22.48]
2023-11-05 21:10:07.351653 INFO pred_bat:  METRIC: [ 157.99,  157.99,  157.99,  250.04,  309.28,  350.53,   365.0,   423.4,  482.12,  501.39,  501.39,  501.39,  501.39,  501.39,  501.39,  501.39,  501.39,  501.39,  501.39,  501.39,  501.39,  503.89,  508.37,  508.37,  508.37,  508.37,  508.37]
2023-11-05 21:10:07.596359 INFO pred_bat: predict best end_record 11-06 23:35:00 final soc 6.81 kwh metric 510.45 p metric_keep 0.0 min_soc 6.58 @ 11-06 23:25:00 kwh load 63.76 pv 14.67
2023-11-05 21:10:07.603809 INFO pred_bat:          [  21:10,   22:10,   23:10,   00:10,   01:10,   02:10,   03:10,   04:10,   05:10,   06:10,   07:10,   08:10,   09:10,   10:10,   11:10,   12:10,   13:10,   14:10,   15:10,   16:10,   17:10,   18:10,   19:10,   20:10,   21:10,   22:10,   23:10]
2023-11-05 21:10:07.609979 INFO pred_bat:     SOC: [   9.83,    9.41,    9.18,    9.63,    9.63,    9.63,    9.63,    9.63,    9.63,    9.47,    9.06,    8.93,     9.1,    9.78,   10.48,   11.52,   12.23,   12.45,    12.3,   11.93,   11.43,   10.28,    8.43,    7.83,    7.38,     7.0,     6.7]
2023-11-05 21:10:07.616210 INFO pred_bat:   STATE: [  g~be-,   g~be-,   g~be-,   g<bf+,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g~be-,   g~be-,   g~be+,   g~be+,   g~be-,   g~be+,   g~be+,   g~be+,   g~be+,   g~be+,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-]
[... cut ...]
2023-11-05 21:10:13.366549 INFO pred_bat: Not setting charging SOC as we are not within the window (now 11-05 21:10:00 target set_soc_minutes 30 charge start time 11-05 23:30:00)
2023-11-05 21:10:13.370278 INFO pred_bat: Adjust reserve to default (as set_reserve_enable is true)
2023-11-05 21:10:13.374265 INFO pred_bat: Inverter 0 Current reserve is 4.0 already at target
2023-11-05 21:10:13.415171 INFO pred_bat: Completed run status Idle
2023-11-05 21:10:13.440593 WARNING AppDaemon: Excessive time spent in callback 'run_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 13.439963 seconds (limit=10)

springfall2008, 2023-11-06:
I'm suspecting the chart just didn't update correctly given everything else seemed okay.

closed 2023-11-13 by springfall2008