springfall2008/batpred #268: Midnight issue

slopemad, 2023-11-06:
I see an issue every day when Predbat does it's midnight run, in that the entire plan for the immediate future is removed.

[image]

The 00:00 run log is here: https://pastebin.com/E4ETAvBZ

And the plan returns in the 00:05 run.  I see this consistently every day.

[image]

The 00:05 run log is here: https://pastebin.com/Kfhvxkqr

Could it be related to this in apps.yaml?

  days_previous: 
    - 1
    - 7
    - 14

  days_previous_weight:
    - 1
    - 0.8
    - 0.6

springfall2008, 2023-11-06:
Good spot, I pushed a fix and it will be in the next release

gcoan, 2023-11-07:
I think I saw the same issue myself last night, and I was running 7.11.8 which the midnight fix #270 was included in.

Just after midnight I looked at predbat to see when the cheap period was to put the washing machine on at was, and was astonished to see that there was no charging or discharging planned for the day and the battery would empty out.
I didn't get much chance to look at it then, or take a screen shot, but when I next looked at 00:05 there was a normal pattern of charging and discharging planned for the day.
So not entirely sure this is fixed.

Here's my logfile for around midnight if you can see what the issue might be?

I will take another look tonight if I can see if happens again

[appdaemon.log.5.txt](https://github.com/springfall2008/batpred/files/13278616/appdaemon.log.5.txt)

springfall2008, 2023-11-07:
Not surprising since I got it the wrong way around!

https://github.com/springfall2008/batpred/releases/tag/v7.11.10

gcoan, 2023-11-08:
Looks like it's fixed now

Before midnight 
[image]

First run after midnight 
[image]

And in the following runs, it didn't change again

springfall2008, 2023-11-08:
Great, thanks for helping with this.

closed 2023-11-08 by springfall2008