springfall2008/batpred #271: Charge freeze

slopemad, 2023-11-06:
It isn't entirely clear why Predbat has decided to set the charge rate to zero at the moment.  

```
2023-11-06 10:55:08.098864 INFO pred_bat: Best charging limit socs [6.36, 8.005333333333333, 0.33, 0.33, 0.33, 3.36, 4.934333333333334, 6.145333333333333, 7.355333333333333, 8.36, 0.33, 0.33, 0.33,>
2023-11-06 10:55:08.297958 INFO pred_bat: Charge window will be: 2023-11-06 11:30:00+00:00 - 2023-11-06 15:00:00+00:00 - current soc 59.target 76
2023-11-06 10:55:08.299983 INFO pred_bat: Not setting charging window yet as not within the window (now 11-06 10:55:00 target set_window_minutes 30 charge start time 11-06 11:30:00
2023-11-06 10:55:08.301977 INFO pred_bat: Include original discharge start 11-05 21:30:00 with our start which is 11-06 10:30:00
2023-11-06 10:55:08.304026 INFO pred_bat: Next discharge window will be: 2023-11-05 21:30:00+00:00 - 2023-11-06 11:01:00+00:00 at reserve 60.0
2023-11-06 10:55:08.306244 INFO pred_bat: Inverter 0 current charge rate is 2600 and new target is 0
2023-11-06 10:55:17.104211 INFO pred_bat: Inverter 0 set charge rate 0 via REST succesfull on retry 0
2023-11-06 10:55:17.162358 INFO pred_bat: Discharge Freeze as discharge is now at/below target - current SOC 4.93 and target 5.016
2023-11-06 10:55:17.165647 INFO pred_bat: Not setting charging SOC as we are not within the window (now 11-06 10:55:00 target set_soc_minutes 30 charge start time 11-06 11:30:00)
2023-11-06 10:55:17.168758 INFO pred_bat: Adjust reserve to default (as set_reserve_enable is true)
2023-11-06 10:55:17.173487 INFO pred_bat: Inverter 0 Current reserve is 4.0 already at target
2023-11-06 10:55:17.209716 INFO pred_bat: Will not recompute the plan, it is 4.983333333333333 minutes old max age 10
2023-11-06 10:55:17.211432 INFO pred_bat: Completed run status Freeze discharging
```

[image]

Full log: https://pastebin.com/6vn1GXXB

There is slightly more solar than household demand at the moment, and Agile rates aren't below export rates, so all that allowing that solar to go out for Export rather than trickling into the battery, means is that we have to import more at a rate higher than export during the upcoming charge windows.

springfall2008, 2023-11-06:
Log wise, do you have the previous iteration when the plan was actually made as this log is just the execution of the previous calculated plan?

kingjpc, 2023-11-06:
Seeing similar behaviour, out at the moment so can’t grab my logs. 
[image]

It is trying to charge for some reason when it shouldn’t need to. The predicted plan is worse than doing nothing :)

springfall2008, 2023-11-06:
Do you maybe have metric keep set to non zero so it’s trying to avoid going below it?

slopemad, 2023-11-06:
The log from the 10:50 run : https://pastebin.com/cANUUZNy

For me, Best SOC Keep is set to 1

Since upgrading to 7.11.7 last night, the overall charge plan is better than it was with 7.10.x before it, it's choosing better windows to charge (rather than charging at slightly more expensive windows then holding at cheaper windows), but overnight there isn't any solar inputs to contend with.

springfall2008, 2023-11-06:
Ultimately I thinks an issue because clouds aren't modelled it looks like solar will cover the house load but it won't always. Turning off freeze charge maybe the sensible thing for now, but I think I might had a simple cloud model

kingjpc, 2023-11-06:
I have my SOC Keep set to 2
I noticed that when solar didn't cover charging the battery + house load it was pulling from the grid in "Charge Freeze" is that expected?

slopemad, 2023-11-06:
What I didn't understand is why it stopped solar charging the battery during a freeze discharge window.  I can understand when import price are low why you might want to set the charge rate to zero, but import prices are generally higher than export at the moment.

Freeze charge, when it sets the target SOC for AC charging to 4%, is working brilliantly.

springfall2008, 2023-11-07:
I'm not sure I fully know why either, I can only assume it's marginal

gcoan, 2023-11-07:
> Ultimately I thinks an issue because clouds aren't modelled it looks like solar will cover the house load but it won't always. Turning off freeze charge maybe the sensible thing for now, but I think I might had a simple cloud model

Can you explain a bit more about how the new cloud model works?

I had been thinking of some kind of automation that changed input_number.predbat_pv_metric10_weight based upon the difference between PV Forecast Today and PV 10% Forecast Today.

e.g. if PV and PV 10% are "close" then weight metric_10 at (say) 10%

If PV and PV 10% are "far apart" (e.g. PV is 10kW and PV10% is 1kW) then weight at say 50% to apply higher weighting to the potentially low PV forecast.

Is this what the new #274 does?  The readme just describes the model as "modulating" but I don't really know what this implies and what it means for different PV scenarios.

Tomorrow looks to be a rubbish PV day, my forecast is 4.4kW and 10% 2.9kW, compared to 16kW and 10kW today.

wilsondw, 2023-11-08:
> What I didn't understand is why it stopped solar charging the battery during a freeze discharge window. I can understand when import price are low why you might want to set the charge rate to zero, [...]

I see the same problem and think it maybe a feature of the inverter firmware. I asked Giv about the odd behaviour and the response I got was “the correct way to pause battery discharge is to use the pause battery (discharge) function. Remember to set the time for the pause”. Guess this maybe another case of Predbat works with some firmware versions and not others and maybe another switch needed to use pause instead of discharge rate and SOC target for pause discharge?

springfall2008, 2023-11-20:
>Can you explain a bit more about how the new cloud model works?

So it's quite crude, it tries to keep the same average solar prediction but predicts a high amount for 5 minutes and then a lower amount for the next 5 minutes after that.

Sort of oOoOoO

The scaling from high to low is determined by the spread between pv 50% and pv 10% scenarios.

springfall2008, 2023-12-29:
Now fixed

closed 2023-12-29 by springfall2008