springfall2008/batpred #276: Documentation rework

iainfogg, 2023-11-06:
Hi @springfall2008 

As the project has grown in features and complexity, I think the README has now got really long, and isn't ideal for people finding their way around it.

What do you think about reworking the content of the README file into a proper documentation site? I'm suggesting using `mkdocs` along with GitHub Pages, and using a CI pipeline to build and publish the docs each time there's a new commit pushed to `master` (or I expect you could also do it on new tags instead).

I'd be happy to have a crack at doing this for you, I've used `mkdocs` before, although I think I'd probably need some permissions temporarily on your repo to do it to configure the settings and publishing. Or obviously feel free to go at it yourself :)

It'd involve taking the content of the current README and breaking it into separate files, and each file would be a separate page in the documentation site. I'd recommend using the `readthedocs` theme within `mkdocs`, so it'd look the same as this site:

https://docs.readthedocs.io/en/stable/intro/getting-started-with-mkdocs.html

I think it'd make it a lot easier for people to make use of the content that's currently in the README.

What do you reckon?

iainfogg, 2023-11-06:
Some useful links:

https://www.mkdocs.org/user-guide/deploying-your-docs/

https://squidfunk.github.io/mkdocs-material/publishing-your-site/

https://blog.elmah.io/deploying-a-mkdocs-documentation-site-with-github-actions/

iainfogg, 2023-11-06:
Sorry, our comments overlapped there :)

Great, I'll give it a crack at some point fairly soon, I'll let you know when I'm doing it.

iainfogg, 2023-11-06:
Yep, I've created a draft PR for now to check it was possible. I'll update you when I have something for you to see.

iainfogg, 2023-11-06:
There's a Settings tab that I need access to on the project when you have a chance @springfall2008 - it's where GitHub Pages is configured.

springfall2008, 2023-11-07:
I think I've turned on pages now, I don't know how to change permissions for collaborators

iainfogg, 2023-11-07:
I think that to have more fine grained control over settings you can give to people, you first need to convert your personal account into an organisation account. Then you can give out more specific permissions. I did that with my account a while ago.

See https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-your-personal-account/converting-a-user-into-an-organization for more info.

If you don't fancy doing that, I'll have a look at settings on my own fork to see what settings are needed, then I'll pass them on to you, as I think there's probably more to it than just turning it on.

iainfogg, 2023-11-07:
Here's what I'm proposing it will look like - what do you think @springfall2008 ?

https://iainfogg.github.io/batpred/

springfall2008, 2023-11-07:
I've changed the settings to match this, what next?

iainfogg, 2023-11-07:
It's now deployed automatically at https://springfall2008.github.io/batpred/ after you made the settings changes.

Have a look and let me know what you think.

I'm still working on one more page, which explains about developing on the code, and building the documentation.

I also need to amend the main README to remove the content and to point to the new docs.

And I need to amend it to publish based on a release, rather than its current temp setup if building based on new commits to the documentation branch.

Let me know if it feels like an improvement though, and I can get those other things sorted fairly soon, then we can get it merged in.

iainfogg, 2023-11-08:
Hi @springfall2008 - I've now:

- updated the README to be pretty barebones now, and to point to the main doc.
- amended the GitHub Actions file to also deploy docs on a release, but I can't test this
- renamed and restructured the menu on the left a bit, to try and make more logical groupings rather than a long list
- fixed a few formatting / link issues

I think it's ready to go if you're happy, but obviously feel free to amend anything or to say if it's not what you were hoping for.

There's one section here https://springfall2008.github.io/batpred/developing/#developing-locally-within-home-assistant which I was hoping you could add to at some point, to describe how you work on the code within HA, but I'd prefer to get this merged in first if possible. I'm trying to avoid any changes happening to the main README while there's a second version of the documentation knocking around, so getting this merged in soon would be ideal, I think, when you're happy.

iainfogg, 2023-11-08:
Just did a few more whitespace  / spelling fixes

iainfogg, 2023-11-09:
Now this is live, if you go to the repo home page and see the About box on the right, click on the cog there - under Website, tick the 'Use your GitHub Pages website' item, and it'll add a link to the documentation site in that box, making it easy for people to spot. 

You may also want to turn Packages off, as that's not being used.

springfall2008, 2023-11-09:
Thanks, Awesome work with the documentation update it's much nicer now :)

iainfogg, 2023-11-09:
Thanks Trefor, glad you like it :)

Don't forget the bit I mentioned earlier, when you get a chance:

> There's one section here https://springfall2008.github.io/batpred/developing/#developing-locally-within-home-assistant which I was hoping you could add to at some point, to describe how you work on the code within HA

Shall I create a separate issue for that, or leave this one open as a reminder?

fboundy, 2023-11-09:
Hi both - I've made some changes to the markdown docs to add some details on getting Solis inverters up and running on PredBat but I'm unclear how to run these through mkdocs. They are in the `Solis Docs` branch

iainfogg, 2023-11-10:
Hi @fboundy thanks for doing that.

The docs are rebuilt and published each time Trefor does a new release.

So once you're happy that the docs look fine, based on running it locally with mkdocs, just do a PR as normal. Then they'll be on main, and will get included and published on the next release.

iainfogg, 2023-11-10:
Or did you mean you're not sure how to check them now while you're working on them?

gcoan, 2023-11-10:
I was wondering the same

Do we edit the docs in the docs folder (which is pretty much the same as editing readme directly only more granular files) and include in the commit as normal or is there a source doc we should edit that is then generated into the docs doc me mkdocs?

fboundy, 2023-11-10:
> Or did you mean you're not sure how to check them now while you're working on them?

Yep - haven't worked out how to run mkdocs but haven't tried very hard. I'm running VSCode locally

iainfogg, 2023-11-10:
Check out https://springfall2008.github.io/batpred/developing/#working-on-the-documentation and let me know whether that covers it

fboundy, 2023-11-10:
> Check out https://springfall2008.github.io/batpred/developing/#working-on-the-documentation and let me know whether that covers it

That works fine in Codespace. If I try to run it on my local machine it can't find mkdocs.

gcoan, 2023-11-10:
Err, no 😵‍💫

I don’t have GitHub setup at all.  Maybe I should but so far I’m not trying to re-establish myself as a programmer… 
Must be 27, 28 years since I last seriously did any programming.  Been system architecture and IT technical leadership since then.

I’m doing direct edits to the python code with the HA visual studio editor, then when I’m happy I open GitHub in a browser, sync the master to my fork, then click edit in the browser and copy/paste any code changes into the GitHub window.
If it’s documentation changes I’ve been updating the doc file directly in a browser window up to now.

Did get as far as installing the GitHub app on my iPad but without a proper keyboard I wouldn’t want to develop on it.

springfall2008, 2023-11-10:
Slightly off topic but I can fully recommend a Mac mini m1 which can be picked up used for a reasonable price on eBay. Pair this with a large monitor (I have a 28 inch 4k one) and you have a great programmers setup.

gcoan, 2023-11-10:
> Slightly off topic but I can fully recommend a Mac mini m1 which can be picked up used for a reasonable price on eBay. Pair this with a large monitor (I have a 28 inch 4k one) and you have a great programmers setup. 

I'm running my home assistant as a VM on a windows 10 i5 tiny PC (off eBay) so doubtless I could turn that into a developer pc if I got forced into it.  Getting back into windows after 5 years using a Mac laptop has been hard !

springfall2008, 2023-11-10:
Mine is also on a PC but I installed HAOS on it so it runs native rather than in a VM

iainfogg, 2023-11-11:
Ok, thanks for the feedback.

- If you want mkdocs running locally without using Codespaces, it's just a case of running `pip install mkdoc`

- You don't need the GitHub app for anything to do with this - it's VS Code that you'd use if you want a dev environment locally based on Codespaces. I know you mentioned the lightweight machine you using - I have successfully run this on a cheap Chromebook, it's actually pretty lightweight and can run on Windows, Mac or Linux, so it's with giving it a go if you'd like an ide. You can even point it to a file share on your HA box so you can live edit your HA setup.

- You can just edit the Markdown files in the browser on GitHub in either the web editor or the web version of Codespaces and the documentation site will still get built, three only reason for running mkdocs while editing the pages is to be able to see a live preview of what the changes will look like - it's not doing anything more than that during editing, so it's not essential - if you're just fixing a spelling mistake, adding a new paragraph it whatever, you don't need it (although I prefer to use it)

Hopefully that helps a bit more, I'll try and update the info in the documentation site to make this stuff more clear.

Ask if there are any more questions though, and I'm happy to tweak it all as needed.

fboundy, 2023-11-11:
Yep that makes sense. I use conda rather than pip but same principle.

I'm just in the process of migrating HA and a bunch of other stuff to ProxMox running on a NUC.

iainfogg, 2023-11-11:
> I'm just in the process of migrating HA and a bunch of other stuff to ProxMox running on a NUC.

Nice, I did the same earlier this year and haven't looked back!

fboundy, 2023-11-11:
HA needs a tidy up and the Mrs is away for a week so seems like a good opportunity...

gcoan, 2023-11-29:
> Ok, thanks for the feedback.
> 
> * If you want mkdocs running locally without using Codespaces, it's just a case of running `pip install mkdoc` then it'll be installed, you can then run `mkdocs serve` as per the docs
> * You don't need the GitHub app for anything to do with this - it's VS Code that you'd use if you want a dev environment locally based on Codespaces. .....
> 
> Hopefully that helps a bit more, I'll try and update the info in the documentation site to make this stuff more clear.
> 
> Ask if there are any more questions though, and I'm happy to tweak it all as needed.

Hi @iainfogg whilst I'm sitting in a dark house with the ASHP off to save electricity :-) I thought I'd have a go at following the instructions you've added to create some pull requests, but it doesn't seem to work for me.

Following the instructions
``
From your fork, click on the Code button, and select the Codespaces tab. You can create multiple environments, or use a single environment and swap between branches in it.

Also, you can choose between running the IDE in the browser, or having your local installation of VS Code connect to the environment that GitHub Codespaces has created for you. The local installation is better in some scenarios e.g. if you need to connect to a specific port, such as if you're working on the documentation.
``

I have my own fork sync'd to the main, but when I click on Code I don't see anything about Codespaces?
[image]

I normally just edit the file directly in the web browser by clicking on the pencil icon at top right, making my changes, then committing them and when I have finished creating a pull request.  Happy to try codespaces if it'll give a superior editing experience especially for previewing changes to docs.
I also want to add a new image to what will probably be a new documentation page, so doing it the "right way" for this would be beneficial.

Reading your instructions further they talk about running `mkdocs serve` which looks like they are tailored to a unix-type command line environment?  My HA is running as a VM under windows 10 so I have just been using the windows 10 PC.
I installed Github desktop on my PC, connected it to gcoan\batpred, fetched the origin into my PC, but am stuck there as well.  I can see the files in Windows Explorer but can't work out how to see the branch contents or edit anything in Github desktop

iainfogg, 2023-11-29:
@gcoan just updated again, for a bit more clarity on local VS Code

gcoan, 2023-11-29:
Thanks @iainfogg for the additions, and in particular the screen shot that showed which Code button to press.  I needed to be pressing it at the fork/branch level, not at the individual file.   It takes stupid people that have never really used github to find the obvious questions.

Don't want the docs for batpred to become a tutorial for how to use github. I was able to follow your revised instructions, created a codespace and find the files for editing and previewing.  VS Code I was familiar enough with from using it inside HA.  Found a couple of changes to make to one of the doc files, made them, found the preview OK, but then couldn't immediately see how to get them out of codespaces and committed back into github.  Then I found the source control option and worked my way through the process, adding the doc change to an existing PR #393 I had created earlier this evening.
Thanks.  Lots to learn!

iainfogg, 2023-11-29:
Great, thanks for the feedback, glad you enjoyed contributing with it!

We'll get the PR merged in soon then with my documentation changes.

PR #277 (2023-11-09): Set up documentation site
This PR takes the existing content from the README and breaks it down into separate files in the `docs` folder, publishing them using GitHub Pages and GitHub Actions.

The aim is to create a more easily accessible, and searchable, site than the current very long README.

It uses `mkdocs` to build the HTML pages for the documentation site, based on the Markdown files.

`mkdocs.yml` defines the navigation bar at the side (this allows controlling the order of pages, which would otherwise be in alphabetic order), and also sets the theme and other settings.

Closes #276
PR #354 (2023-11-30): Documentation tidy-up
In response to some of the comments on #276 from @fboundy and @gcoan I've added some more info to the section in the documentation about working on the documentation.

Hopefully the combination of what was already there plus these comments makes the description sufficiently clear.

If not, please do add suggestions to improve it.

@springfall2008 this also stops publishing the documentation site on commits to `documentation` branch, as that was only a temp setup while building out the site. It'll now only publish on new releases.

Closes #276