springfall2008/batpred #278: Different discharge slots for the two inverters

gcoan, 2023-11-06:
This isn't affecting my discharge but I've noticed it since 7.11.7 that the start time for inverter 1 discharge is being set to 00:00 not the actual time.  Error 8n a variable name or using an uninitialised variable?

[image]
[image]

springfall2008, 2023-11-06:
I think its maybe as expected, in that the start time isn't changed if it's in the past to avoid constantly updating it.

gcoan, 2023-11-06:
I’m not sure. I’ve pretty much seen the two inverters always move in sync.

It would make sense to not change the start time if it’s in the past, except of course when it’s setting up a discharge slot for the future

I’ll set inverter 1 discharge time to the same as inverter 0 and see what happens at the next discharge period

gcoan, 2023-11-07:
There's been a couple of discharges this afternoon and in both occasions the inverters have moved the same together 

[image]
[image]

So can only come conclude that the dissimilar times before were as a result of a manual discharge I'd done 

Will close this as not being an issue

closed 2023-11-07 by gcoan