springfall2008/batpred #279: Status Warning

SheriffOfNotts, 2023-11-06:
Hi Trefor, I've been trying, without success to get rid of a status "Warn - Return bad float value unavailable from
car_charging_soc". I'm not sure exactly what this means?

This is the extract from apps.yaml:

 car_charging_soc:
    - sensor.id_3_state_of_charge
    - sensor.volvo_charge_level

Both sensors are numbers - float(0) and round (0)

adamwylde, 2023-11-06:
Innocent bystander here, but maybe you are missing the quotes? Mine looks like this…

car_charging_soc: 
    - 're:sensor.cupra_born_state_of_charge'

springfall2008, 2023-11-07:
It's saying the sensor is returning 'unavailable' rather than a value

SheriffOfNotts, 2023-11-07:
Ah ok, thanks Trefor.

springfall2008, 2023-11-13:
Did you re-open on purpose or was this a mistake?

closed 2023-11-13 by SheriffOfNotts