springfall2008/batpred #284: Gaps in charging on consecutive charge slots

wilsondw, 2023-11-08:
With combine charge slots = false, Predbat calculates separate SOC targets tor each 30 min slot in the 3 hour flux cheap rate period. Towards the end of each slot, the battery charge often matches or exceeds the target and the charging pauses until the next Predbat run completes and the new target is set. When trying to charge from empty to 100% SOC in the three hour cheap rate window, these gaps in charging of a few minutes every half an hour mean that the battery doesn’t reach 100% by the end of the cheap rate. 

Would it be possible to have a mode in between combine slots true and false where Predbat calculates the plan based on charge slots false but then combines consecutive charge slots in the plan sets the end target SOC for the whole block to maximise the chances of successfully charging to that target?

springfall2008, 2023-11-08:
So I thought I recently made a change to do just that in v7.11.10, what version are you on?

wilsondw, 2023-11-08:
I’m on 7.11.10. Is there a switch to enable this or should it be automatic when combine charge slots is false?

springfall2008, 2023-11-08:
Should be automatic

wilsondw, 2023-11-08:
Don't think it is quite working then :-( The picture shows the grid to battery power flow for last night when it was trying to charge to 100% from empty. It got to 97% - so not far off but would probably have made it another percent or two without the gaps at the end of the half hour slots when SOC reached the target predbat had caclulated for the slot.  Looks like another full charge tonight so will try to grab the logs in the morning. 

[image]

wilsondw, 2023-11-09:
No log from last night as encountered a different problem (div by zero) which resulted in no overnight charge... Will raise a separate issue on that one

wilsondw, 2023-11-10:
Logs and power flow for today's cheap rate charge period attached. There were some REST errors again before the charge window but no div by zero problem so charging went ahead as planned, thanks. Separately, I'll increase the log level on GivTCP to see if it gives any hints on why the REST errors are happening.
[appdaemon log.zip](https://github.com/springfall2008/batpred/files/13316908/appdaemon.log.zip)
[image]

closed 2023-12-13 by springfall2008