springfall2008/batpred #286: Sub Optimal Agile Charging Plan

fboundy, 2023-11-08:
Having got PredBat working with my Solis inverter I don’t understand the plan it is creating. This is tonight on Agile where the cheapest slots are 2-5 am. So why is it charging from 10pm? And why is the cost of Best higher than the cost of Base? I’ve compared with my own (much simpler!) algorithm in the below. This is with the recommended Agile settings.

[image]

If I reduce the maximum Charge/Discharge window count from 48 to 24 I get something which looks much better and has a cost for Best which is lower than Base:

[image]

What have I got wrong?

gcoan, 2023-11-08:
If it’s charging too early then I’ve found this is either caused by issues with the house load data that it’s trying to plan your battery level to support or your min soc which is the minimum battery level to retain is too high.

Can you share the data prediction graph?

The new html plan is also very useful to see what predbat is planning on doing and why

fboundy, 2023-11-08:
Here you are. Hope the chart is readable.

gcoan, 2023-11-08:
Didn't get any chart or html plan, sorry.  Could see the charts on the original submission, just not this reply.

If you reply by email I think it loses attachments 🙁

fboundy, 2023-11-08:
[image]
[image]

gcoan, 2023-11-08:
Yes that doesn’t make any sense.  Here for comparison is my plan, also on agile, and whilst there’s a bit of charging to get to increase the soc to support my higher load, until the cheap period, most of the charging isn’t until the cheap period 

[image]

I’m wondering if the issue is your charge rate that predbat thinks you need to charge for longer to get the battery to 100%?

fboundy, 2023-11-09:
I don't think it's that - it has me at 100% before the cheapest rates even start.

Bart39, 2023-11-09:
Your import /export numbers look a little odd, actual / predicted import and export seem to suggest you are exporting everything you import

fboundy, 2023-11-09:
Think that is a function of the time window the numbers were calculated over.

I’m wondering if I’ve managed to inadvertently break something in the core PredBat class with my mods to the Inverter.

I’ll temporarily revert to the release version and see if the plan makes more sense. It won’t control the inverter correctly but in Read Only it should be ok.

gcoan, 2023-11-09:
I was thinking about it last night.  What do you have your inverter losses set to, charging, discharging, in the inverter and is hybrid set in or off?
I've seen some aggressive charging before when my losses were set higher (eg 5%, 5%, 4%) because of the way predbat valued the electricity.  This may not be all of it because your soc doesn't drop too much, but worth a try?  My losses now set to 3%, 3%, 0%

fboundy, 2023-11-09:
Inverter losses are at the default which seems to give a good match in the Calibration plot but I will have a play. I thought it might be the “battery cost metric” (I forget the exact name) that is set to 3p but this made little difference

gcoan, 2023-11-09:
I find the losses configuration makes quite a difference and I've had to set them lower to what I think they should be.

Charge 3%, discharge 3%, inverter 0% and battery metric 0:
[image]

5%, 5%, 0%, 0:
[image]

5%, 5%, 4%, 0 - results similar to yours:
[image]

fboundy, 2023-11-09:
> Your import /export numbers look a little odd, actual / predicted import and export seem to suggest you are exporting everything you import

This was indeed the issue! SImple error in the yaml. All looks much better now!

closed 2023-11-09 by fboundy