springfall2008/batpred #287: ERROR: Exception raised float division by zero

wilsondw, 2023-11-09:
At 23:30 last night, the predbat log shows there was a problem reading from givTCP over REST API. Then on every subsequent run overnight predbat quits with  ERROR: Exception raised float division by zero. givTCP log does not show any errors and inverter/battery data is available in Home assistant history. Restarting appdaemon this morning solved the problem and the next predbat run completed ok. Predbat log attached.
[float div be zero - 1 appdaemon.log](https://github.com/springfall2008/batpred/files/13304549/float.div.be.zero.-.1.appdaemon.log)

springfall2008, 2023-11-09:
I think I have found the issue, will push a fix today, but the real issue is why REST has stopped working

springfall2008, 2023-11-09:
I've released a fix for this now, I hope it won't happen again

closed 2023-11-09 by springfall2008

wilsondw, 2023-11-09:
Thanks - have updated