springfall2008/batpred #288: Status Error

SheriffOfNotts, 2023-11-09:
Hi Trefor, I keep getting similar messages to the following:

9 November 2023
Changed to Invertor time is 2023-11-09 05:32:30+00:00 AppDeamon
time 2023-11-09 08:45:00.013692+00:00 this is -192.52 minutes
• skewed, Predbat may not function correctly, please fix this by updating
your inverter or fixing AppDeamon time zone
08:45:06 - 44 seconds ago - Supervisor

I've checked and the inverter is set to the correct time and the AppDemon timezone is correct? I also use Chronosync to update the HA time which is correct?

SheriffOfNotts, 2023-11-09:
Restarting MQTT and GivTCP seems to have resolved the error for now - any ideas?

springfall2008, 2023-11-09:
I'm not sure, it seems like a problem with GivTCP rather than Predbat as such

SheriffOfNotts, 2023-11-09:
Thanks Trefor, I’ll raise it with GivTCP