springfall2008/batpred #293: Solis inverter error when enable "Set discharge window"

marcoaddario, 2023-11-09:
When setting batpred to work with Solis inverter `inverter_type: "GS"`, as soon as I set the Discharge window flag `switch.predbat_set_discharge_window` I got an error because the app cannot set the `inverter_mode` as it does not exist on Solis.
This is the trace:

```
2023-11-09 12:45:36.748878 INFO pred_bat: Not setting charging window yet as not within the window (now 11-09 12:45:00 target set_window_minutes 30 charge start time 11-09 23:30:00
2023-11-09 12:45:36.750060 INFO pred_bat: Next discharge window will be: 2023-11-10 23:20:00+00:00 - 2023-11-10 23:31:00+00:00 at reserve 15
2023-11-09 12:45:36.751480 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 11-10 23:20:00 - 11-10 23:30:00
2023-11-09 12:45:36.752883 INFO pred_bat: ERROR: Exception raised argument of type 'NoneType' is not iterable
2023-11-09 12:45:36.760719 WARNING pred_bat: ------------------------------------------------------------
2023-11-09 12:45:36.761537 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2023-11-09 12:45:36.762335 WARNING pred_bat: Worker Ags: {'id': 'ad7f465b0a33416da24a3806eac640a8', 'name': 'pred_bat', 'objectid': 'b8fb5f976fc84d7c8579fba5925c30d4', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7fe4732e1dd0>>, 'pin_app': True, 'pin_thread': 2, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-2'}}
2023-11-09 12:45:36.762507 WARNING pred_bat: ------------------------------------------------------------
2023-11-09 12:45:36.765252 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/config/appdaemon/apps/batpred/predbat.py", line 6655, in update_time_loop
    raise e
  File "/config/appdaemon/apps/batpred/predbat.py", line 6651, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/config/appdaemon/apps/batpred/predbat.py", line 6281, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/config/appdaemon/apps/batpred/predbat.py", line 5613, in execute_plan
    inverter.adjust_inverter_mode(False)
  File "/config/appdaemon/apps/batpred/predbat.py", line 865, in adjust_inverter_mode
    entity = self.base.get_entity(self.base.get_arg('inverter_mode', indirect=False, index=self.id))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 3188, in get_entity
    self._check_entity(namespace, entity)
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 231, in inner_sync_wrapper
    f = run_coroutine_threadsafe(self, coro(self, *args, **kwargs))
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 313, in run_coroutine_threadsafe
    result = future.result(self.AD.internal_function_timeout)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/usr/lib/python3.11/site-packages/appdaemon/adapi.py", line 588, in _check_entity
    if "." not in entity:
       ^^^^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable

2023-11-09 12:45:36.765462 WARNING pred_bat: ------------------------------------------------------------
```

fboundy, 2023-11-09:
Are you running the main release from HACS or is this from the Inverter Support branch on GitHub?

I’ve realised that having put this out into the wild I need to tidy up the versioning on these Beta releases..

marcoaddario, 2023-11-10:
I'm on the 7.11.11 from the main branch (from HACS).
I'm happy to pull the branch version and test it, if you need

fboundy, 2023-11-10:
The main branch on GitHub should be good:7.11.12 - it’s not been release to HACS yet I think.

marcoaddario, 2023-11-11:
Error is no longer present in 7.11.12

Thanks

closed 2023-11-11 by marcoaddario