springfall2008/batpred #298: Feature Request: Group Configs to Simplify

fboundy, 2023-11-10:
I am finding the number of config parameters a bit overwhelming. Would it be possible to set up a Select entity to automatically select a set of entities as per the advice in the Configuration Guide in the docs?  Users can always go "under the hood" and tweak individual parameters if they want but this would help a lot of people get a reasonably good setup.

iainfogg, 2023-11-10:
Sightly different approach, but a way I'd thought about solving the complexity was have some sort of UI, or maybe just a drop down like GivTCP uses, where you can set your tariff. Predbat would then set a while load of parameters in the background for you based on your selection.

Ideally it'd tell you what it was going to do, and what it's done.

When I started I was overwhelmed with the number of options, and I still don't understand some of the settings.

Something like this could make it easy easier for people to get going.

springfall2008, 2023-11-10:
I really like the idea, to be honest it would be fairly instant to use as it would just change other HA settings.

iainfogg, 2023-11-10:
Sorry @fboundy I just re-read your original post and realised I said pretty much the same as you 🤦

fboundy, 2023-11-10:
It just needs a select entity which then sets the configs.

On a related note I think it would be really helpful to have a `description` attribute for each of the `CONFIG_ITEMS`. I've done a couple in a branch called 'config-description'. They are simply published to an attribute called `description` in the HA entity. I haven't worked out quite the best way to use them yet mind...

fboundy, 2023-11-11:
First bash at this: #306

closed 2023-12-10 by fboundy