springfall2008/batpred #299: Freeze Charging

fboundy, 2023-11-10:
Hi Trefor

Trying to get my head around Freeze Charging. Two scenarios:

1. What is the desired behaviour when:

Current SOC is 30%
Target SOC is set to 15%
Status is Freeze Charging
PV is 2kW

Export is 15p and Import is 20p

2. Same conditions as above but at night so no PV

Thanks

fboundy, 2023-11-10:
And a follow-up: Why would you ever turn off `Set Soc Enable`?

springfall2008, 2023-11-10:
Freeze charging is just a shortcut for hold the current battery charge level, this means excess solar would charge the battery but excess demand is from the grid. The cloud simulation is meant to stop it accidentally importing due to clouds in this scenario.

set soC enable when off means you can manually set it yourself but leave predbat to calculate the windows

closed 2023-11-14 by fboundy