springfall2008/batpred #300: ERROR: Exception raised can only concatenate str (not "NoneType") to str

stefan-west1, 2023-11-10:
Hi,

Since updating to V 7.11.11 yesterday, I seem to have this repeating error. I don't believe I had this with previous versions:

2023-11-10 12:15:32.540087 INFO pred_bat: ERROR: Exception raised can only concatenate str (not "NoneType") to str
2023-11-10 12:15:32.548622 WARNING pred_bat: ------------------------------------------------------------
2023-11-10 12:15:32.549152 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2023-11-10 12:15:32.549677 WARNING pred_bat: Worker Ags: {'id': 'aa04e4320df04297a3fc3a43930ae776', 'name': 'pred_bat', 'objectid': '75b70705cc964292bb11149d833e3f53', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x1552118edc00>>, 'pin_app': True, 'pin_thread': 1, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-1'}}
2023-11-10 12:15:32.550140 WARNING pred_bat: ------------------------------------------------------------
2023-11-10 12:15:32.553655 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/conf/apps/batpred/predbat.py", line 6672, in run_time_loop
    raise e
  File "/conf/apps/batpred/predbat.py", line 6668, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/local/lib/python3.10/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
  File "/conf/apps/batpred/predbat.py", line 6318, in update_pred
    self.record_status(status, debug="best_charge_limit={} best_charge_window={} best_discharge_limit= {} best_discharge_window={}".format(self.charge_limit_best, self.charge_window_best, self.discharge_limits_best, self.discharge_window_best), notify=True, extra=status_extra)
  File "/conf/apps/batpred/predbat.py", line 2196, in record_status
    self.call_notify("Predbat status change to: " + message + extra)
  File "/conf/apps/batpred/predbat.py", line 1406, in call_notify
    self.call_service("notify/" + device, message=message)
TypeError: can only concatenate str (not "NoneType") to str

stefan-west1, 2023-11-10:
Should add this is not the HACS version. I'm running this directly from here in Appdaemon docker

springfall2008, 2023-11-10:
This is indicating your notify_devices in your apps.yaml is incorrect and has a None in it, what is it set to?

closed 2023-12-13 by springfall2008