springfall2008/batpred #309: Unexpected charging in peak period with 7.11.12

gcoan, 2023-11-11:
7.11.12 introduced changes to stop unnecessary freeze charging but I've just noticed it is now charging in the peak period when I'm almost on 100% soc and there will be lower rate charging periods before the soc drops to my min soc of 1.5

Here's the plan:
[image]

And the logfile
[appdaemon (5).log](https://github.com/springfall2008/batpred/files/13325957/appdaemon.5.log)

springfall2008, 2023-11-11:
I've broken something, just made a new release which I hope resolves it (my discharge didn't work properly)

gcoan, 2023-11-11:
just realised as well some of the discharge is wrong, its discharging at 11pm at 16.93p only to then charge at 18.26p soon after :-(

[image]

I'll put the new release in and see what it now plans

This looks better, there's no 11pm discharge now
[image]
[image]

And there's no planned peak period charging tomorrow afternoon.  It does seem to have given up on Monday morning, but this is all in the 'unknown octopus rates' period so I'm not too fussed as it'll have the octopus rates before it gets there, but is a bit strange

gcoan, 2023-11-12:
I thought I was running the latest 7.12, but found I was running 7.11.12 so maybe I messed up installing the latest version??

Was surprised to see its just started discharging at 7:20pm which is after the peak period.

Looking at the plan its discharging now at 16.93 only to then freeze charge at 21.1 & 20.79.
Surely it should be retaining/using this charge level now and not discharging to only then need to import again soon after and then the 20:30 slot is cheaper so better to freeze charge then rather than 19:30-20:30?

[image]

Upgraded now to 7.12. 

We're now beyond the 19:30 slot so I can't compare that, but the 20:00 charge looks incorrect, soc still above min and the next slot at 20:30 is slightly cheaper to charge in.
Same at midnight, why freeze charge when import is 17.6 and then charge at 00:30 when import is more expensive at 18.19.  Would have thought these should be the other way round?

[image]

[appdaemon (7).log](https://github.com/springfall2008/batpred/files/13328953/appdaemon.7.log)

springfall2008, 2023-11-12:
>We're now beyond the 19:30 slot so I can't compare that, but the 20:00 charge looks incorrect, soc still above min and the next slot at 20:30 is slightly cheaper to charge in.

The text is a little misleading, notice it says 'charge' but the target is 32% and you are at 35% so it's really not charging.

> Same at midnight, why freeze charge when import is 17.6 and then charge at 00:30 when import is more expensive at 18.19. Would have thought these should be the other way round?

The rates are rounded to the nearest pence so 17.6 and 18.19 are both 18p band. I could probably make it more accurate at some point but there is some speed impact.

I hope that makes sense?

gcoan, 2023-11-13:
Yes that does make sense, thanks.  I hadn't appreciated that the rates were being rounded to the nearest penny and so 17.6 and 18.19 were both being treated as equivalent 18p charge slots.  
9% to 25% soc is a 16% increase of my 14.6kW battery capacity, or 2.3kW so the cost difference is 1.4p from charging at the 00:30 rate rather than the 00:00 rate.
In the grand scheme of things not something I can get too fussed about.  I was just looking at the rate difference numerically and seeing something that didn't make sense.

Might be worth adding something to the docs to explain that rates are rounded to the nearest penny in the optimisation logic. (if this isn't there already)

gcoan, 2023-11-13:
I've installed 7.12.1 now and now understanding the rounding, the plan looks fine

[image]

But, has the logic for discharging priority changed?  Its doing a discharge at the beginning of the flux peak export period and I'm fine with the export as there's excess capacity and can charge later on at a lower rate.  But, I'm sure predbat used to do the discharges at the end of peak periods before which I'd have thought was preferable because if there is unexpected extra load and the export is done early there's a risk that the battery could be exhausted prematurely?  I'm sure the discharges used to retain the charge for as long as possible and then discharge towards the end of the peak period?

springfall2008, 2023-11-13:
> I'm sure the discharges used to retain the charge for as long as possible and then discharge towards the end of the peak period?

That does seem a little odd, do you have the log?

gcoan, 2023-11-13:
Here's the latest plan, has now added a second discharge, but they are both early on in the peak period
[image]

And latest logfile
[appdaemon (8).log](https://github.com/springfall2008/batpred/files/13335525/appdaemon.8.log)

fboundy, 2023-11-13:
So this kind of reinforced my view that has evolved over the last week of tinkering around the edges. Bear with me and please don't take this as criticism: 

We are dealing with a problem of optimisation under uncertainty: both the load and the solar forecast are imprecise (though they are hopefully accurate) forecasts subject to error bars that narrow but never go away. We are attempting to optimise a system based on these forecasts.

This happens to be pretty much what I do in my day job, albeit in a different sphere.

What 30 years of experience has taught me is that there is limited benefit it building too complex a model and attempting to optimise the model parameters  because reality frequently fails to perform as expected. It is better to run many alternative, simpler models and to make decisions based on the range of possible outcomes. 

My feeling is that predbat at the moment is trying to over-optimise. At a fundamental level I don't really understand the need for so many parameters. I have my own, simpler, model and all it uses is the solar and load forecasts, the price and some basic parameters describing the PV system. The problem to be solved is the same whatever the tariff set: minimise the overall cost and the controls are the same: forced battery charge and discharge profiles. In the case where you have a poor export tariff the optimiser should never give you any discharge slots and it should never bring up issues like this which seem to me to be down to complexity in the model being prioritised over accuracy.

I've run tonight with the Eco7/Bad Export and the Agile parameters: opposite ends of the spectrum. There is 4p difference in a £6 total or <1% so the answer is correct but it doesn't need so many parameters to define it. Given that the prices have been rounded to the nearest penny (~1% error maybe more). There is more uncertainty in all the inputs.  My much simpler model (PVOpt in the below) gives near enough the same SOC and grid import profile but with approximately 1/8 the number of lines of code and zero parameters to control:

[image]

So what am I saying? 

I guess it is that the underlying model is good but it doesn't need more bells and whistles because it is always going to be slightly wrong. My experimentation suggests that (for me at least) so long as the inverter and battery are defined correctly the other inputs make little difference and just confuse things. They also make the code much more cumbersome. To improve predbat I would focus on:

- Simplifying the inputs
- Improving the UX and UI
- Building a more flexible code structure allowing for more code re-use 
- Allowing different inverter brands to "plug" in to PredBat - this is tricky as  they all seem to have such different controls I think it can be done with careful thought about the I/O required

Happy to share my code if you are interested. It is much more basic and doesn't fully handle discharge yet. It uses pandas for most of the calculations (and hence numpy beneath that) which may speed it up. It assumes you charge at the full power for the full 30 minute slot which is probably my equivalent of your price rounding.

springfall2008, 2023-11-13:
@gcoan I think it's the second pass thing that's messing it up, try it with it turned off. I was wondering about removing that feature anyhow as I don't see it helps much if any.

springfall2008, 2023-11-13:
@fboundy thanks for your insights, you make some interesting points.

Fundamentally Predbat isn't that complex, it has a model of the Solar/Battery/Load which produces a cost result for a scenario and an optimiser that tries to give the best results.

The challenge is that you can't try all combinations it's an N! type problem and so you have to iterate in some way. This iteration obviously isn't perfect and leads to multiple good solutions in some cases. The problem is the user thinks they can do a better job (and sometimes they can) but as it's automated it has to work in all cases.

How does your simple model work with Agile, there are so many combinations and complexities ?

fboundy, 2023-11-13:
Yes, the key thing is a good model of the PV system which is quite simple when you've figured it out. It took me ages to work out how to deal with Agile but when you have you can generalise that model to everything else. The workflow is this:

1. Run the base model with no forced charge or discharge - yours and mine give identical results.
2. Look for the most expensive import half hour in the plan window.
3. See if we can buy that electricity (accounting for necessary efficiencies beforehand. The window to buy it in is from the slot in question to the the later of now and the point where SOC is 100%. This is Slot 0.
4. Charge the battery at max power in Slot 0 and re-run the model.
5. Repeat 2-4 but exclude Slot 0 because it's been used. Fill Slot 1
6. Repeat until there are no slots where you can buy the electricity earlier.
7. That's the charge cycle done with a max of 48 iterations for a 24 hour window but in reality it takes typically 10-15.

The discharge logic then runs the same but in reverse: what slots could I sell that I can buy cheaper. I haven't quite finished coding this.

That's it. It reduces the N! problem to a much simpler subset.

I've invited you into the repo. You can install it in parallel to PredBat as it doesn't write to any inverter controls as yet - I do that in Automations and Scripts.

fboundy, 2023-11-13:
It needs pandas installed in the AppDaemon config under Python packages:
[image]

iainfogg, 2023-11-13:
Seeing as @fboundy has gone big on his suggestions, I thought I'd cool in too.

I think Predbat is doing a great job for loads of people, and @springfall2008 has been super generous in adding an ever increasing number of options for people's different scenarios.

But I feel the cost of that has been a really complex set of options, plenty of which I still don't properly understand despite being around the project for a while, and despite working with software for a living.

Also, I think it's led to functions getting increasingly long, which makes understanding and testing the code increasingly difficult.

I'm not personally fussed about different inverter support, but that's only because it already works for me (but I certainly see the benefit of it if it can be done in a way that's not adding complexity to the core code, she we have a good way of making sure that we keep the interface in sync between different inverter code).

But if I'd been asked to write a list of where to focus, I think I'd have written the same things that @fboundy wrote:

* Simplifying the inputs
* Improving the UX and UI
* Building a more flexible code structure allowing for more code re-use

I'd been wondering about a few of us having a chat some time to think about direction, but had been planning on waiting a bit longer before suggesting it. But now @fboundy has brought this stuff, we could do that if you fancy it @springfall2008 - or we can just discuss stuff via an issue or whatever if you prefer.

fboundy, 2023-11-13:
It also currently relies on an HA Event to trigger it: either PV_OPT with minimal logging or PV_DEBUG with more verbose logging. I have these on a 10 minute loop in an Automation.

The output is simply a dict of force charge slots which are written to the attributes of an entity. I've then got a couple of scripts that handle interaction with the inverter.

fboundy, 2023-11-13:
> Seeing as @fboundy has gone big on his suggestions, I thought I'd cool in too.
>
> I think Predbat is doing a great job for loads of people, and @springfall2008 has been super generous in adding an [...]

I actually think there is an opportunity to build a really useful tool. To support other inverters we really need people to help develop it who have that model and are prepared to put the time in to make it work. One of the eye-openers for me so far has been just how different each brand is in terms of how you control it. That's another argument for modularising the code.

springfall2008, 2023-11-13:
> Yes, the key thing is a good model of the PV system which is quite simple when you've figured it out. It took me ages to work out how to deal with Agile but when you have you can generalise that [...]

I actually think you will end up with something very similar, you could try just your algorithm inside predbat as an alternative option if you wanted. There’s many complexities which you haven’t addressed yet eg car charging, hybrid inverter DC charging where you can avoid clipping, exporting solar directly without going through the battery etc…

springfall2008, 2023-11-13:
Dare I say we need a project meeting?!

iainfogg, 2023-11-13:
> I actually think there is an opportunity to build a really useful tool. To support other inverters we really need people to help develop it who have that model and are prepared to put the time in to [...]

I feel comfortable with supporting other inverters as long as it can be done in a way which avoids complexity in the core code. That should be possible through good design, but I don't know how much breaking apart of stuff it would need to achieve that with the current starting point.

If what's being passed out is a plan for the inverter specific code to implement, that would hopefully do it (although there may be aspects of a plan which some inverters may or may not be able to implement).

iainfogg, 2023-11-13:
> Dare I say we need a project meeting?!

I'm in 😃

fboundy, 2023-11-13:
> I actually think you will end up with something very similar, you could try just your algorithm inside predbat as an alternative option if you wanted. There’s many complexities which you haven’t [...]

I do think it's a more efficient algorithm - quick test on PredBat with Charge only vs PVOpt gives a 3-fold speed-up

fboundy, 2023-11-13:
> I feel comfortable with supporting other inverters as long as it can be done in a way which avoids complexity in the core code. That should be possible through good design, but I don't know how much [...]

What needs to be passed to the inverter is actually quite simple and I think a clear definition of this would help add more brands. At it's simplest it is one of 4 commands:

1 - Do nothing - ie "Self-Use"
2 - Charge at (x watts or x/50 amp)s for (y minutes or until SOC is z)
3 - Discharge at (x watts or x/50 amps) for (y minutes or until SOC is z)
4 - Hold an SOC of z for y minutes or I tell you otherwise

Everything else is just a variation on that theme.

fboundy, 2023-11-13:
> I do think it's a more efficient algorithm - quick test on PredBat with Charge only vs PVOpt gives a 3-fold speed-up

Though to be fair I think it may be PredBat's extensive logging that slows things down. Mine went from 11s to 22s when I put it in Debug mode...

gcoan, 2023-11-13:
> @gcoan I think it's the second pass thing that's messing it up, try it with it turned off. I was wondering about removing that feature anyhow as I don't see it helps much if any.

No problem.  I'm past the peak period discharge for today now, but here's the bit of the plan for the same period tomorrow with 2nd pass on. The discharge slot is from 18:00-19:00

fboundy, 2023-11-13:
> On the logging, the logging predbat does makes it easier to diagnose when something is awry but wonder whether there needs to be much less logging by default and more of the output produced in various levels of debug mode? Be interesting to test to see what speed up it would make or whether its so negligible that we should leave it where it is.

I've done a bit of testing and the effect of the logging is actually pretty minimal so I'd say stick with plenty for now as it makes troubleshooting much easier.

gcoan, 2023-11-14:
Fair enough. If its minimal improvement then leave it logging.

Thanks for testing

gcoan, 2023-11-14:
> > @gcoan I think it's the second pass thing that's messing it up, try it with it turned off. I was wondering about removing that feature anyhow as I don't see it helps much if any.
> 
> I'll switch second pass back on and see if I see it doing strange behaviour with the discharge plan in future days, and if it does, can test if turning it off makes a difference

@springfall2008 Next morning and the latest plan (with second pass turned on) has a similar pattern, planning a lot of early and then a bit of late peak period discharges. The best metric cost to midnight (actually 23:45) was 592.16:
[image]

At 7:45 I turned off second pass. In the logfile I can see the plan that then is produced (finishing at 7:49) has a best metric cost at 23:45 of 606p. 
But then the next run that starts at 7:50, I hadn't changed anything but the metric cost to midnight (actually 23:50) is increased again to 621p??
The plan for this second "second pass off" is quite a bit simpler and doesn't have the early peak period discharges now, but has jumped in price 28p so does suggest that second pass is making a notable cost improvement?
[image]

Logfile covering these runs 
[appdaemon (9).log](https://github.com/springfall2008/batpred/files/13346104/appdaemon.9.log)

fboundy, 2023-11-14:
I'm also seeing that the recommended Agile settings aren't working. I'm using the groups as specified in the Help. For a 24 hour plan I get:

| Group | Base | Best |
|:----|:----|:----|
| Eco7 / No Export | 471 | 484 |
| Eco 7 / Export | 475 | 471 |
| Agile | 507 | 556 |
| Agile 2nd Pass | 507 | 531 |

So in many cases Best is worse than Base....

springfall2008, 2023-11-14:
I'm not sure you are comparing like for like, in that at the end of your plan the battery ends at 89% which for a 9.5kWh battery is maybe 8.5 kWh which is worth say £1 so in reality Best is probably at least 50p better than base (which I assume is do nothing).

Also do you have metric keep or min_soc set to something non zero?

fboundy, 2023-11-14:
Not sure what `metric_keep` is?

I was running a dev branch so I've switched back to main to check out what is going on

fboundy, 2023-11-14:
OK - that makes a huge difference. What does it do?

fboundy, 2023-11-14:
It was set to 2 which shouldn't make a huge difference then since my min SOC is 1.5

fboundy, 2023-11-14:
OK - weird. My `config-grouping` branch is as I would expect (even with SOC Keep at 2) but `main` is keeping it very high. I'll send you the two logs and see if you can diagnose it. Will be later.

fboundy, 2023-11-14:
> For agile you probably want both min soC and keep as 0 to get an optimal plan

I think that depends on how your inverter/battery reports SOC. With my Puredrive batteries you can’t go below 15% SOC so the nominal 10kWh is a working 8.5 capacity.

springfall2008, 2023-12-17:
>I think that depends on how your inverter/battery reports SOC. With my Puredrive batteries you can’t go below 15% SOC so the nominal 10kWh is a working 8.5 capacity.

Isn't that the same as saying reserve is 15%?

springfall2008, 2024-02-10:
I think this is now fixed, feel free to open a new ticket if not.

closed 2024-02-10 by springfall2008