springfall2008/batpred #334: AppDaemon 0.15.0 Moves Everthing!

fboundy, 2023-11-15:
Just a headsup that this is a Major Update. I'm still working through the consequences and how to get Predbat running again:

https://github.com/hassio-addons/addon-appdaemon/releases/tag/v0.15.0

springfall2008, 2023-11-15:
It seems the simple solution is:

> Add: app_dir: /homeassistant/appdaemon/apps/ to your appdaemon.yaml

I've not tested it yet

fboundy, 2023-11-15:
Not that simple. My apps have been moved to /addon_configs/a0d7b954_appdaemon

It also appears that you can no longer access via File Editor. Not that I do.

iainfogg, 2023-11-15:
With the file editor, there's a config option for the add-on called Enforce Basepath - disabling that should allow access outside of the/config folder

fboundy, 2023-11-15:
OK - I was about to shift everything over to my new NUC anyway so this has just prompted me to do that. All running OK on there. And much quicker than the Pi

DJBenson, 2023-11-17:
Raised a defect against HACS (https://github.com/hacs/integration/issues/3351) to address HACS not being able to see the addons_config folder. If anybody has any information I've missed (or got wrong) please chime in.

closed 2023-12-10 by fboundy