springfall2008/batpred #336: battery reserve issue

andrewmusty, 2023-11-15:
log as requested 

2023-11-15 22:49:53.811165 INFO pred_bat: Adjust reserve to target charge % (set_reserve_enable is true)
2023-11-15 22:49:53.824723 INFO pred_bat: Inverter 0 Current Reserve is 14.0 % and new target is 55 %
2023-11-15 22:50:09.327174 INFO pred_bat: Set inverter 0 reserve 55 via REST successful on retry 0
2023-11-15 22:50:09.473762 INFO pred_bat: Completed run status Charging
2023-11-15 22:50:09.526210 WARNING AppDaemon: Excessive time spent in callback 'update_time_loop() in pred_bat', Thread 'thread.thread-1' - now complete after 86.525428 seconds (limit=30.0)
2023-11-15 22:50:09.585862 INFO pred_bat: --------------- PredBat - update at 2023-11-15 22:50:09.583725+00:00 with clock skew 0 minutes, minutes now 1370
2023-11-15 22:50:09.588236 INFO pred_bat: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 12.183333333333334 minutes
2023-11-15 22:50:09.590600 INFO pred_bat: Predbat version v7.12.1 currently running, latest version is v7.12.2
2023-11-15 22:50:09.592808 INFO pred_bat: Updating HA config version to None
2023-11-15 22:50:09.616094 INFO pred_bat: Inverter type GE max_windows 48 num_cars 0 debug enable is False calculate_plan_every 10
2023-11-15 22:50:12.181642 INFO pred_bat: Found 21600 load_today datapoints going back 14 days
2023-11-15 22:50:13.300774 INFO pred_bat: Car charging hold False threshold 6.0
2023-11-15 22:50:13.303754 INFO pred_bat: Current data so far today: load 18.72 kWh import 10.0 kWh export 0.7 kWh pv 9.9 kWh
2023-11-15 22:50:13.306928 INFO pred_bat: Downloading import rates directly from url https://api.octopus.energy/v1/products/AGILE-FLEX-22-11-25/electricity-tariffs/E-1R-AGILE-FLEX-22-11-25-H/standard-unit-rates/
2023-11-15 22:50:13.310020 INFO pred_bat: Return cached octopus data for https://api.octopus.energy/v1/products/AGILE-FLEX-22-11-25/electricity-tariffs/E-1R-AGILE-FLEX-22-11-25-H/standard-unit-rates/ age 12.15 minutes
2023-11-15 22:50:13.313595 INFO pred_bat: Downloading export rates directly from url https://api.octopus.energy/v1/products/OUTGOING-FIX-12M-BB-23-02-09/electricity-tariffs/E-1R-OUTGOING-FIX-12M-BB-23-02-09-A/standard-unit-rates/
2023-11-15 22:50:13.317551 INFO pred_bat: Return cached octopus data for https://api.octopus.energy/v1/products/OUTGOING-FIX-12M-BB-23-02-09/electricity-tariffs/E-1R-OUTGOING-FIX-12M-BB-23-02-09-A/standard-unit-rates/ age 12.133333333333333 minutes
2023-11-15 22:50:13.327857 INFO pred_bat: Standing charge is set to 42.7665 p
2023-11-15 22:50:14.320028 INFO pred_bat: Rate min forward looking: now 17.4195 at end of forecast 17.4195
2023-11-15 22:50:14.325072 INFO pred_bat: Import rates min 17.42 max 52.51 average 27.0
2023-11-15 22:50:14.426666 INFO pred_bat: Adding rate {'date': '2023-11-16', 'start': '17:30:00', 'end': '18:30:00', 'rate': 225} => 11-15 17:30:00 to 11-15 18:30:00 @ 225.0 date 2023-11-16 00:00:00
2023-11-15 22:50:14.432802 INFO pred_bat: Export rates min 15.0 max 225.0 average 19.38
2023-11-15 22:50:14.437790 INFO pred_bat: Rate thresholds (for charge/discharge) are import 52.01p (0.0) export 15.5p (0.0)
2023-11-15 22:50:14.444609 INFO pred_bat: High export rate found rates in range 99 to 225.0
2023-11-15 22:50:14.600695 INFO pred_bat: High export rate windows [0: 11-16 17:30:00 - 11-16 18:00:00 @ 225.0, 1: 11-16 18:00:00 - 11-16 18:30:00 @ 225.0]
2023-11-15 22:50:14.663994 INFO pred_bat: Low Import rate found rates in range 17.42 to 28.22
2023-11-15 22:50:14.846715 INFO pred_bat: Low import rate windows [0: 11-15 22:30:00 - 11-15 23:00:00 @ 18.02, 1: 11-15 23:00:00 - 11-15 23:30:00 @ 19.18, 2: 11-15 23:30:00 - 11-16 00:30:00 @ 17.42, 3: 11-16 00:30:00 - 11-16 01:00:00 @ 20.29, 4: 11-16 01:00:00 - 11-16 01:30:00 @ 19.2, 5: 11-16 01:30:00 - 11-16 02:00:00 @ 18.94, 6: 11-16 02:00:00 - 11-16 02:30:00 @ 18.96, 7: 11-16 02:30:00 - 11-16 03:00:00 @ 17.64, 8: 11-16 03:00:00 - 11-16 03:30:00 @ 19.18, 9: 11-16 03:30:00 - 11-16 04:00:00 @ 19.2, 10: 11-16 04:00:00 - 11-16 04:30:00 @ 20.35, 11: 11-16 04:30:00 - 11-16 05:00:00 @ 17.9, 12: 11-16 05:00:00 - 11-16 05:30:00 @ 19.4, 13: 11-16 05:30:00 - 11-16 06:00:00 @ 18.74, 14: 11-16 06:00:00 - 11-16 06:30:00 @ 22.95, 15: 11-16 06:30:00 - 11-16 07:00:00 @ 23.57, 16: 11-16 08:00:00 - 11-16 08:30:00 @ 23.11, 17: 11-16 08:30:00 - 11-16 09:00:00 @ 26.64, 18: 11-16 19:30:00 - 11-16 20:00:00 @ 24.72, 19: 11-16 20:00:00 - 11-16 20:30:00 @ 26.52, 20: 11-16 20:30:00 - 11-16 21:00:00 @ 23.15, 21: 11-16 21:00:00 - 11-16 21:30:00 @ 26.52, 22: 11-16 21:30:00 - 11-16 22:00:00 @ 22.05, 23: 11-16 22:00:00 - 11-16 23:00:00 @ 19.2, 24: 11-16 23:00:00 - 11-16 23:30:00 @ 19.18, 25: 11-16 23:30:00 - 11-17 00:30:00 @ 17.42, 26: 11-17 00:30:00 - 11-17 01:00:00 @ 20.29, 27: 11-17 01:00:00 - 11-17 01:30:00 @ 19.2, 28: 11-17 01:30:00 - 11-17 02:00:00 @ 18.94, 29: 11-17 02:00:00 - 11-17 02:30:00 @ 18.96, 30: 11-17 02:30:00 - 11-17 03:00:00 @ 17.64, 31: 11-17 03:00:00 - 11-17 03:30:00 @ 19.18, 32: 11-17 03:30:00 - 11-17 04:00:00 @ 19.2, 33: 11-17 04:00:00 - 11-17 04:30:00 @ 20.35, 34: 11-17 04:30:00 - 11-17 05:00:00 @ 17.9, 35: 11-17 05:00:00 - 11-17 05:30:00 @ 19.4, 36: 11-17 05:30:00 - 11-17 06:00:00 @ 18.74, 37: 11-17 06:00:00 - 11-17 06:30:00 @ 22.95, 38: 11-17 06:30:00 - 11-17 07:00:00 @ 23.57, 39: 11-17 08:00:00 - 11-17 08:30:00 @ 23.11, 40: 11-17 08:30:00 - 11-17 09:00:00 @ 26.64, 41: 11-17 09:30:00 - 11-17 10:00:00 @ 28.22, 42: 11-17 19:30:00 - 11-17 20:00:00 @ 24.72, 43: 11-17 20:00:00 - 11-17 20:30:00 @ 26.52, 44: 11-17 20:30:00 - 11-17 21:00:00 @ 23.15, 45: 11-17 21:00:00 - 11-17 21:30:00 @ 26.52, 46: 11-17 21:30:00 - 11-17 22:00:00 @ 22.05, 47: 11-17 22:00:00 - 11-17 23:00:00 @ 19.2]
2023-11-15 22:50:14.981782 INFO pred_bat: Todays energy import 10.0 kwh export 0.7 kwh cost 234.58 p import 245.08 p export -10.5 p
2023-11-15 22:50:14.993062 INFO pred_bat: PV Data for pv_forecast_today total 10.94 kWh
2023-11-15 22:50:15.004498 INFO pred_bat: PV Data for pv_forecast_tomorrow total 1.47 kWh
2023-11-15 22:50:15.017243 INFO pred_bat: PV Data for pv_forecast_d3 total 7.2 kWh
2023-11-15 22:50:15.027389 INFO pred_bat: PV Data for pv_forecast_d4 total 1.21 kWh
2023-11-15 22:50:15.085463 INFO pred_bat: Today's load divergence 122.44 % in-day adjustment 100.0 % damping 1.0x, Predicted so far 12.36 kWh with 0.0 kWh car excluded and 5.51 kWh import ignored and 0.0 forecast extra, Actual so far 16.08 KWh with 0.0 kWh car excluded and 2.63 kWh import ignored
2023-11-15 22:50:15.203996 INFO pred_bat: Historical data totals for days [7, 14] are [19.03, 15.32] - min 15.32
2023-11-15 22:50:15.210228 INFO pred_bat: Historical days now [7, 14] weight [1, 1]
2023-11-15 22:50:15.213715 INFO pred_bat: Inverter 0 using Rest API http://192.168.1.95:6345
2023-11-15 22:50:16.509808 INFO pred_bat: Invertor time 2023-11-15 22:49:52+00:00 AppDaemon time 2023-11-15 22:50:09.583725+00:00 difference -0.3 minutes
2023-11-15 22:50:16.515353 INFO pred_bat: Reserve min: 10.0% Battery_min:4.0%
2023-11-15 22:50:16.520933 INFO pred_bat: New Inverter 0 with soc_max 19.05 kWh nominal_capacity 19.05 kWh battery rate raw 3000.0 w charge rate 3.0 kw discharge rate 3.0 kw battery_rate_min 0.0 w ac limit 3.0 kw export limit 3.0 kw reserve 10.0 % current_reserve 55.0 %
2023-11-15 22:50:17.802350 INFO pred_bat: Inverter 0 SOC: 9.71 kw 51 % Current charge rate 3000.0 w Current discharge rate 3000.0 w Current power -2252.0 w Current voltage52.0
2023-11-15 22:50:17.807180 INFO pred_bat: Inverter 0 scheduled charge enable is True
2023-11-15 22:50:17.811077 INFO pred_bat: Inverter 0 charge windows currently [{'start': 1350, 'end': 1380, 'average': 0}, {'start': 2790, 'end': 2820, 'average': 0}, {'start': 4230, 'end': 4260, 'average': 0}]
2023-11-15 22:50:17.815252 INFO pred_bat: Inverter 0 Charge settings: 11-15 22:30:00-11-15 23:00:00 limit 55.0 power 3.0 kw
2023-11-15 22:50:17.819573 INFO pred_bat: Inverter 0 scheduled discharge enable is False
2023-11-15 22:50:17.824149 INFO pred_bat: Inverter 0 discharge windows currently [{'start': 1058, 'end': 1062, 'average': 0}, {'start': 2498, 'end': 2502, 'average': 0}]
2023-11-15 22:50:17.827302 INFO pred_bat: Found 1 inverters totals: min reserve 1.9 current reserve 10.48 soc_max 19.05 soc 9.71 charge rate 3.0 kw discharge rate 3.0 kw battery_rate_min 0.0 w ac limit 3.0 export limit 3.0 kw loss charge 3 % loss discharge 3 % inverter loss 4 %
[... cut ...]
2023-11-15 22:50:20.201414 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 11-16 17:30:00 - 11-16 18:30:00
2023-11-15 22:50:20.205737 INFO pred_bat: Inverter 0 Current Target SOC is 55.0 already at target
2023-11-15 22:50:20.210212 INFO pred_bat: Adjust reserve to target charge % (set_reserve_enable is true)
2023-11-15 22:50:20.214861 INFO pred_bat: Inverter 0 Current reserve is 55.0 already at target
2023-11-15 22:50:20.261058 INFO pred_bat: Will not recompute the plan, it is 1.4 minutes old max age 10
2023-11-15 22:50:20.264263 INFO pred_bat: Completed run status Charging

andrewmusty, 2023-11-15:
and after predbat has run again the battery is no longer discharging even when battery is @ 53% and the house is running from grid only. log below

2023-11-15 23:00:00.054480 INFO pred_bat: Inverter type GE max_windows 48 num_cars 0 debug enable is False calculate_plan_every 10
2023-11-15 23:00:01.744255 INFO pred_bat: Found 21600 load_today datapoints going back 14 days
2023-11-15 23:00:02.639922 INFO pred_bat: Car charging hold False threshold 6.0
2023-11-15 23:00:02.643383 INFO pred_bat: Current data so far today: load 19.56 kWh import 10.6 kWh export 0.7 kWh pv 9.9 kWh
2023-11-15 23:00:02.646804 INFO pred_bat: Downloading import rates directly from url https://api.octopus.energy/v1/products/AGILE-FLEX-22-11-25/electricity-tariffs/E-1R-AGILE-FLEX-22-11-25-H/standard-unit-rates/
2023-11-15 23:00:02.650221 INFO pred_bat: Return cached octopus data for https://api.octopus.energy/v1/products/AGILE-FLEX-22-11-25/electricity-tariffs/E-1R-AGILE-FLEX-22-11-25-H/standard-unit-rates/ age 21.966666666666665 minutes
2023-11-15 23:00:02.654274 INFO pred_bat: Downloading export rates directly from url https://api.octopus.energy/v1/products/OUTGOING-FIX-12M-BB-23-02-09/electricity-tariffs/E-1R-OUTGOING-FIX-12M-BB-23-02-09-A/standard-unit-rates/
2023-11-15 23:00:02.658435 INFO pred_bat: Return cached octopus data for https://api.octopus.energy/v1/products/OUTGOING-FIX-12M-BB-23-02-09/electricity-tariffs/E-1R-OUTGOING-FIX-12M-BB-23-02-09-A/standard-unit-rates/ age 21.95 minutes
2023-11-15 23:00:02.664525 INFO pred_bat: Standing charge is set to 42.7665 p
2023-11-15 23:00:03.653127 INFO pred_bat: Rate min forward looking: now 17.4195 at end of forecast 17.4195
2023-11-15 23:00:03.658572 INFO pred_bat: Import rates min 17.42 max 52.51 average 27.01
2023-11-15 23:00:03.753796 INFO pred_bat: Adding rate {'date': '2023-11-16', 'start': '17:30:00', 'end': '18:30:00', 'rate': 225} => 11-15 17:30:00 to 11-15 18:30:00 @ 225.0 date 2023-11-16 00:00:00
2023-11-15 23:00:03.759875 INFO pred_bat: Export rates min 15.0 max 225.0 average 19.38
2023-11-15 23:00:03.764860 INFO pred_bat: Rate thresholds (for charge/discharge) are import 52.01p (0.0) export 15.5p (0.0)
2023-11-15 23:00:03.772237 INFO pred_bat: High export rate found rates in range 99 to 225.0
2023-11-15 23:00:03.936261 INFO pred_bat: High export rate windows [0: 11-16 17:30:00 - 11-16 18:00:00 @ 225.0, 1: 11-16 18:00:00 - 11-16 18:30:00 @ 225.0]
2023-11-15 23:00:03.983158 INFO pred_bat: Low Import rate found rates in range 17.42 to 28.22
2023-11-15 23:00:04.162604 INFO pred_bat: Low import rate windows [0: 11-15 23:00:00 - 11-15 23:30:00 @ 19.18, 1: 11-15 23:30:00 - 11-16 00:30:00 @ 17.42, 2: 11-16 00:30:00 - 11-16 01:00:00 @ 20.29, 3: 11-16 01:00:00 - 11-16 01:30:00 @ 19.2, 4: 11-16 01:30:00 - 11-16 02:00:00 @ 18.94, 5: 11-16 02:00:00 - 11-16 02:30:00 @ 18.96, 6: 11-16 02:30:00 - 11-16 03:00:00 @ 17.64, 7: 11-16 03:00:00 - 11-16 03:30:00 @ 19.18, 8: 11-16 03:30:00 - 11-16 04:00:00 @ 19.2, 9: 11-16 04:00:00 - 11-16 04:30:00 @ 20.35, 10: 11-16 04:30:00 - 11-16 05:00:00 @ 17.9, 11: 11-16 05:00:00 - 11-16 05:30:00 @ 19.4, 12: 11-16 05:30:00 - 11-16 06:00:00 @ 18.74, 13: 11-16 06:00:00 - 11-16 06:30:00 @ 22.95, 14: 11-16 06:30:00 - 11-16 07:00:00 @ 23.57, 15: 11-16 08:00:00 - 11-16 08:30:00 @ 23.11, 16: 11-16 08:30:00 - 11-16 09:00:00 @ 26.64, 17: 11-16 19:30:00 - 11-16 20:00:00 @ 24.72, 18: 11-16 20:00:00 - 11-16 20:30:00 @ 26.52, 19: 11-16 20:30:00 - 11-16 21:00:00 @ 23.15, 20: 11-16 21:00:00 - 11-16 21:30:00 @ 26.52, 21: 11-16 21:30:00 - 11-16 22:00:00 @ 22.05, 22: 11-16 22:00:00 - 11-16 23:00:00 @ 19.2, 23: 11-16 23:00:00 - 11-16 23:30:00 @ 19.18, 24: 11-16 23:30:00 - 11-17 00:30:00 @ 17.42, 25: 11-17 00:30:00 - 11-17 01:00:00 @ 20.29, 26: 11-17 01:00:00 - 11-17 01:30:00 @ 19.2, 27: 11-17 01:30:00 - 11-17 02:00:00 @ 18.94, 28: 11-17 02:00:00 - 11-17 02:30:00 @ 18.96, 29: 11-17 02:30:00 - 11-17 03:00:00 @ 17.64, 30: 11-17 03:00:00 - 11-17 03:30:00 @ 19.18, 31: 11-17 03:30:00 - 11-17 04:00:00 @ 19.2, 32: 11-17 04:00:00 - 11-17 04:30:00 @ 20.35, 33: 11-17 04:30:00 - 11-17 05:00:00 @ 17.9, 34: 11-17 05:00:00 - 11-17 05:30:00 @ 19.4, 35: 11-17 05:30:00 - 11-17 06:00:00 @ 18.74, 36: 11-17 06:00:00 - 11-17 06:30:00 @ 22.95, 37: 11-17 06:30:00 - 11-17 07:00:00 @ 23.57, 38: 11-17 07:00:00 - 11-17 07:30:00 @ 27.56, 39: 11-17 08:00:00 - 11-17 08:30:00 @ 23.11, 40: 11-17 08:30:00 - 11-17 09:00:00 @ 26.64, 41: 11-17 09:30:00 - 11-17 10:00:00 @ 28.22, 42: 11-17 19:30:00 - 11-17 20:00:00 @ 24.72, 43: 11-17 20:00:00 - 11-17 20:30:00 @ 26.52, 44: 11-17 20:30:00 - 11-17 21:00:00 @ 23.15, 45: 11-17 21:00:00 - 11-17 21:30:00 @ 26.52, 46: 11-17 21:30:00 - 11-17 22:00:00 @ 22.05, 47: 11-17 22:00:00 - 11-17 23:00:00 @ 19.2]
2023-11-15 23:00:04.293068 INFO pred_bat: Todays energy import 10.6 kwh export 0.7 kwh cost 245.4 p import 255.9 p export -10.5 p
2023-11-15 23:00:04.303989 INFO pred_bat: PV Data for pv_forecast_today total 10.94 kWh
2023-11-15 23:00:04.315642 INFO pred_bat: PV Data for pv_forecast_tomorrow total 1.47 kWh
2023-11-15 23:00:04.326686 INFO pred_bat: PV Data for pv_forecast_d3 total 7.2 kWh
2023-11-15 23:00:04.337932 INFO pred_bat: PV Data for pv_forecast_d4 total 1.21 kWh
2023-11-15 23:00:04.396941 INFO pred_bat: Today's load divergence 125.43 % in-day adjustment 100.0 % damping 1.0x, Predicted so far 12.32 kWh with 0.0 kWh car excluded and 6.09 kWh import ignored and 0.0 forecast extra, Actual so far 16.66 KWh with 0.0 kWh car excluded and 2.84 kWh import ignored
[... cut ...]
2023-11-15 23:00:19.171560 INFO pred_bat: Adjust reserve to target charge % (set_reserve_enable is true)
2023-11-15 23:00:19.176687 INFO pred_bat: Inverter 0 Current reserve is 53.0 already at target
2023-11-15 23:00:19.216569 INFO pred_bat: Will not recompute the plan, it is 5.0 minutes old max age 10
2023-11-15 23:00:19.220282 INFO pred_bat: Completed run status Hold charging
2023-11-15 23:00:19.251864 WARNING HASS: Error calling Home Assistant service default/notify/notify
2023-11-15 23:00:19.253776 WARNING HASS: Code: 500, error: 500 Internal Server Error

Server got itself in trouble

springfall2008, 2023-11-17:
Is set_reserve enabled?

I think if not you need to turn down the current reserve back to the default

springfall2008, 2023-11-17:
Sorry re-reading it the code is saying the plan is to charge to 53% and hence it's just holding at that percentage.

What does your battery chart or charging plan look like?

andrewmusty, 2023-11-17:
Hi,

It was working ok yesterday / today. Should the problem reoccur I will let you know

Thanks

closed 2023-12-01 by springfall2008