springfall2008/batpred #337: Feature Request: Tariff Comparison

fboundy, 2023-11-16:
Agile rates are going up so am I on the best tariff? It's not simple because your usage on a different tariff would be different but this is a question that Predbat should be able to answer. 

You'd  have to take the past 7 days (say) and optimise for the different tariffs in Read Only and then compare the total costs (including standing charges).

arti273, 2023-11-16:
Have a look at the Octopus Compare app.  It has a load-shifting prediction.  That's not to say it wouldn't be a useful addition to Predbat, just flagging what is already available.
The issue I found with the comparison is you need to take a view over a month, which may exceed most users' historical data in HA. One week in September showed I should jump ship from Agile but I held my nerve and in doing so I saved £127 due to Storm Agnes to Debi causing price plunges.
Predbat would need to access all current relevant (time-of-use) tariffs and ignore the rest - there are 41 current Octopus tariffs, and most are single rate but I don't know how Predbat would know to ignore them for comparison as new products are launched (maybe there is a time-of-use flag in the API?).  Then, would it check EON Drive, OVO, etc.
Then some sort of configuration for the minimum scale factor above baseload that would be shifted and a shift time exclusion range, if users wouldn't run heavy load appliances overnight, etc.  
A useful feature would be for a notification to alert the user that a better tariff might be available.  That's something the apps don't do at present.
However, my gut feeling is it could generate a lot of additional support requests when a similar result could be gained by existing apps. 
Long term though, (particularly with additional inverter support) it could include a referral link to switch to another provider and that could support further development.  That sort of feels out of scope and into the realms of uSwitch though!

gcoan, 2023-11-16:
I agree that the historical usage can help answer this question and predbat's future rate planning can help with optimising battery strategies for the tariff, something that a simple compare of historical usage (eg Octopus compare app) can't do.

But at the moment predbat has no internal knowledge of the different rates, it gets your current rate information from the octopus integration.
If/when we get the ability for quick setup of rate defaults via groups I wonder if this can be achieved with read only mode, setting manual rate overrides and setting a long forecast period?  Then some documentation on how to do a what-if run?

Personally I'd rather not add more complexity to the core code for an edge case.  Maybe a supplement routine if need be.

closed 2023-12-10 by fboundy