springfall2008/batpred #340: Should be charging on Agile pre-peak

fboundy, 2023-11-16:
Batttery at 15% due to tumble dryer on for 1st time in a week. Prices through the roof at 4 but PredBat isn't scheduling a charge:

[image]

[image]

[20231116.log](https://github.com/springfall2008/batpred/files/13378219/20231116.log)

springfall2008, 2023-11-16:
The plan showed you had enough power with solar to get past the peak before charging ?!

springfall2008, 2023-11-16:
It seems soC was stuck on 15, did you turn off discharge ?

springfall2008, 2023-11-16:
Actually maybe your reserve is 15%? 

It seems the rates before peak were not in your low rate threshold, either it’s set too high or if it’s automatic you don’t have enough windows ?

fboundy, 2023-11-16:
15% is the maximum DOD for Puredrive batteries. I suspect it’s to do with thresholds like you say. I don’t really understand why you need a threshold. It’s cheaper that the slot later. That seems to me to be the only logic you need.

fboundy, 2023-11-16:
And there's no point in discharging when the export price is 15p vs min import of >20

springfall2008, 2023-11-17:
The threshold for low/high if set to 0 are automatic, but there's a max windows setting which needs to be high enough to cover the slots you want and so might need increasing.

I don't see any exports at 15?

closed 2023-12-10 by fboundy