springfall2008/batpred #341: Moving away from appdaemon

springfall2008, 2023-11-17:
Appdeamon was an easy way for me to write my first HA app but I’m wondering if it’s time to make Predbat be a 1st class add-on instead.

does anyone know how hard this might be to do, especially get state history from HA sensors?

DJBenson, 2023-11-17:
As an alternative to a complete rewrite, might it be possible to package up the Docker version of AppDaemon along with the Predbat code and make it available as an add-on with the backend (yaml) configuration items surfaced as config entities in the add on ui? You'd then have control over anything like this happening in future as you'd control whether AppDaemon gets updated or not.

JRascagneres, 2023-11-18:
Hey @springfall2008 - Looking at using predbat soonish with an install I'm getting soon and was poking around here and noticed this - I'd like a "1st class" integration version. Accessing history is fully viable. I went and played around in my integration and put together a mini example for you:

```
test = history.state_changes_during_period(
    hass,
    start,
    end,
    "sensor.national_grid_current_carbon_intensity",
    include_start_time_state=True,
    no_attributes=True,
)
```

Can see we can an array:
[image]

Digging into it, can see state:
[image]

springfall2008, 2023-11-18:
> Hey @springfall2008 - Looking at using predbat soonish with an install I'm getting soon and was poking around here and noticed this - I'd like a "1st class" integration version. Accessing history is fully viable. I went and played around in my integration and put together a mini example for you:

That's awesome, would you be able to share the source code of this example?

iainfogg, 2023-11-18:
I don't have any answers for this, but I think this may give an opportunity to use dependency injection to get objects passed in to Predbat that support getting state, rather than the Predbat class inheriting a load of methods for getting state, logging etc that makes having unit tests on Predbat code far harder.

Happy to detail this more some time if you like @springfall2008

iainfogg, 2023-11-18:
And I'd personally avoid the "Docker it all up" approach and concentrate on a simpler implementation

JRascagneres, 2023-11-18:
@springfall2008 My integration doesn't have this code in source - I just poked around to see if I could help find a solution for you... Is there anything else I can help with? Happy to write more examples or more actual source code for the proposed integration.

Yeah I'm kinda against the docker it up method. Seems like a cheat and not a good way to build a real integration.

@iainfogg What do you mean RE dependencies - you should and pretty much need to use the tools and methods offered by home assistant?

iainfogg, 2023-11-18:
@JRascagneres I was referring to the way in which, with Appdaemon you end up with your class having to extend an Appdaemon class. That causes your class to automatically have a bunch of methods for logging, getting state etc.

I don't know if you're familiar with dependency injection, it means you would pass in an object into the constructor for your class that can do, for example, logging, and call methods on that object to do the logging rather than having the log method in your own class.

The advantage is that when you write unit tests, you can pass in a mocked object that looks like a logger but it doesn't actually log during the test. It means you can test a small amount of your code in isolation.

I'm just typing on my phone so can't really pull together an example here.

So it doesn't mean you would use HA for getting state, it's more about doing it by having your code more split up, and passing objects in that can get state rather than doing it directly in your code.

So with the state scenario, you can write a unit test where the function calls on object to get state, but pass in a mock object that will provide the state for the scenario you want to test.

As I say, not easy to do this on my phone, hopefully that helps a bit. I can type something better up another day if you like.

springfall2008, 2023-11-18:
> @springfall2008 My integration doesn't have this code in source - I just poked around to see if I could help find a solution for you... Is there anything else I can help with? Happy to write more examples or more actual source code for the proposed integration.

I guess what I'm after is something I can copy/paste which takes care of the framework and then I can integrate it with the Predbat code. 

Which project did you edit to try this, was it HA National Grid?

JRascagneres, 2023-11-18:
@iainfogg ah good old mocking interfaces. The bane of my existence. Makes sense. You can totally have methods to obtain state, obtain history etc that could make testing easier.

@springfall2008 yup that's the one I have. I can try to look at throwing something together if you'd want

springfall2008, 2023-11-18:
> 
> @springfall2008 yup that's the one I have. I can try to look at throwing something together if you'd want

That would be great, just a dummy application that I can install that can read/write HA entities would allow me to port Predbat into it.

iainfogg, 2023-11-18:
@JRascagneres yep, I'd looked at adding tests to Predbat but because so many methods come from the class that Predbat inherits from (the Appdaemon class), it's hard to separate those out from the code you want to test.

springfall2008, 2023-11-18:
@iainfogg i thought you could mock member functions in a class ?

iainfogg, 2023-11-18:
It's not the normal way of doing it, but I think it may be possible. I have a branch on my repo where I'd started looking at adding tests, and I was part way through trying to mock member functions and hadn't yet cracked it.

Ideally I'd personally like to see smaller self contained classes, and all dependencies passed in via the constructor.

In this case, if you're busy trying to move away from Appdaemon, the easiest way is to keep Predbat as it's, but pass in one or a number of objects that do logging, getting and saving state etc. and simply save them in the constructor (e.g. saving the locker as self.logger) and use them in the code in place of self.log() and things like that, so they become self.logger.log() for example.

JRascagneres, 2023-11-18:
@springfall2008 Sure. I'll throw something together today. Something that can read / write states and something that runs every 5 minutes. Something I'll have to investigate is how you set the GivTCP stuff... I assume you set the states of other integrations? Or is it done in a different way?

springfall2008, 2023-11-18:
Sounds good.

the inverter is controlled via HA states or via a REST api

the framework is running every 5 minutes, creating certain switches/inputs, Reading inverter status in HA and then writing inverter controls in HA

JRascagneres, 2023-11-18:
@springfall2008 Still working on it...

https://github.com/JRascagneres/DummyIntegration

It'll be here. At the moment its barely doing anything and doesn't fully work :P You can poke it tho

JRascagneres, 2023-11-19:
@springfall2008 Okay, take a peek at that. Has a single sensor which increments every minute. Also has some commented out lines which read / write other states to non-internal sensors.

springfall2008, 2023-11-19:
Thanks that’s great, I’ll take a look as soon as I get time and see if I understand the code

JRascagneres, 2023-11-19:
There may be better ways to satisfy the use case here but the one I've given you 'works' so this can be tweaked and messed with until  you're happy with it. My integration is essentially just a tonne of sensors from an API so its a bit different

iainfogg, 2023-11-21:
@springfall2008 just wanted to get a sense from you of how time-critical moving away from AppDaemon is? Do you think it's a 'need to do it now to sort out the mess from the last AppDaemon release' issue, or just a 'move away when we get a chance to rework it over some weeks', or something else?

JRascagneres, 2023-12-06:
Is this still a plan?

springfall2008, 2023-12-09:
> Is this still a plan?

Yes I'd still like to do it it's just I haven't had time. 

@iainfogg I think we have a workaround for now, but I'm concerned appdeamon will continue to get broken in future.

iainfogg, 2023-12-09:
I've started having a bit of a look, but it was way more complicated than I was hoping, so I've not got far. I can see why you started with Appdaemon.

However, if it's a proper HA integration, you can have better support for things like unit testing, which is a positive.

Also, you can run your code in proper debug mode and step through line by line in your IDE which can be really helpful, I've been doing that to try and fix issues in another integration that I use recently.

So I think it could be good for Predbat, it's just far from simple.

JRascagneres, 2023-12-09:
Yeah its defo a good choice to move away. Reduce depencies, make it easier for users, make it easier for you to maintain etc. 

@iainfogg 
It largely depends what your after - What particular bits are difficult? I can try to help with solutions. I've found that docs aren't amazing but that honestly reading existing built-in integrations from github is super helpful.

iainfogg, 2023-12-10:
@JRascagneres thanks I think there were a few things that surprised me when I started reading the docs, especially as I've heard people talking about how good the docs are.

Firstly, the docs around building something new felt really unclear about what the difference is between a platform and an integration, and also when you'd build an integration v. when you'd pbuild an add-on.

Also, everything seemed geared up around developing something that connects with an external device, rather than something like Predbat that connects with HA entities to calculate new info and carry out actions against them.

And the example code for a new integration felt like it didn't really explain why the particular methods it creates are there, and how to customise it for your own integration.

And I was surprised by how much code there was for a hello world type integration.

I tried creating a basic integration and it never got picked up by HA, even though it was in custom_components. Didn't figure this out, but I've not tried since, so may still be able to get that sorted.

I work with code a lot, but this felt way harder than I expected.

Having said that, getting up a debug instance of HA was documented really well and I got that side going quickly and easily.

JRascagneres, 2023-12-10:
@iainfogg 

So yeah... I agree on that one, they're not clear. From my perspective:
- Add-ons are essentially things that are disconnected from home assistant. ie. Could run entirely independently as a usual docker container on another machine. It just may be advantageous to the user to run on the same machine. Things like Frigate and GivTCP are good examples - They can run on another machine but as they are connected to the smart home stuff people might want them on the same machine.
- Integrations are... well... integrated into home assistant. They directly write / read data from within the home assistant software. For example, you couldn't run the Octopus integration on another machine, it couldn't write / read to home assistant sensors (yes, I know MQTT exists but still, its not the same)

Yeah a lot of integrations are about setting up sensors and reading / writing to them, thats the bread and butter of home assistant. However, you CAN write to things like predbat wants to do, see my example code I posted. 

Yeah, there is a hell of a lot of boilerplate that takes a while to get running right but once its there its fairly simple and you build on it.

See my dummy integration code I posted, you can copy that in and have a play.

Yeah I'm a software engineer by day and this was a huge learning curve for me. Honestly I went and read and searched a bunch of the default built-in integration code on the home-assistant core repo to understand and help build my own.

Agreed, the easy setup is nice.

springfall2008, 2023-12-27:
So I'm wondering what would be the pro/cons of making Predbat be an add-on like GivTCP instead. The reason I was thinking down this lines is it appears the MQTT can be used to access HA history and create and track it's own sensors and switches. It seems that the port to this model would be quite easy as it's just a few function calls I would have to change to MQTT. This would also mean Predbat could run outside the HA machine if you wanted.

On the other hand an integration seems nicer but looks really quite difficult.

iainfogg, 2023-12-27:
@springfall2008 I'm definitely no expert - quite the opposite in fact!

I've been spending hours here and there trying to figure out HA integrations, and it certainly is hard, as the documentation seems to do a pretty poor job of explaining what you need to know - the docs that explain about building integrations don't even seem to match up with the starter code that their wizard type script generates.

Still, I think once we have a framework in place that does work in HA, it'll be better to be integrated. It's less CPU load because you're not running a container, allows for configuration UI through the HA config settings (that's the main thing I've worked on so far), and allows easy integration with other sensors, HA events etc.

I'd hoped this would give a good opportunity to break Predbat up into smaller chunks - maybe one for gathering load data, one for building a plan based on load data and config, one for passing that plan out to an inverter, then plenty of inverter classes that can implement a plan in the way that suits that particular inverter.

Those chunks could potentially be available as services in HA that you can trigger manually if that's useful, or people could set their own frequency for recalculating the plan, for example (to reduce CPU load).

And a component should be able to define its own sensors and switches, and also respond to events (e.g. new Octopus rates being published, or whatever).

But as I said, it's pretty hard figuring out how to do all that in a well organised way.

For now I'll keep tinkering with what I've got, and I'll shout if I get a useful framework in place, unless you decide on the add-on route - feel free to do that though and I can park this stuff.

And in favour of the add-on route, I guess you can code it how you like, without having to fit into an existing application - I'm just not clear on how you do integrate with HA via MQTT, but that could be easier.

gcoan, 2023-12-27:
Stepping back and thinking why change from app daemon to a different delivery method:
- lower "footprint", less dependencies to install & maintain
- more consistent end user experience 
- better logging access
- easier to setup through GUI config

Either an add on or an integration would meet these needs.  Personally I think most end users don't really appreciate the difference between add on's and integrations and don't really understand why HA has two ways of doing much the same thing (three if you include HACS).

I think an integration would be the better experience for an end user as they see most extensions to HA as being via integrations, and there is less CPU overhead with an integration.

But, two big reasons why add-on's would be better:
- you don't need to restart HA when upgrading an add-on
- add-on's better support version control, automatically taking a backup that you can restore if need be

iainfogg, 2023-12-27:
I'm not sure add-ons help with "less dependencies to install and maintain", as I assume you'll be responsible for a full Docker image of packages etc., you'll potentially be running a web server and your own full web UI - it feels like there's more to maintain there than in Appdaemon. But you'd get a lot more control in an add-on.

I'd be interested to know if add-ons get any logging support for free, or whether they have to run their own logging too.

And any UI consistency presumably needs doing manually for add-on UIs, copying HA styles etc., I'm guessing.

But there are lots of guesses and assumptions here, so please do correct anything here that's not right :)

JRascagneres, 2023-12-27:
@iainfogg Did you take a look at the dummy integration I threw together. Admittedly it isn't easy to get started but once the basic functions are in place its not all that difficult. I would definately admit though that the docs aren't great. Like I say I just copied other integrations available in the HA repo.

I really don't think add-on is the best plan. I understand the advantage of being able to run this externally as a separate 'thing' to HA but does anyone REALLY do that?  

> add-on's better support version control, automatically taking a backup that you can restore if need be
I'd argue that integrations have benefits too in the way you can easily change versions for a given integration (in HACS) rather than needing the whole backup process. 

I wonder how much work it'd be to switch to an add-on though? I really really think integration is the way to go but if add-on is super easy to add then it'd be preferable in the near term to the current situation.

iainfogg, 2023-12-27:
@JRascagneres yes thanks, I used that, plus an article I found, plus looking at other examples in the HA repo.

It's frustrating that the starter code that HA can generate doesn't match the docs though.

JRascagneres, 2023-12-27:
@iainfogg yeah I gave up with that generation stuff

springfall2008, 2023-12-27:
I've been playing with an add-on, here is a dummy one:

https://github.com/springfall2008/batpred_addon

I got as far as reading the state of all the HA entities and printing them out.

I copied some template stuff from GivTCP so I can cut/paste it's MQTT code to create sensors if required.

Let me know what you think vs the Integration in terms of effort.

iainfogg, 2023-12-27:
Most of the code in there is API handling code, as you're not running in HA itself. I'm guessing an add-on does end up with a bunch of code for handling things that HA already does.

I expect you'd accomplish the same thing with less code in an integration.

At the moment my experimentation is mostly focused on allowing entry and modification of config, rather than creating and modifying entities - that was the next thing I was starting to look at.

Once I have the entities stuff worked out I'll post some code so you can see @springfall2008

springfall2008, 2023-12-27:
If you are on the case then please continue, just looking at options

springfall2008, 2023-12-31:
I had a look at the quick and dirty solution of packaging Predbat with AppDaemon as an add-on and it works 

https://github.com/springfall2008/appdaemon-predbat

The nice thing is that it doesn't need the latest Predbat version as once installed it can be updated from inside Predbat and so no need to make new releases.

Let me know what you think, it would solve some things but obviously it's a change and maybe not worth doing if we can get a real HA integration sometime soon.

[image]
[image]
[image]

JRascagneres, 2023-12-31:
I think it's probably a good idea for  even a temporary solution. Would be nice if you're able to integrate the config UI too.

gcoan, 2023-12-31:
Great stuff Trefor, very quick to get it working.

For new users this means one thing less to install, configure and have to keep upgraded.

If it's running as a separate docker container how does changing apps.yaml and logging work - still to HA filesystem?

And tweaking predbat.py (experts only), the same?

Or do you have to ssh etc?

springfall2008, 2023-12-31:
Effectively it’s no different from appdaemon, you get a new directory in add_ons which contains the apps.yaml and predbat.py. The only real difference is that you don’t need HACS to install Predbat

iainfogg, 2024-01-01:
I know it won't apply for most people, but for some it'll mean they have two Appdaemon instances running, I've no idea what the repercussions of this would be.

Personally I'd lean to just leaving it as it is for now, bearing in mind you've got all the documentation and videos that are already in place.

I'm making slow progress on the HA integration. I've got two pieces of code, one at https://github.com/iainfogg/predbat_ha which is based on a template I found, which I'm slowly customising and experimenting with.

The other is just local code, and it's where I've experimented with building a config flow, similar to what you see with the Octopus integration and others, where you set config on install, and can further modify it post install.

My plan is to get the two codebases into one, with ideas on how to either wrap the existing code, or to break it up.

However, as I've moaned about before, the HA documentation is disappointing so it's taking way longer than I expected, and there's a lot to consider with the asynchronous code model that HA follows, plus lots of undocumented helper functions, means there's a lot of trial and error.

I'm happy to keep plugging away with it, but feel free to go at it yourself if you've got more time than me.

Realistically it'll probably be another month before I've got something that feels like a good starting point, I expect @springfall2008

iainfogg, 2024-01-01:
One thing you could do @springfall2008 is to add Predbat to the list of HA brands at https://github.com/home-assistant/brands so a custom integration will show an icon for Predbat.

Ideally I'd suggest doing it for 'predbat' and 'predbat_ha' (I'm using the latter in my repo so it doesn't clash with the existing entities, but we would potentially revert to 'predbat' at some point).

JRascagneres, 2024-01-01:
@iainfogg Let me know if you have any questions or anything I can help with. I've struggled through a lot of the config flow stuff and the rest of it 😂

iainfogg, 2024-01-01:
Having said all that, I just saw the post on Facebook about this and it looks like people are positive about it. I guess the big win I'd not appreciated earlier is significantly simplifying the installation which I know plenty of people struggle with, so ignore my previous hesitation about this temporary approach 😃

iainfogg, 2024-01-01:
> @iainfogg Let me know if you have any questions or anything I can help with. I've struggled through a lot of the config flow stuff and the rest of it 😂 

Thanks, the config and option flow stuff feels ok now. I've got a few things I'm experimenting with, then I'll maybe shout up.

I'd be interested to know your thoughts on how to run a long running piece of code like Predbat, bearing in mind the asynchronous event loop - guessing a task or a service would be the best place? @JRascagneres

JRascagneres, 2024-01-01:
@iainfogg Hey. I don't have a tonne of experience in that but you can see what I did here - https://github.com/JRascagneres/HA-NationalGrid/blob/master/custom_components/national_grid/coordinators/national_grid.py#L41-L61

Essentially runs that async update function every 5 minutes. I think you can probably do similar?

iainfogg, 2024-01-02:
Thanks, that looks good, I'll have a good look through it in a few days, cheers @JRascagneres

springfall2008, 2024-01-02:
> Having said all that, I just saw the post on Facebook about this and it looks like people are positive about it. I guess the big win I'd not appreciated earlier is significantly simplifying the [...]

Yeh its just a stop-gap to make the install easier until we get something better :)

JRascagneres, 2024-01-02:
@springfall2008 I don't run it myself **yet** but will this conflict with another app daemon install? If so how / why? Is there a way to separate it?

divenal, 2024-01-11:
Rather than starting a new issue, my question doesn't seem _too_ off-topic for this one...
To what extent does this actually depend on home assistant ?  Would it be feasible to structure it in such a way that it _can_ run within home assistant, but doesn't have to?   Which I imagine would mean putting the core functionality in a separate module, which the HA integration class calls into, but which can also be called from a standalone wrapper, for example.

Absolutely fine if you're not interested - I can just fork and have a play around myself.  I have absolutely nothing against HA in principle, but having to install it seems overkill if it's merely required to provide a container in which the custom component runs. I have an always-on pi running various stuff, and I'm sure I could get HA running on it, or on the other pi I have lying around, but if all it actually needs is just a wrapper which calls the fn to update the prediction every so often, that doesn't sound too hard. And avoids overheads of having all the HA code running on there too.

iainfogg, 2024-01-14:
I think there's a potential argument for some of the logic being in a separate component. However, significant blocks of data are probably / possibly easiest to get from within HA, such as:

- house load data
- solar forecasts from Solcast
- Octopus rate info from the Octopus Energy integration

They could all be retrieved from other sources, but doing so without using HA integrations would add a bunch of complexity that probably negates the benefit of trying to move too much of it into a separate non-HA, I reckon.

I can still imagine you could move some of the core calculation code into a non-HA environment, but you'd need to create a non-HA way of wrapping all those data sets I mentioned earlier (and probably others I didn't mention).

How's that sounds @divenal ?

divenal, 2024-01-14:
Ah, I hadn't noticed that the solcat was coming from another sensor: I had just assumed this code was just connecting to the solcast server directly and pulling it from there. Okay, yes, that does make it more awkward. Not insurmountable, but I can appreciate that you'd probably prefer not to have maintain additional abstractions around those various data sources. Not to worry, then...

springfall2008, 2024-01-14:
Right now Predbat supports reading from the GE Cloud but it can't control the inverter without GivTCP.

Direct access to Solcast and Octopus isn't that hard so could be done if there was a point, but I'm trying to avoid re-writing code that others maintain.

I guess it depends what you want to achieve really?

iainfogg, 2024-01-14:
GivTCP, I believe, is written in such a way that you can use it with or without HA.

But I'm still not sure there's enough benefit to carving out parts of Predbat without adding a lot of extra code that's currently covered by that other integrations.

divenal, 2024-01-16:
Certainly wasn't my intention that any extra code be added. More that it was organised in such a way that it could be more easily replaced if desired. Eg rather than having fetch_pv_datapoints() inline, it could be in a separate module Sensors or something. Exactly the same code, but makes it easier to drop in a different implementation of Sensors. Then the core of badpred can be used unchanged.

iainfogg, 2024-01-16:
Personally I'm hoping that as part of moving to a new way of delivering Predbat's functionality, we can also break it apart into a variety of classes which can be swapped out and replaced as long as you follow the interface.

There's a long way to go, but I think it'll help in the long run.

springfall2008, 2024-08-10:
I think this is sort of done now as the new add-on and the code has been split up (first pass).

@iainfogg do we want to resolve the ticket?

iainfogg, 2024-08-11:
> I think this is sort of done now as the new add-on and the code has been split up (first pass).
> 
> @iainfogg do we want to resolve the ticket?

Yep @springfall2008 sounds good, well done getting it done 😎

closed 2024-08-12 by springfall2008