springfall2008/batpred #355: Charge freeze force to true causing draw from grid when there is plenty of battery

AGLAISTER, 2023-11-21:
The removal of charge freeze as a configuration option (not even available in expert mode) is causing my SOC to be frozen at some high levels (37%+) causing me to draw from the grid at prices higher than when charging will occur later.

Please restore charge freeze to be a configurable option in expert mode

springfall2008, 2023-11-21:
Sure I can certainly put it back if you like

My idea was to delete stuff and see who wants it

AGLAISTER, 2023-11-21:
I would prefer to have the option either through expert mode or another way that doesn't get overwritten when an update happens.
Being on Agile import only my biggest savings in winter come from timeshifting load, loss of the charge freeze control is causing the battery level to be held at the expense of importing more - i've gone from predicted charges of less than £3-4 for 48 hours to over £7 on the new version

springfall2008, 2023-11-22:
I've restored this toggle now in expert mode in the latest release https://github.com/springfall2008/batpred/releases/tag/v7.13.5

I have also skewed the metric a bit so when enabled it happens less often

gcoan, 2023-11-22:
In the changelog for 7.13.5 https://github.com/springfall2008/batpred/compare/v7.13.4...v7.13.5 it reintroduces predbat_inverter_soc_reset in the documentation, but predbat doesn't seem to be outputting this

AGLAISTER, 2023-11-22:
Thanks new update has fixed the issue - did take me having to toggle freeze on -> off but now back to previous condition

closed 2023-11-22 by AGLAISTER

springfall2008, 2023-11-22:
>is causing my SOC to be frozen at some high levels (37%+) causing me to draw from the grid at prices higher than when charging will occur later.

Just as an observation if you have around 14% round trip losses (AC->inverter(4%)->charge(3%)->discharge(3%)y->inverter(4%)->AC) then charging the battery at 20p is the same as using the grid at around 23p. If you enable metric_battery_cycle at 2p then this becomes 25p. I'm not sure if you are accounting for this?