springfall2008/batpred #359: PV data not showing in Power chart and does not seem to match predbat plan

AGLAISTER, 2023-11-22:
Since update to 7.13.x I do not seem to be seeing PV data in my power graphs and I suspect it is not feeding into my plan
[image]

Also looking at the predbat plan the values of PV data do not seem to be matching those shown in the log
[image]

2023-11-22 00:25:06.989008 INFO pred_bat:          [  00:25,   01:25,   02:25,   03:25,   04:25,   05:25,   06:25,   07:25,   08:25,   09:25,   10:25,   11:25,   12:25,   13:25,   14:25,   15:25,   16:25,   17:25,   18:25,   19:25,   20:25,   21:25,   22:25,   23:25,   00:25,   01:25,   02:25,   03:25,   04:25,   05:25,   06:25,   07:25,   08:25,   09:25,   10:25,   11:25,   12:25,   13:25,   14:25,   15:25,   16:25,   17:25,   18:25,   19:25,   20:25,   21:25,   22:25,   23:25]
2023-11-22 00:25:06.997121 INFO pred_bat:     SOC: [    3.9,    3.51,    3.14,    2.87,    2.57,    2.36,    2.08,    1.81,    1.38,    1.16,    0.87,    0.59,    0.63,    0.53,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.39,    0.38,    0.38,    0.56,    0.74,    0.77,     0.6,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38]
2023-11-22 00:25:07.004902 INFO pred_bat:   STATE: [  g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<bf+,   g<bf+,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g~be-,   g<be-,   g<be-,   g~be-,   g~be-,   g~be-,   g~be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g~be~,   g<be-]
2023-11-22 00:25:07.013085 INFO pred_bat:    LOAD: [    0.0,    0.36,     0.7,    0.95,    1.23,    1.42,    1.68,    1.93,    2.36,    2.71,    3.24,    3.87,    4.27,    4.76,    5.56,    6.03,    6.73,    7.44,    7.86,    8.17,     8.6,    8.89,    9.29,    9.62,   10.11,   10.35,   10.55,   10.78,   10.93,   11.47,    11.8,    12.1,   12.58,   12.94,   13.56,   14.62,   15.17,   15.57,   15.95,   16.21,    16.5,   16.99,   17.73,   18.71,   22.18,   22.92,   23.49,   24.04]
2023-11-22 00:25:07.022042 INFO pred_bat:      PV: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    0.05,     0.2,    0.48,    0.87,    1.36,    1.81,    2.04,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.13,    2.27,    2.58,    3.13,    3.83,    4.47,    4.92,    5.03,    5.04,    5.04,    5.04,    5.04,    5.04,    5.04,    5.04,    5.04]
2023-11-22 00:25:07.030364 INFO pred_bat:  IMPORT: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    0.44,    0.86,    1.56,    2.28,     2.7,    3.01,    3.44,    3.72,    4.13,    4.46,    4.95,    5.18,    5.39,    5.61,    5.77,    6.31,    6.63,    6.94,    7.37,    7.61,    7.92,    8.47,    8.56,    8.56,    8.56,    8.56,    8.64,    9.13,    9.88,   10.85,   14.32,   15.06,   15.63,   16.19]
2023-11-22 00:25:07.038563 INFO pred_bat:  EXPORT: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0]
2023-11-22 00:25:07.047046 INFO pred_bat:    CAR0: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0]
2023-11-22 00:25:07.055494 INFO pred_bat:  METRIC: [  44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   44.12,   53.29,   61.83,   82.27,   110.6,  126.74,  136.14,  145.04,  151.15,  158.83,  163.31,  216.88,  221.68,   225.6,  229.73,  232.42,  242.19,  248.87,  255.89,  266.33,  271.93,  278.64,  289.36,  291.09,  291.09,  291.09,  291.09,  294.19,  313.92,  342.32,  372.52,  445.34,  461.63,  472.48,  479.71]
2023-11-22 00:25:07.646258 INFO pred_bat: Battery has 13.17 hours left - now at 3.9
2023-11-22 00:25:07.761474 INFO pred_bat: predict base10 end_record 11-24 00:25:00 final soc 0.38 kwh metric 591.81 p metric_keep 466.42 min_soc 0.38 @ 11-22 11:15:00 kwh load 24.65 pv 2.08
2023-11-22 00:25:07.770445 INFO pred_bat:          [  00:25,   01:25,   02:25,   03:25,   04:25,   05:25,   06:25,   07:25,   08:25,   09:25,   10:25,   11:25,   12:25,   13:25,   14:25,   15:25,   16:25,   17:25,   18:25,   19:25,   20:25,   21:25,   22:25,   23:25,   00:25,   01:25,   02:25,   03:25,   04:25,   05:25,   06:25,   07:25,   08:25,   09:25,   10:25,   11:25,   12:25,   13:25,   14:25,   15:25,   16:25,   17:25,   18:25,   19:25,   20:25,   21:25,   22:25,   23:25]
2023-11-22 00:25:07.778655 INFO pred_bat:     SOC: [    3.9,    3.51,    3.14,    2.87,    2.57,    2.36,    2.08,    1.81,    1.37,     1.1,    0.73,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38,    0.38]
2023-11-22 00:25:07.786493 INFO pred_bat:   STATE: [  g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g~be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<bf+,   g<bf+,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g~be~,   g<be-]
2023-11-22 00:25:07.794713 INFO pred_bat:    LOAD: [    0.0,    0.36,     0.7,    0.95,    1.23,    1.42,    1.68,    1.93,    2.36,    2.71,    3.24,    3.87,    4.27,    4.76,    5.56,    6.03,    6.73,    7.44,    7.86,    8.17,     8.6,    8.89,    9.29,    9.62,   10.11,   10.35,   10.55,   10.78,   10.93,   11.47,    11.8,    12.1,   12.58,   12.94,   13.56,   14.62,   15.17,   15.57,   15.95,   16.21,    16.5,   16.99,   17.73,   18.71,   22.18,   22.92,   23.49,   24.04]
2023-11-22 00:25:07.803049 INFO pred_bat:      PV: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    0.03,    0.13,    0.33,    0.58,    0.86,    1.11,    1.22,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.26,    1.27,    1.32,    1.41,    1.58,    1.79,    1.97,    2.05,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08]
2023-11-22 00:25:07.811431 INFO pred_bat:  IMPORT: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    0.07,    0.21,    0.48,    1.16,     1.6,     2.3,    3.02,    3.44,    3.75,    4.17,    4.46,    4.87,     5.2,    5.68,    5.92,    6.13,    6.35,     6.5,    7.04,    7.37,    7.68,    8.14,    8.45,    8.98,    9.89,   10.24,   10.48,   10.76,    11.0,   11.29,   11.78,   12.52,    13.5,   16.97,   17.71,   18.28,   18.83]
2023-11-22 00:25:07.819718 INFO pred_bat:  EXPORT: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0]
2023-11-22 00:25:07.828007 INFO pred_bat:    CAR0: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0 [... cut ...]  23.49,   24.04]
2023-11-22 00:25:08.545995 INFO pred_bat:      PV: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    0.05,     0.2,    0.48,    0.87,    1.36,    1.81,    2.04,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.08,    2.13,    2.27,    2.58,    3.13,    3.83,    4.47,    4.92,    5.03,    5.04,    5.04,    5.04,    5.04,    5.04,    5.04,    5.04,    5.04]

springfall2008, 2023-11-22:
I’m seeing the same issue, will try to fix it tomorrow

springfall2008, 2023-11-24:
I've made a release with a fix for this now: https://github.com/springfall2008/batpred/releases/tag/v7.13.8

AGLAISTER, 2023-11-24:
Thanks just installed the latest version. Had to edit the card to get it to
spring back to life but it's now working again

Many thanks

closed 2023-11-24 by springfall2008