springfall2008/batpred #362: Plan 'state' explanations in documentation

slopemad, 2023-11-22:
Might it be worth adding something to the documentation to explain the various 'states' in the Predbat plan?  Some are more obvious than others, but it might be worth a little explainer as to what 'FreezeChrg', 'Charge/Discharge' and 'FreezeDis' might actually mean from the point of view of what it would change on the inverter?

For example, I'm a little confused about what the plan actually might intend to do during the 'Charge/Discharge' state here...?
[image]

gcoan, 2023-11-22:
+1 from me on this point. I’ve looked and it isn’t covered that I could find

The charge and discharge means within the 30 min slot, some of the 5 minute predbat runs will be charging and some will be discharging

slopemad, 2023-11-22:
That's what I thought,  but then I also have "Calculate Discharge on charge slots" set to False...  I've since upgraded to 7.13.6, that Charge/Discharge part of the plan has gone away.

gcoan, 2023-11-22:
I've got calculate discharge on charge slots on

Makes sense to me to have it on for periods like now when export rate is higher than import rate

[image]

Only issue it keeps freeze charging not charging #357

slopemad, 2023-11-22:
What I have right now is a flat battery and a plan to charge right now even though the current slot is the most expensive for the next few hours (it isn't charging as it's in read only, it doesn't really need to start charging until 1.30am.  Might be an artifact of the recent feature of adding the next 4 hours into the plan?

[image]

springfall2008, 2023-11-23:
This will likely be the result of metric keep being non zero forcing a charge

slopemad, 2023-11-23:
So here's what I want to try and understand, specifically around FreezeDis and FreezeChrg, and what they are intended to mean.  Here is the current plan:
[image]
A bunch of 'FreezeDis' for the next few hours.  To me, 'FreezeDis' means "No more discharging of the battery", whereas actually it seems to really mean 'No more charging of the battery'.  In 'FreezeDis', the inverter's Battery Charge Rate is changed to zero, which in practice just means 'export excess solar'.

(I added this extra show_limit line into the code at line 5236 to visualise what 'limit' means)
```
                    state += "FreezeDis&rarr;"
                    show_limit = str(int(limit))
```

So, does FreezeDis simply mean 'Eco mode, export excess'?

Meanwhile, FreezeChrg seems to mean 'No Charging, no Discharging either, run the house off mains', i.e. Pause the battery.  The logs suggest that at the last change to Freeze charging, SOC was set to 4, and discharge rate was set to 0. (Which would, of course, mean no mains charge, no discharge at all, excess solar into the battery though).

Is this the intention with these two states?

springfall2008, 2023-11-23:
So maybe the names are confusing

Charge -> Battery is charging -> Grid charges battery and covers load
Charge Freeze -> Battery is held at the current charge level (charging with soc target == current level) -> Grid covers load, battery can charge from solar only
Discharge -> Battery is force discharging -> Battery covers load and exports to grid
Discharge Freeze -> Battery is held at the current discharge level (discharging with target == current_level) -> Solar is exported, battery will cover the load

springfall2008, 2023-11-23:
Added some documentation here: https://github.com/springfall2008/batpred/blob/main/docs/what-does-predbat-do.md

gcoan, 2023-11-23:
Thanks Trevor.

It looks like the freeze discharge ha# got muddled/copy paste partly from an earlier status

Discharge Freeze - A slot where the current battery is held at the current level. The battery covers the load. Any solar generated will be exported.

The battery can’t be held at current level if it’s also supporting house load?

I think there’s some more intro needed.  You hav3 these predbat modes under battery terminology, but they’re not really battery terms, they are describing wha5 predbat does with the battery.  The descriptions talk about slots and the earlier text on the page doesn’t introduce the 30 minute slot and plan concept.
Also Eco mode IS a battery term but predbat corresponding status is idle and the difference isn’t clear

springfall2008, 2023-11-24:
I've updated the wording again, have a look and see what you think

gcoan, 2023-11-24:
Found it in the main branch. Reads much better thanks.

closed 2023-11-24 by springfall2008