springfall2008/batpred #366: Low priority enhancement: populate battery_hours_left when charging

gcoan, 2023-11-24:
I display the battery hours left entity on my dashboard as it's a useful predbat indicator of how long the battery will support house load (-pv generation) but when the battery is being charged it's set to zero which reduces its usefulness.  Is it possible to continue population based on current soc etc?

[image]

gcoan, 2023-11-24:
[image]

springfall2008, 2023-11-24:
I found a recent bug and hopefully fixed this

gcoan, 2023-11-24:
Its populated now when charging (and soc>4%) but the figures look a bit "generous"

0.37kWh=1.75 hours
[image]

When discharging earlier, 0.42kwh=2.17 hours
[image]

current load is 1.33kwh for this slot

slopemad, 2023-11-24:
31kWh in your Heat Pump though! Mine's used about 14kWh today, and my house has been over 20 degrees all day.  Probably not a Predbat issue ;-)

gcoan, 2023-11-25:
> 31kWh in your Heat Pump though! Mine's used about 14kWh today, and my house has been over 20 degrees all day. Probably not a Predbat issue ;-)

Yeah the heat pump can use a lot.  Yesterday was only 14kW and the day before 21. Its very weather dependent.  The total house bill is Around £5-7 a day and since I was generating about £5 a day in summer, I'm OK with that.

closed 2023-11-28 by gcoan