springfall2008/batpred #375: Feature request: safety margin / keep for peak period

wilsondw, 2023-11-26:
For those of us with a battery that isn’t large enough to see us through a grey winter’s day and on flux or similar tariffs where the day rate either side or peak is the same, you have to set SOC keep to 0 to get a good plan out of Predbat. In this mode, it will allow discharge during the day, but hold the charge at sufficient level to (just) get through the peak period before battery runs out and the using grid through to next cheap rate period.

This is great so long as:
- consumption in peak rate is not higher than predicted (I set load scaling of 1.2 to try to account for this)
- your battery doesn’t suffer from SOC crash / low end adjustments. 

Unfortunately, my battery suffers from low end SOC adjustments almost daily… so whilst predbat’s prediction is good, the battery will decide to stop discharging  before it reaches 4% so I end up importing towards the end of peak. This is a side effect or the latest firmware for me so suspect others will start to see the same.

Would it be possible to either:
- change SOC keep behaviour so that it aims to keep the minimum level at the end of peak periods rather than during the whole plan?
Or
- add a safety margin parameter to set the min SOC planned during peak periods?
- 
Thanks

springfall2008, 2023-12-17:
What I've done instead in the latest release is two things:

1. There is now a load_scaling10 setting which default to 1.1 (+10%) which will skew the worse case a bit more towards higher load - so if you increase this setting the plan will be more conservative.
2. The keep setting has been changed to be softer, it only applies at a nominal 50% weighting and the next 4 hours are weighted even lower.

Please have a play with the new release and see if it helps with your concerns

wilsondw, 2023-12-18:
Thanks Trefor, will play over next few days and let you know

wilsondw, 2023-12-22:
@springfall2008 I've experimented with the two options over the last few days without a lot of luck. I can get the initial plan to look good but then it is not followed as the day progresses. I think something else has changed that is causing predbat to discharge during peak rate as it recalculates what to do over the next few sessions. This discharge doesn't make a lot of sense given (a) flux peak export rate = standard import rate and the battery would be used covering load at standard import rate and (b) it may reduce income from future savings sessions. No discharge is shown in the plan at the start of the day so this looks to be something in the optimisation logic for the next 8 sessions (although I haven't had a chance to look at the plan just before peak rate to confirm this). Other than keep and load_scaling10, no other settings have changed so I'm not clear what is causing predbat to think the export is a good idea (which wasn't the case with prior versions). I tried turning expert mode off to see if it was any of my other settings causing this and the result was more extreme as freeze charge kicked in and kept extending so that the battery was at 100% at the start of the peak rate... followed by the discharge.

More broadly, the revised SOC keep behaviour in the long term plan looks most promising as the initial plan each day looks good so long as I set the keep less than 0.4 (e.g. 0.2 or 0.3). This results in an initial plan that holds the charge at 100  for a couple of hours in the morning after the cheap rate, then allows discharge to empty by around 9/10pm - safely after the evening peak period. If this happened, it would be perfect as it then gives the battery a few hours to do its offline OCV and adjust itself to minimise SOC problems the following day. Unfortunately, although this plan looks good, it is not followed and the near term plan adjustments result in charge being held for longer followed by the discharge at peak. 

FYI, if keep is set 0.4 or higher, the long term plan becomes rather inefficient as rather than holding SOC it seems to allow discharge to house load and then tops up with import at standard rate. It also tries to stretch things out so that the keep level is maintained through to the next overnight cheap rate.

wilsondw, 2023-12-27:
Updated to 7.14.26 and behaviour has changed. With SOC keep of less than 0.5, the plan simply charges overnight cheap rate and then discharges to cover house load through the day reaching empty and importing during peak rate. 
[image]

With SOC keep set to 0.5, the plan looks more sensible and gets through peak, but then charges up again at standard rate after peak and before the overnight cheap rate.

For now I”ve reverted to 7.14.19.

wilsondw, 2024-01-01:
@springfall2008: Good news: Updated to 7.14.28 (and now to .30) with SOC Keep of 0 and load_scaling 1.1 and freeze charging off, the plan now looks great - charging overnight cheap rate and then holding the charge level until late morning before allowing discharge to cover house load, comfortably getting me through the peak rate with a safety margin before the battery runs out of juice a couple of hours later. This then gives a couple of hours for the battery to do an offline OCV, sort itself out before the cheap rate charge window. For my system, at this time of year, that's the perfect plan. Many thanks for all the hard work on predbat. Happy to close this issue.

closed 2024-01-01 by wilsondw