springfall2008/batpred #387: Saving session import rate

adamwylde, 2023-11-29:
Tiny bug I just noticed - during the saving session it’s showing the IMPORT rate to have the saving session premium applied as well as to export.

[image]

Really make no difference but thought I’d let you know. 

Thanks for the amazing Predbat

RobinCu, 2023-11-29:
It’s not a bug. The saving session payments are based on net reduction, so any import is effectively costing £4/kWh more.

gcoan, 2023-11-29:
I agree with @RobinCu that the session saving payments are based on net reduction, but thinking about it further I think @adamwylde is more right and import rate shouldn't see the saving session rate included.

My logic for this is that if we have a battery and we're on agile or a time of day tariff then we're going to be using it anyway and avoiding peak rate imports.  So our normal pattern of use will only be imports caused by home demand exceeding the output rate of the inverter.  i.e. imports we can't avoid anyway.  So predbat loading the import price to avoid importing energy isn't going to change our consumption pattern any more than normal anyway.
And, if we have excess energy in the battery (or we just want to receive the very high saving session export rate) then at the moment with import>export, predbat won't be exporting in the saving session period:
[image]

Personally I want to export in the saving session period even if it exhausts the battery and I have to import in that period and in the rest of the peak as the export rate is so good it will compensate for the peak rate import

adamzest, 2023-11-29:
> It’s not a bug. The saving session payments are based on net reduction, so any import is effectively costing £4/kWh more.

right yeah I suppose that makes sense as the net saving would be right.

> Personally I want to export in the saving session period even if it exhausts the battery and I have to import in that period and in the rest of the peak as the export rate is so good it will compensate for the peak rate import

I see now though that importing in that period isn't beneficial as it will count against the saving session - so it wont compensate it will just get wiped out.

adamwylde, 2023-11-29:
> It’s not a bug. The saving session payments are based on net reduction, so any import is effectively costing £4/kWh more.

right yeah I suppose that makes sense as the net saving would be right.

> Personally I want to export in the saving session period even if it exhausts the battery and I have to import in that period and in the rest of the peak as the export rate is so good it will compensate for the peak rate import

I see now though that importing in that period isn't beneficial as it will count against the saving session - so it wont compensate it will just get wiped out.

gcoan, 2023-11-29:
Oh really, I hadn’t seen anywhere explicit that it was difference in net import/export

I’d assumed it was difference in import + difference in export

adamwylde, 2023-11-29:
The FAQ is ambiguous and you could have spotted a gap. I guess its just an assumption given they are basing it on the half hour smart meter readings they receive I figure that if you export 3kwh and import 1kwh they'll conclude that is 2kwh export net?

They do clearly include export in the baseline (although this could still be as you describe two independent measures):
> we calculate that you usually export 2kWh during that time and don't import anything, then if you export 5kWh during the Saving Session, we'd credit you for this extra 3kWh that you're giving back to the grid (5kWh - 2kWh).

Which I think lends weight to the fact its the net import/export?

adamwylde, 2023-11-29:
From the DNO:

https://www.nationalgrideso.com/document/267731/download

How we evidence load shedding?
A: For each unit, during each contracted settlement period, participants will need to submit 3 datasets to ESO:
1) total half-hour baseline of their participating consumers
2) total half-hour readings for the outturn demand of their participating consumers
3) total half-hour calculation for the delivery of their service by their participating consumers, the delivery
volume will be the difference between the metering and baseline.

Suggests theres just a baseline and demand figure not split by import/export for the purpose of the calculation.

gcoan, 2023-11-29:
> From the DNO:
> 
> Suggests theres just a baseline and demand figure not split by import/export for the purpose of the calculation.

Agreed, but the DNO submission is a total for all participating customers, not broken down by individual customer.

I was assuming import and export were treated separately by Octopus as they are two separate meters, billed separately.  The Octopus FAQ as you say is ambiguous as to how the calc  is done

closed 2023-11-29 by adamwylde