springfall2008/batpred #389: Charging to 100%

Philhews, 2023-11-29:
When charging to any soc below 100 predbat disables the charge and sets the reserve once it gets there. This seems to work fine, but when charging to full it looks like it fails to set the reserve so the battery then discharges a bit only to be recharged by predbat on its next calculation.

I am unable to manually set the reserve to 100 either via GivTCP (it immediately reverts to the previous value rather than setting it) so I think it's an inverter thing not a predbat thing but I was wondering if there's any way that the method of holding charge can be changed (config or otherwise to use the pause option maybe) or I can limit where predbat will charge to at 99 instead (which works)? 
[image]
[image]

springfall2008, 2023-11-29:
Maybe try this in apps.yaml:

  # Workaround to limit the maximum reserve setting, some inverters won't allow 100% to be set
  # inverter_reserve_max : 99

Philhews, 2023-11-29:
Will give it a go, thanks.

springfall2008, 2023-12-03:
I've changed the default apps.yaml to be 98% now to avoid these issues

closed 2023-12-03 by springfall2008