springfall2008/batpred #392: Supressing "Predbat status change to: Idle" notifications

Speshman, 2023-11-29:
Is it possible to supress only these notifcations from the phone app?  I seem to get quite a few of them and I'm not sure I need them but want to keep all the others (Android if that makes a difference)

springfall2008, 2023-11-29:
You can either turn them off with the notify settings in home assistant or you can change the notified devices list in apps.yaml

Speshman, 2023-11-29:
Hi trefor,  I want the notifications just I seem to get quite alot of idle notifications throughout the day.

springfall2008, 2023-11-30:
Oh I see, in theory you only get it when it changes to idle from something else e.g. charging->idle, is that not the case?

gcoan, 2023-11-30:
I see a lot of notifications as well, it's better than it was with the previous notification approach but I get multiple idle notifications and notifications when predbat swaps between charging and hold charging - I can see that happening in the soc chart as well 

I think @Speshman just wants to surpress the Idle notifications, but I see a lot regardless 

[image]
[image]
[image]
[image]

Speshman, 2023-11-30:
I seem to get idle messages on  repeat even when there hasn't actually been a state change.

springfall2008, 2023-12-04:
> I seem to get idle messages on repeat even when there hasn't actually been a state change.

Do you have multiple inverters?

Speshman, 2023-12-04:
Nope just one. Although it seems to no longer be doing it on the latest release

Speshman, 2023-12-05:
[image]

Randomly happened again, frequency is less.

Speshman, 2023-12-14:
[image]

springfall2008, 2023-12-16:
Are you restarting home assistant or rebooting appdaemon? I wouldn't expect this normally unless something changed.

Speshman, 2024-05-11:
Finally worked out what's causing it. Losing API connection to Renault and then picking it back up. When it does pick it up the predbat idle status is going to notifications.

[image]

closed 2024-09-01 by springfall2008