springfall2008/batpred #404: Cannot access local variable ‘best_price_discharge’ where it is not associated with a value

b1nuzz, 2023-12-02:
2023-12-02 06:39:35.086871 WARNING pred_bat: ------------------------------------------------------------
2023-12-02 06:39:35.087423 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2023-12-02 06:39:35.087840 WARNING pred_bat: Worker Ags: {'id': 'e0f5a2212f08479b82f8b080d1f4e8eb', 'name': 'pred_bat', 'objectid': '55e2b137decf49f28b0596f22e9c7b13', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7ffbe59b8950>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2023-12-02 06:39:35.088151 WARNING pred_bat: ------------------------------------------------------------
2023-12-02 06:39:35.090940 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9227, in update_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9222, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 8708, in update_pred
    self.calculate_plan(recompute=recompute)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 7619, in calculate_plan
    self.optimise_all_windows(self.end_record, load_minutes_step, pv_forecast_minute_step, pv_forecast_minute10_step, metric, metric_keep)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 6920, in optimise_all_windows
    self.rate_best_cost_threshold_discharge = best_price_discharge
                                              ^^^^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'best_price_discharge' where it is not associated with a value

2023-12-02 06:39:35.091381 WARNING pred_bat: ------------------------------------------------------------
2023-12-02 06:44:19.443017 WARNING pred_bat: ------------------------------------------------------------
2023-12-02 06:44:19.443484 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2023-12-02 06:44:19.443880 WARNING pred_bat: Worker Ags: {'id': '215f970e729f43f5aba69dfd5910e1e7', 'name': 'pred_bat', 'objectid': '55e2b137decf49f28b0596f22e9c7b13', 'type': 'scheduler', 'function': <bound method PredBat.run_time_loop of <predbat.PredBat object at 0x7ffbe59b8950>>, 'pin_app': True, 'pin_thread': 0, 'kwargs': {'interval': 300, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-0'}}
2023-12-02 06:44:19.444243 WARNING pred_bat: ------------------------------------------------------------
2023-12-02 06:44:19.445933 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9249, in run_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9245, in run_time_loop
    self.update_pred(scheduled=True)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 8708, in update_pred
    self.calculate_plan(recompute=recompute)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 7619, in calculate_plan
    self.optimise_all_windows(self.end_record, load_minutes_step, pv_forecast_minute_step, pv_forecast_minute10_step, metric, metric_keep)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 6920, in optimise_all_windows
    self.rate_best_cost_threshold_discharge = best_price_discharge
                                              ^^^^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'best_price_discharge' where it is not associated with a value

2023-12-02 06:44:19.446319 WARNING pred_bat: ------------------------------------------------------------

springfall2008, 2023-12-02:
I’ll fix this but today, but do you have calculate charge disabled?

b1nuzz, 2023-12-02:
Hi. 
Thanks Trefor, 
It’s enabled.

springfall2008, 2023-12-02:
Fixed in https://github.com/springfall2008/batpred/releases/tag/v7.13.20 let me know how you get on

b1nuzz, 2023-12-02:
Fixed for me. Thank you!

closed 2023-12-02 by springfall2008