springfall2008/batpred #405: Respecting SOC Keep

JonathanLew1s, 2023-12-02:
I have SOC Keep at 2.5kWh, expecting that the system will work to maintain SOC at a % above the reserve, yet I do not see the system planning to hold/maintain charge at that level.

2.5 would equate to around 12%, plus reserve at 4%. Shouldn't the system at a minimum hold the charge at 16% in the low rate slot and not plan to get to 4% before holding? Strange it calculates down to 4% and then hold in a medium rate slot.

Tried both 24 and 48 hour windows, tried adjusting SOC keep. 

[image]

JonathanLew1s, 2023-12-02:
Changed it to 3.5 and no change in plan.

[image]

[_appdaemon_logs.txt](https://github.com/springfall2008/batpred/files/13536526/_appdaemon_logs.txt)

springfall2008, 2023-12-03:
This is because your rate_low_threshold is set to 0.8 and so Predbat can't use any charge slots earlier in the day. Either set it to 0 for automatic or go to something like 0.85 or 0.9

closed 2023-12-10 by JonathanLew1s