springfall2008/batpred #412: Frequency of releases

iainfogg, 2023-12-03:
@springfall2008 one of the things that's great about Predbat is how rapidly you keep innovating to add new features etc.

And one of the things that's tricky about Predbat is how rapidly you keep innovating and doing new releases ;)

I'm thinking especially of less confident users, and wondering whether it's better to have two release schedules. One (which we could call beta), which would be exactly as it is now. And another one (stable) which would only be released every so often, maybe monthly (or even less frequently), which would bundle up all the beta changes (as long as they felt stable enough) - and the only other releases on the stable release would be bug-fixes for features already included in a stable release.

That way, people who don't want tons of updates can update once every month or two and follow a clear, laid out document that tells them any modifications they need to make (or confirms there are none), and they know that anything else that pops up in between is a bug fix without any features / modifications needed.

HACS would support the beta idea, so people would need to opt in to get the very frequent releases. These releases could be done off `main`.

And adding a `stable` branch which is taken off `main` ready for release and tagged there would cover the stable releases. You could still do bugfixes on `stable` and merge them back into `main` so that stays up to date with the fixes too.

It adds a bit of complexity, but not much, and I wonder if it would help less technical users who get a bit more swamped with updates.

The trickiest problem is doing a bugfix on `stable` on some code that's been modified on `main`, but that's a merge conflict you'd just need to resolve when merging it back in.

What do you think? Might be worth floating the idea on FB and seeing if there are people who would find it beneficial. I've noticed a few posts recently about updates, so thought it was worth floating.

iainfogg, 2023-12-03:
Also thought of dependencies on other integrations - people on the `stable` release would also need to avoid whacking on updates of the things Predbat relies on otherwise they'd potentially break Predbat by upgrading one of the others, and would then need to go onto the beta release to fix it.

That feels a bit messy now I think about it, but interested what you think.

springfall2008, 2023-12-03:
I think you have a valid point, although managing multiple branches is also hard work.

Maybe I should make more use of beta releases and then only do production releases when they are stable?

iainfogg, 2023-12-03:
Yeah, that's another way of doing it, but I think you still potentially hit the same issue unless you have a beta branch or something - if you only have main, and you tag a proper release commit, then do some beta commits and merge them to main and tag them, you've got it all together on the same branch. Then if your stable release needs a bugfix you don't have a way to do that without releasing all your beta features, unless you have a second branch (or create one at that point, I guess).

You may well be right, yours is a simpler way of handling it.

iainfogg, 2023-12-05:
Yep, I agree finding a good way to give all the necessary upgrade info together is important, if I'm upgrading a bunch of versions in one go I sometimes miss changes I should have made.

springfall2008, 2024-09-01:
I've sort of slowed down to once a week unless its an urgent fix now

closed 2024-09-01 by springfall2008