springfall2008/batpred #416: Next planned charging period?

gcoan, 2023-12-03:
I thought it would be useful to put on my dashboard what the current period rate and when predbat is next due to charge and discharge and at what rates.

current period rate is predbat.rates and predbat.rates_export

But next charge and discharge seem to be harder to find.

I've got next low rate start, end and cost and same for high rate:
[image]

But looking at the plan, 20:00-20:30 is not planned for import nor 20:00-21:00 for export (and in fact the next best export period would be tomorrow's peak):
[image]
and then a minute later the 20:00 slot became freeze discharge (soc is currently 4% so its a moot point)
[image]
And then at 20:20 the latest plan swapped back from 20:00 being a freeze discharging slot!

I looked at the attributes of predbat status and the best_charge_window is encoded there, but not sure what the start and end refer to - minutes from now?
[image]

Is this information easily accessible?  Are the next low and high rates being populated correctly?

springfall2008, 2023-12-04:
I think the problem is for Agile the next low and high rate slots aren't very meaningful as most slots are considered but many aren't used.

Do you want the next charge slot and the next force export slot or something else?

gcoan, 2023-12-04:
Agreed. I’d like to know the next charge slot and next force export slot, although the export is far lesser priority as very little force exporting at the moment and I know when the higher rate export slots are.

Thanks

gcoan, 2023-12-06:
Thanks @springfall2008, so making sure I understand the results attribute correctly, its a list of charging limits and start times.  So the next charge slot will be the first array entry in results[] where the % is >0,  which will be results[0] if we're currently in a charge slot, or results[1] if its in the future?
[image]

springfall2008, 2023-12-06:
I’d ignore the results part it’s for the chart.

There’s a start and end time sensor and then the rate attribute gives you the import rate

gcoan, 2023-12-06:
I upgraded to the main branch.  Can see predicted charge start time sensors now, don't know if these are new or I Just never saw them before?

Curiously there seems to be gaps in the charge start time sensor and its currently undefined:
[image]

[image]

But the plan clearly shows charging in the octopus power up event this afternoon:
[image]

Here's the logfile
[appdaemon (1).log](https://github.com/springfall2008/batpred/files/13581801/appdaemon.1.log)

springfall2008, 2023-12-06:
Are you looking at the best version of these sensors ?

springfall2008, 2023-12-06:
Start and end time best are what’s planned. The other ones are what’s configured currently

gcoan, 2023-12-06:
OK found them:
          - entity: predbat.best_charge_start
          - entity: predbat.best_charge_end
          - entity: predbat.best_discharge_start
          - entity: predbat.best_discharge_end
Thank you

springfall2008, 2023-12-06:
Yes sorry, and limit has the rate encoded