springfall2008/batpred #419: charge being held too high

AGLAISTER, 2023-12-04:
I updated from 7.13.14 -> 7.13.23
Immediately as I did I noticed that the charge was never being allowed to fall below about 74%

[image]
I added the new Set Reserve Enable to my dashboard and set it to disabled (response of the dashboard was slow and was flicking back to previous settings

I rolled back to 7.13.14 initially this looked OK  but then after altering min metric improvement the plan has gone back to not letting the charge below 74%

[image]

I am now barely an hour away from peak prices a near full battery and it is looking like using grid during peak times

[appdaemon.log](https://github.com/springfall2008/batpred/files/13547703/appdaemon.log)

AGLAISTER, 2023-12-04:
All seems to have resolved now - I don't understand what has happened because the settings now are identical to after the update
However its back up and working so I'm closing this issue
[image]

closed 2023-12-04 by AGLAISTER

springfall2008, 2023-12-04:
Is it working with the latest version now?

What I meant was when set_reserve_enable was on Predbat may have set your reserve and then you disabled it leaving the reserve set higher

AGLAISTER, 2023-12-04:
Yes I've switched back to the latest version and seems to be working now