springfall2008/batpred #422: Helping people give the right info for problem solving

iainfogg, 2023-12-04:
There are loads of Facebook posts of people asking for help, but giving inconsistent pieces info about what's going on - they may show the plan card, or bits of the settings, or maybe part of the logs.

I wondered if there's a way of the users clicking on something that dumps all the info needed for problem solving in a consistent way, I think it could make it easier to support people without you having to ask for all the stuff they've not provided.

No idea what this looks like technically though, sorry!

marcoaddario, 2023-12-05:
for github I think it may be useful to create an issue template with the relevant infos that can help identify the problem, the environment, the specific events that make the issue happen and the support info (log, screenshot, etc).
What all this info are in detail? I don't know yet...
For the issues reported on FB... I don't think there is the option to create a templated post... but it may help creating a pinned post with the instruction on what is a must, what is best to have and what nice to have while reporting issues.

springfall2008, 2023-12-13:
I've created the issue template for bug and features now which might help.

springfall2008, 2023-12-15:
I added a self check when Predbat starts to try to catch appdaemon configuration issues

iainfogg, 2023-12-15:
Sounds good @springfall2008 

What info would you want people to post on Facebook when asking for help? Could that go in an FAQ maybe? 

And are there common questions you get asked that could also go in FAQs so you and others can just post a link to that, to save typing the same thing many times over? I don't know how you do it!

springfall2008, 2023-12-16:
Added a new feature when debug mode is on it will write a yaml file predat_debug.yaml containing all the settings and the plan

springfall2008, 2023-12-16:
> Sounds good @springfall2008
>
> What info would you want people to post on Facebook when asking for help? Could that go in an FAQ maybe?
>
> And are there common questions you get asked that could also go [...]

Sounds like a good idea, I must do that

iainfogg, 2023-12-16:
> Added a new feature when debug mode is on it will write a yaml file predat_debug.yaml containing all the settings and the plan

That's a great idea! Hopefully that'll make a big difference.

closed 2024-09-01 by springfall2008