springfall2008/batpred #423: solis - continuously disable charging, even if in charge window or in read-only mode

marcoaddario, 2023-12-05:
a couple of days ago I have found in the morning that the inverter did not charged the battery during the night (even if the plan was to charge from 23:30 to 5:30). Yesterday night I kept an eye on the log and I have seen at the window start time that the target SOC was not correct even if in the plan it was set to 100%

```
2023-12-04 23:30:02.616237 INFO pred_bat: Inverter 0 SOC: 0.71 kw 10 % Current charge rate 3599.0 w Current discharge rate 3599.0 w Current power 78.0 w Current voltage45.9                                                                                                                                                                           │
2023-12-04 23:30:02.618718 INFO pred_bat: Inverter 0 scheduled charge enable is False                                                                                                                                                                                                                                                                  │
2023-12-04 23:30:02.621243 INFO pred_bat: Inverter 0 charge windows currently []                                                                                                                                                                                                                                                                       │
2023-12-04 23:30:02.624865 INFO pred_bat: Solis Energy Control Switch setting 3 is correct for disable grid  charging                                                                                                                                                                                                                                  │
2023-12-04 23:30:02.627558 INFO pred_bat: Current SOC 10% is greater than Target SOC 0.0. Grid Charge disabled.                                                                                                                                                                                                                                        │
2023-12-04 23:30:02.630250 INFO pred_bat: Inverter 0 Charge settings: timed charged is disabled, power 3.599 kw                                                                                                                                                                                                                                        │
2023-12-04 23:30:02.634203 INFO pred_bat: Inverter 0 scheduled discharge enable is False
2023-12-04 23:30:02.637201 INFO pred_bat: Inverter 0 discharge windows currently [{'start': 1439, 'end': 1439, 'average': 0}, {'start': 2879, 'end': 2879, 'average': 0}]                                                                                                                                                                              │
2023-12-04 23:30:02.639161 INFO pred_bat: Found 1 inverters totals: min reserve 1.06 current reserve 1.06 soc_max 7.1 soc 0.71 charge rate 3.599 kw discharge rate 3.599 kw battery_rate_min 100.0 w ac limit 4.6 export limit 4.6 kw loss charge 3 % loss discharge 3 % inverter loss 4 %                                                             │
2023-12-04 23:30:02.678773 INFO pred_bat: Base charge    window [  ]                                                                                                                                                                                                                                                                                   │
2023-12-04 23:30:02.680225 INFO pred_bat: Base discharge window [ 04-12 23:59:00 - 04-12 23:59:00 @ 0.0p 100.0%, 05-12 23:59:00 - 05-12 23:59:00 @ 0.0p 100.0% ]                                                                                                                                                                                       │
2023-12-04 23:30:02.681153 INFO pred_bat: Plan was last updated on 2023-12-04 23:20:00.015020+00:00 and is now 10.0 minutes old                                                                                                                                                                                                                        │
2023-12-04 23:30:02.682203 INFO pred_bat: Best charge    window [ 04-12 23:30:00 - 05-12 05:30:00 @ 7.5p 7.1%, 05-12 23:30:00 - 06-12 05:30:00 @ 7.5p 7.1% ]                                                                                                                                                                                           │
2023-12-04 23:30:02.683209 INFO pred_bat: Best discharge window [  ]                                                                                                                                                                                                                                                                                   │
2023-12-04 23:30:02.685060 INFO pred_bat: PV Forecast 4.7 kWh and 10% Forecast 1.9 kWh pv cloud factor 1.0                                                                                                                                                                                                                                             │
2023-12-04 23:30:02.707844 INFO pred_bat: predict base end_record 12-06 23:30:00 final soc 1.06 kwh metric 1173.52 p metric_keep 350.4 min_soc 1.06 @ 12-04 23:50:00 kwh load 51.65 pv 6.08                                                                                                                                                            │
2023-12-04 23:30:02.710330 INFO pred_bat:          [  23:30,   00:30,   01:30,   02:30,   03:30,   04:30,   05:30,   06:30,   07:30,   08:30,   09:30,   10:30,   11:30,   12:30,   13:30,   14:30,   15:30,   16:30,   17:30,   18:30,   19:30,   20:30,   21:30,   22:30,   23:30,   00:30,   01:30,   02:30,   03:30,   04:30,   05:30,   06:30,   0│
2023-12-04 23:30:02.712608 INFO pred_bat:     SOC: [   0.71,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.08,    1.08,    1.07,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    1.06,    │
2023-12-04 23:30:02.714784 INFO pred_bat:   STATE: [  g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g<be-,   g│
2023-12-04 23:30:02.717029 INFO pred_bat:    LOAD: [  14.28,   15.24,   16.24,   16.91,   17.42,   17.82,   18.18,   18.53,   18.99,   19.54,   20.15,   20.72,   21.49,   22.27,   22.98,   23.66,   24.36,   25.24,   25.94,   27.14,   27.79,   29.24,   30.23,   30.89,   31.54,    32.8,   33.42,   35.23,   38.03,   40.35,   40.72,   41.07,   4│
2023-12-04 23:30:02.719238 INFO pred_bat:      PV: [    1.4,     1.4,     1.4,     1.4,     1.4,     1.4,     1.4,     1.4,     1.4,    1.43,    1.59,    1.92,    2.38,    2.86,    3.23,    3.42,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    3.45,    │
2023-12-04 23:30:02.721491 INFO pred_bat:  IMPORT: [  12.69,   13.69,   14.69,   15.36,   15.87,   16.27,   16.63,   16.98,   17.43,   17.96,   18.42,   18.69,   19.02,   19.34,   19.68,   20.19,   20.85,   21.73,   22.43,   23.63,   24.28,   25.73,   26.72,   27.38,   28.03,    29.3,   29.91,   31.72,   34.52,   36.84,   37.22,   37.56,   3│
2023-12-04 23:30:02.723720 INFO pred_bat:  EXPORT: [    0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,     0.0,    │
2023-12-04 23:30:02.725990 INFO pred_bat:    CAR0: [   62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    62.0,    │
2023-12-04 23:30:02.954294 INFO pred_bat: Battery has 0.33 hours left - now at 0.71
[... cut ...]                                                            │
│2023-12-04 23:50:04.508630 INFO pred_bat: Inverter 0 Charge settings: timed charged is disabled, power 3.599 kw                                                                                                                                                                                                                                        │
│2023-12-04 23:50:04.511330 INFO pred_bat: Inverter 0 scheduled discharge enable is False
```

springfall2008, 2023-12-06:
I've made a fix on master, would you be able to test it please?

marcoaddario, 2023-12-06:
Thank you very much for the fix Trefor.
I have started testing it and the read-only mode is now preventing the app to change the inverter mode.
Luckily today is a very sunny day and I should be able to test it on different scenarios.
I'll check tonight with the charge slot if it works as it should, but I have noticed something odd (I don't know if si related to this fix or not. If not I think it would be better to create a new issue to track it properly)

The plan shown the `FreezeDis` status (due to sun + full batteries) but the inverter started to discharge them.:
[image]

and this is how predbat set the inverter:
[image]
[predbat-20231206_1100-1105.log](https://github.com/springfall2008/batpred/files/13582009/predbat-20231206_1100-1105.log)

marcoaddario, 2023-12-07:
The night charge session went well. The inverter charger during the expected slot and then switch back to the original state.

springfall2008, 2023-12-07:
I'm not sure what's happening here, I don't think force discharge is really supported with Solis so probably you should turn off calculate discharge by using the Predbat mode of Control Charge

closed 2023-12-08 by marcoaddario