springfall2008/batpred #430: Predbat discharging at the start of the day instead of the end

iainfogg, 2023-12-06:
I'm on 7.14.2. Export mode off, mode control charge and discharge.

Intelligent just finished charging my car at 0700, and Predbat has straight into discharging the house battery, as per the screenshot.

[image]

The normal routine is it exports at the end of the day prior to the start of the next cheap rate.

Doubt it's relevant, but I've currently got my car SOC commented out in the YAML for because the night before it became unavailable (due to issues with the car integration) and Predbat stopped charging the home battery, so I commented it out last night.

I turned mode to Monitor to stop it discharging, and just tried routing it back on, and now I've got this:

[image]

Not as extreme as the first, but still not what I'd expect.

springfall2008, 2023-12-06:
I'm not sure monitor worked, the best line should really go away in this case...

springfall2008, 2023-12-06:
Can you send me a log, I wonder if the new 'tweak' feature is adding back in the discharge early on. Are the export rates fixed?

iainfogg, 2023-12-06:
PS you asked about export - I'm on 15p any time of day, so nothing special about the morning. Every night normally, it dumps remaining battery to finish discharging at 23:30.

springfall2008, 2023-12-07:
So it looks like you had some left-over settings when you turned predbat off, did you perhaps use read-only mode and hence leave the inverter in a bad state?

iainfogg, 2023-12-07:
No, I only changed the mode. I didn't touch read only. I've checked in the history to confirm that too.

Last night was also a bit weird. It mostly charged, but twice stopped to do small discharged, it didn't charge to 100%, and after charging finished it kept doing a small block of freeze discharge which kept moving forward as time went on.

I'll post some screenshots from my phone here, and I'll grab some logs and will post separately.

[image]

Above, at just after 05:30 when my Intelligent slot finished, it dropped into a small freeze discharge slot. Also you can see the two times it discharged during the night, and that it never got to 100%. Normally, every night finishes at 100%, and there are no discharge slots during the night. I have 'set discharge during charge' set to off.

[image]

And in this one above, an hour later at 06:30, you can see the freeze discharge block has simply shifted to the right.

iainfogg, 2023-12-07:
And here are all my switch settings:

[image]

iainfogg, 2023-12-07:
Log from the run at 23:30, where it planned to charge all the way until 05:30:

```txt
2023-12-06 23:30:00.005111 INFO pred_bat: --------------- PredBat - update at 2023-12-06 23:30:00.004133+00:00 with clock skew 0 minutes, minutes now 1410
2023-12-06 23:30:00.006165 INFO pred_bat: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 50.0 minutes
2023-12-06 23:30:00.007156 INFO pred_bat: Predbat version v7.14.3 currently running, latest version is v7.13.23
2023-12-06 23:30:00.017187 INFO pred_bat: Inverter type GE forecast_hours 48 max_windows 48 num_cars 1 debug enable is False calculate_plan_every 10
2023-12-06 23:30:00.018775 INFO pred_bat: Cars 1 charging from battery False planned [False], charging_now [False] smart [False], plan_time ['07:00:00'], battery size [100.0], limit [100.0], rate [7.4]
2023-12-06 23:30:00.020136 INFO pred_bat: Predbat mode is set to Control charge & discharge
2023-12-06 23:30:00.297488 INFO pred_bat: Found 11521 load_today datapoints going back 8 days
2023-12-06 23:30:00.672162 INFO pred_bat: Current data so far today: load 29.1 kWh import 37.9 kWh export 13.1 kWh pv 2.4 kWh
2023-12-06 23:30:00.673347 INFO pred_bat: Fetching futurerate data from None
2023-12-06 23:30:00.687031 INFO pred_bat: Car 0 using Octopus Intelligent, charging limit 80.0, ready time 07:30:00 - battery size 100.0
2023-12-06 23:30:00.688403 INFO pred_bat: Current Car SOC kwh: [0.0]
2023-12-06 23:30:00.698336 INFO pred_bat: Joined Octopus saving session: 2023-11-16T17:30:00+00:00 - 2023-11-16T18:30:00+00:00 at rate 225.0 state False
2023-12-06 23:30:00.699850 INFO pred_bat: Joined Octopus saving session: 2023-11-29T17:00:00+00:00 - 2023-11-29T18:30:00+00:00 at rate 400.0 state False
2023-12-06 23:30:00.701314 INFO pred_bat: Joined Octopus saving session: 2023-12-01T16:30:00+00:00 - 2023-12-01T18:00:00+00:00 at rate 400.0 state False
2023-12-06 23:30:00.702836 INFO pred_bat: Standing charge is set to 0.0 p
2023-12-06 23:30:00.709927 INFO pred_bat: Octopus Intelligent slot at 12-06 00:30:00-12-06 01:00:00 assumed price 7.5
2023-12-06 23:30:00.711657 INFO pred_bat: Octopus Intelligent slot at 12-06 00:00:00-12-06 00:30:00 assumed price 7.5
2023-12-06 23:30:00.713347 INFO pred_bat: Octopus Intelligent slot at 12-06 01:00:00-12-06 01:30:00 assumed price 7.5
2023-12-06 23:30:00.715128 INFO pred_bat: Octopus Intelligent slot at 12-06 02:30:00-12-06 03:00:00 assumed price 7.5
2023-12-06 23:30:00.716841 INFO pred_bat: Octopus Intelligent slot at 12-06 02:00:00-12-06 02:30:00 assumed price 7.5
2023-12-06 23:30:00.718513 INFO pred_bat: Octopus Intelligent slot at 12-06 01:30:00-12-06 02:00:00 assumed price 7.5
2023-12-06 23:30:00.720368 INFO pred_bat: Octopus Intelligent slot at 12-06 03:30:00-12-06 04:00:00 assumed price 7.5
2023-12-06 23:30:00.722186 INFO pred_bat: Octopus Intelligent slot at 12-06 03:00:00-12-06 03:30:00 assumed price 7.5
2023-12-06 23:30:00.723927 INFO pred_bat: Octopus Intelligent slot at 12-06 04:00:00-12-06 04:30:00 assumed price 7.5
2023-12-06 23:30:00.725787 INFO pred_bat: Octopus Intelligent slot at 12-06 05:30:00-12-06 06:00:00 assumed price 7.5
2023-12-06 23:30:00.727496 INFO pred_bat: Octopus Intelligent slot at 12-06 05:00:00-12-06 05:30:00 assumed price 7.5
2023-12-06 23:30:00.729212 INFO pred_bat: Octopus Intelligent slot at 12-06 04:30:00-12-06 05:00:00 assumed price 7.5
2023-12-06 23:30:00.730922 INFO pred_bat: Octopus Intelligent slot at 12-06 06:00:00-12-06 06:30:00 assumed price 7.5
2023-12-06 23:30:00.732613 INFO pred_bat: Octopus Intelligent slot at 12-06 07:00:00-12-06 07:30:00 assumed price 7.5
2023-12-06 23:30:01.115250 INFO pred_bat: Rate min forward looking: now 7.499899999999999 at end of forecast 7.499899999999999
2023-12-06 23:30:01.117024 INFO pred_bat: Import rates min 7.5 max 29.9 average 24.3
2023-12-06 23:30:01.124455 INFO pred_bat: Export rates min 15.0 max 15.0 average 15.0
2023-12-06 23:30:01.126188 INFO pred_bat: Rate thresholds (for charge/discharge) are import 29.4p (0.0) export 15.0p (0.0)
2023-12-06 23:30:01.132897 INFO pred_bat: High export rate found rates in range 15.0 to 15.0
2023-12-06 23:30:01.136369 INFO pred_bat: Low Import rate found rates in range 7.5 to 7.5
2023-12-06 23:30:01.138007 INFO pred_bat: Car 0 is using Octopus intelligent schedule
2023-12-06 23:30:01.175639 INFO pred_bat: Todays energy import 37.9 kwh export 13.1 kwh cost 224.92 p import 421.42 p export -196.5 p
2023-12-06 23:30:01.179721 INFO pred_bat: PV Data for pv_forecast_today total 1.49 kWh
2023-12-06 23:30:01.183223 INFO pred_bat: PV Data for pv_forecast_tomorrow total 0.42 kWh
2023-12-06 23:30:01.186853 INFO pred_bat: PV Data for pv_forecast_d3 total 0.81 kWh
2023-12-06 23:30:01.190480 INFO pred_bat: PV Data for pv_forecast_d4 total 1.31 kWh
2023-12-06 23:30:01.209008 INFO pred_bat: Today's load divergence 130.58 % in-day adjustment 129.05 % damping 0.95x, Predicted so far 5.58 kWh with 18.06 kWh car excluded and 1.83 kWh import ignored and 0.0 forecast extra, Actual so far 8.04 KWh with 16.47 kWh car excluded and 4.72 kWh import ignored
2023-12-06 23:30:01.238844 INFO pred_bat: Historical data totals for days [7] are [14.3] - min 14.3
2023-12-06 23:30:01.240579 INFO pred_bat: Historical days now [7] weight [1]
2023-12-06 23:30:01.241828 INFO pred_bat: Inverter 0 using Rest API http://192.168.68.58:6345
[... cut ...]
2023-12-06 23:30:18.686157 INFO pred_bat: Inverter 0 Current Target SOC is 100.0 already at target
2023-12-06 23:30:18.692536 INFO pred_bat: Adjust reserve to target charge % (set_reserve_enable is true)
2023-12-06 23:30:18.695292 INFO pred_bat: Inverter 0 Current Reserve is 4.0 % and new target is 99 %
2023-12-06 23:30:31.072126 INFO pred_bat: Set inverter 0 reserve 99 via REST successful on retry 0
2023-12-06 23:30:31.119775 INFO pred_bat: Will not recompute the plan, it is 5.0 minutes old and max age is 10 minutes
```

iainfogg, 2023-12-07:
A record of predbat.status from when the IO slot started at 23:30 last night until 06:30-ish when I set it to Monitor mode and made sure the GivTCP status was correct (it was). Here the status message goes with the time immediately below it:

```txt
December 7, 2023
changed to Idle
6:31:26 AM - 12 hours ago - Supervisor
changed to Freeze discharging
6:01:06 AM - 12 hours ago - Supervisor
changed to Inverter 0 charge rate changed to 0
6:01:06 AM - 12 hours ago - Supervisor
changed to Inverter 0 charge rate changed to 3600
6:00:21 AM - 12 hours ago - Supervisor
changed to Freeze discharging
5:31:01 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 4.0
5:31:01 AM - 13 hours ago - Supervisor
changed to Inverter 0 Set discharge mode to Eco
5:30:49 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 55
5:30:15 AM - 13 hours ago - Supervisor
changed to Discharging
5:22:22 AM - 13 hours ago - Supervisor
changed to Inverter 0 set soc to 4.0
5:22:22 AM - 13 hours ago - Supervisor
changed to Inverter 0 charge rate changed to 0
5:22:01 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 54
5:21:40 AM - 13 hours ago - Supervisor
changed to Inverter 0 set discharge slot to 04:45:00 - 07:01:00
5:21:32 AM - 13 hours ago - Supervisor
changed to Inverter 0 Set discharge mode to Timed Export
5:21:32 AM - 13 hours ago - Supervisor
changed to Inverter 0 discharge rate changed to 3600
5:21:08 AM - 13 hours ago - Supervisor
changed to Inverter 0 Turned off scheduled charge
5:20:56 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 93
5:20:29 AM - 13 hours ago - Supervisor
changed to Inverter 0 set soc to 93
5:20:21 AM - 13 hours ago - Supervisor
changed to Charging
5:15:26 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 95
5:15:26 AM - 13 hours ago - Supervisor
changed to Inverter 0 set soc to 95
5:15:17 AM - 13 hours ago - Supervisor
changed to Charging
5:05:30 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 96
5:05:30 AM - 13 hours ago - Supervisor
changed to Inverter 0 set soc to 96
5:05:18 AM - 13 hours ago - Supervisor
changed to Charging
5:01:37 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 97
5:01:37 AM - 13 hours ago - Supervisor
changed to Inverter 0 set soc to 97
5:01:29 AM - 13 hours ago - Supervisor
changed to Inverter 0 Set discharge mode to Eco
5:01:16 AM - 13 hours ago - Supervisor
changed to Inverter 0 Turned on charge enable
5:01:07 AM - 13 hours ago - Supervisor
changed to Inverter 0 discharge rate changed to 0
5:00:45 AM - 13 hours ago - Supervisor
changed to Inverter 0 charge rate changed to 3600
5:00:38 AM - 13 hours ago - Supervisor
changed to Discharging
4:55:15 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 62
4:55:15 AM - 13 hours ago - Supervisor
changed to Discharging
4:50:35 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 63
4:50:15 AM - 13 hours ago - Supervisor
changed to Discharging
4:45:49 AM - 13 hours ago - Supervisor
changed to Inverter 0 charge rate changed to 0
4:45:49 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 62
4:45:36 AM - 13 hours ago - Supervisor
changed to Inverter 0 Set discharge mode to Timed Export
4:45:28 AM - 13 hours ago - Supervisor
changed to Inverter 0 discharge rate changed to 3600
4:45:19 AM - 13 hours ago - Supervisor
changed to Hold charging
4:41:25 AM - 13 hours ago - Supervisor
changed to Inverter 0 set reserve to 91
4:41:25 AM - 13 hours ago - Supervisor
changed to Inverter 0 set soc to 91
4:41:07 AM - 13 hours ago - Supervisor
changed to Inverter 0 set discharge slot to 04:45:00 - 06:01:00
4:40:59 AM - 13 hours ago - Supervisor
changed to Inverter 0 Turned off scheduled charge
4:40:45 AM - 14 hours ago - Supervisor
changed to Charging
1:01:35 AM - 17 hours ago - Supervisor
changed to Inverter 0 Charge window change to: 00:00:00 - 05:30:00
1:01:35 AM - 17 hours ago - Supervisor
changed to Inverter 0 set reserve to 99
1:01:00 AM - 17 hours ago - Supervisor
changed to Inverter 0 Set discharge mode to Eco
1:00:39 AM - 17 hours ago - Supervisor
changed to Inverter 0 discharge rate changed to 0
1:00:26 AM - 17 hours ago - Supervisor
changed to Inverter 0 charge rate changed to 3600
1:00:19 AM - 17 hours ago - Supervisor
changed to Discharging
12:51:54 AM - 17 hours ago - Supervisor
changed to Inverter 0 charge rate changed to 0
12:51:54 AM - 17 hours ago - Supervisor
changed to Inverter 0 set reserve to 25
12:51:42 AM - 17 hours ago - Supervisor
changed to Inverter 0 set discharge slot to 00:00:00 - 01:01:00
12:51:20 AM - 17 hours ago - Supervisor
changed to Inverter 0 Set discharge mode to Timed Export
12:51:20 AM - 17 hours ago - Supervisor
changed to Inverter 0 discharge rate changed to 3600
12:50:58 AM - 17 hours ago - Supervisor
changed to Inverter 0 Charge window change to: 01:00:00 - 05:30:00
12:50:44 AM - 17 hours ago - Supervisor
December 6, 2023
changed to Charging
11:30:31 PM - 19 hours ago - Supervisor
changed to Inverter 0 set reserve to 99
11:30:31 PM - 19 hours ago - Supervisor
changed to Inverter 0 discharge rate changed to 0
11:30:18 PM - 19 hours ago - Supervisor
```

iainfogg, 2023-12-07:
Logs from 04:45 and 04:50 where some discharging happened (it happened in other places too, but these are examples):

```txt
2023-12-07 04:45:00.004503 INFO pred_bat: --------------- PredBat - update at 2023-12-07 04:45:00.003552+00:00 with clock skew 0 minutes, minutes now 285
2023-12-07 04:45:00.647118 INFO pred_bat: Predbat version v7.14.3 currently running, latest version is v7.13.23
2023-12-07 04:45:00.656430 INFO pred_bat: Inverter type GE forecast_hours 48 max_windows 48 num_cars 1 debug enable is False calculate_plan_every 10
2023-12-07 04:45:00.657973 INFO pred_bat: Cars 1 charging from battery False planned [False], charging_now [False] smart [False], plan_time ['07:00:00'], battery size [100.0], limit [100.0], rate [7.4]
2023-12-07 04:45:00.659276 INFO pred_bat: Predbat mode is set to Control charge & discharge
2023-12-07 04:45:00.935170 INFO pred_bat: Found 11521 load_today datapoints going back 8 days
2023-12-07 04:45:01.202046 INFO pred_bat: Current data so far today: load 4.42 kWh import 17.7 kWh export 0.5 kWh pv 0.0 kWh
2023-12-07 04:45:01.203398 INFO pred_bat: Fetching futurerate data from None
2023-12-07 04:45:01.215805 INFO pred_bat: Car 0 using Octopus Intelligent, charging limit 80.0, ready time 07:30:00 - battery size 100.0
2023-12-07 04:45:01.217218 INFO pred_bat: Current Car SOC kwh: [0.0]
2023-12-07 04:45:01.227405 INFO pred_bat: Joined Octopus saving session: 2023-11-16T17:30:00+00:00 - 2023-11-16T18:30:00+00:00 at rate 225.0 state False
2023-12-07 04:45:01.228921 INFO pred_bat: Joined Octopus saving session: 2023-11-29T17:00:00+00:00 - 2023-11-29T18:30:00+00:00 at rate 400.0 state False
2023-12-07 04:45:01.230376 INFO pred_bat: Joined Octopus saving session: 2023-12-01T16:30:00+00:00 - 2023-12-01T18:00:00+00:00 at rate 400.0 state False
2023-12-07 04:45:01.231898 INFO pred_bat: Standing charge is set to 0.0 p
2023-12-07 04:45:01.636786 INFO pred_bat: Rate min forward looking: now 7.499899999999999 at end of forecast 7.499899999999999
2023-12-07 04:45:01.638554 INFO pred_bat: Import rates min 7.5 max 29.9 average 24.3
2023-12-07 04:45:01.646179 INFO pred_bat: Export rates min 15.0 max 15.0 average 15.0
2023-12-07 04:45:01.647905 INFO pred_bat: Rate thresholds (for charge/discharge) are import 29.4p (0.0) export 15.0p (0.0)
2023-12-07 04:45:01.653224 INFO pred_bat: High export rate found rates in range 15.0 to 15.0
2023-12-07 04:45:01.656434 INFO pred_bat: Low Import rate found rates in range 7.5 to 7.5
2023-12-07 04:45:01.658475 INFO pred_bat: Car 0 is using Octopus intelligent schedule
2023-12-07 04:45:01.702724 INFO pred_bat: Todays energy import 17.7 kwh export 0.5 kwh cost 125.25 p import 132.75 p export -7.5 p
2023-12-07 04:45:01.706499 INFO pred_bat: PV Data for pv_forecast_today total 0.42 kWh
2023-12-07 04:45:01.710853 INFO pred_bat: PV Data for pv_forecast_tomorrow total 0.8 kWh
2023-12-07 04:45:01.714469 INFO pred_bat: PV Data for pv_forecast_d3 total 0.78 kWh
2023-12-07 04:45:01.718030 INFO pred_bat: PV Data for pv_forecast_d4 total 1.12 kWh
2023-12-07 04:45:01.735112 INFO pred_bat: Today's load divergence 100.0 % in-day adjustment 100.0 % damping 0.95x, Predicted so far 0.0 kWh with 0.08 kWh car excluded and 7.31 kWh import ignored and 0.0 forecast extra, Actual so far 0.1 KWh with 0.07 kWh car excluded and 4.25 kWh import ignored
2023-12-07 04:45:01.775269 INFO pred_bat: Historical data totals for days [7] are [12.34] - min 12.34
2023-12-07 04:45:01.777035 INFO pred_bat: Historical days now [7] weight [1]
2023-12-07 04:45:01.778232 INFO pred_bat: Inverter 0 using Rest API http://192.168.68.58:6345
2023-12-07 04:45:03.007224 INFO pred_bat: Invertor time 2023-12-07 04:44:30+00:00 AppDaemon time 2023-12-07 04:45:00.003552+00:00 difference -0.52 minutes
2023-12-07 04:45:03.008764 INFO pred_bat: Reserve min: 4.0% Battery_min:4.0%
2023-12-07 04:45:03.010378 INFO pred_bat: New Inverter 0 with soc_max 19.05 kWh nominal_capacity 19.05 kWh battery rate raw 3600.0 w charge rate 3.6 kw discharge rate 3.6 kw battery_rate_min 200.0 w ac limit 5.0 kw export limit 5.0 kw reserve 4.0 % current_reserve 91.0 %
2023-12-07 04:45:04.245929 INFO pred_bat: Inverter 0 SOC: 17.14 kw 90 % Current charge rate 3600.0 w Current discharge rate 200.0 w Current power -291.0 w Current voltage52.0
2023-12-07 04:45:04.248528 INFO pred_bat: Inverter 0 scheduled charge enable is False
2023-12-07 04:45:04.251869 INFO pred_bat: Inverter 0 charge windows currently []
2023-12-07 04:45:04.254726 INFO pred_bat: Inverter 0 Charge settings: timed charged is disabled, power 3.6 kw
2023-12-07 04:45:04.257821 INFO pred_bat: Inverter 0 scheduled discharge enable is False
2023-12-07 04:45:04.260701 INFO pred_bat: Inverter 0 discharge windows currently [{'start': 285, 'end': 361, 'average': 0}, {'start': 1725, 'end': 1801, 'average': 0}]
2023-12-07 04:45:04.263125 INFO pred_bat: Found 1 inverters totals: min reserve 0.76 current reserve 17.33 soc_max 19.05 soc 17.14 charge rate 3.6 kw discharge rate 0.2 kw battery_rate_min 200.0 w ac limit 5.0 export limit 5.0 kw loss charge 5 % loss discharge 5 % inverter loss 0 %
2023-12-07 04:45:04.595002 INFO pred_bat: Base charge    window [  ]
2023-12-07 04:45:04.597095 INFO pred_bat: Base discharge window [ 07-12 04:45:00 - 07-12 06:01:00 @ 0.0p 100.0%, 08-12 04:45:00 - 08-12 06:01:00 @ 0.0p 100.0% ]
2023-12-07 04:45:04.598058 INFO pred_bat: Plan was last updated on 2023-12-07 04:40:00.004400+00:00 and is now 5.0 minutes old
[... cut ...]
2023-12-07 04:50:35.041360 INFO pred_bat: Discharging now - current SOC 16.95 and target 12.0015
2023-12-07 04:50:35.042796 INFO pred_bat: Inverter 0 Adjust force discharge to True, change times from 04:45:00 - 06:01:00 to 04:45:00 - 06:01:00
2023-12-07 04:50:35.044172 INFO pred_bat: Inverter 0 Current reserve is 63.0 already at target
2023-12-07 04:50:35.045660 INFO pred_bat: Not setting charging SOC as we are not within the window (now 12-07 04:50:00 target set_soc_minutes 30 charge start time 12-09 00:00:00)
2023-12-07 04:50:35.060040 INFO pred_bat: Completed run status Discharging
```

iainfogg, 2023-12-08:
I had the same issue last night. Here's the plan at the start of the night, showing it charging through the night then going idle after that:

[image]

By the early hours, the plan had changed to dumping 60% of the battery immediately after the cheap rate finished:

[image]

[image]

I had to turn the mode to Control Charge to avoid that discharge.

Around 07:30 I tried changing back to Control Charge & Discharge, and it went straight back to discharging immediately. I'll grab the logs for that plan and will post below.

There's either an issue with some residual settings, or a problem with the plan calculation in the code, as this is the opposite of what Predbat normally does, and I can't make this release work for me. I'm on 7.14.5, by the way. @springfall2008

iainfogg, 2023-12-08:
Also, the Calculate Discharge on Charge switch is still visible in non expert mode, but it sounds like it's being ignored and treated as true in the calculations from what you said @springfall2008

springfall2008, 2023-12-09:
> Also, the Calculate Discharge on Charge switch is still visible in non expert mode, but it sounds like it's being ignored and treated as true in the calculations from what you said @springfall2008

Is it not marked as 'disabled"?

springfall2008, 2023-12-09:
I've made a new release which turns off the tweak calculation by default and should hopefully fix the issue you reported: https://github.com/springfall2008/batpred/releases/tag/v7.14.6

iainfogg, 2023-12-09:
> > Also, the Calculate Discharge on Charge switch is still visible in non expert mode, but it sounds like it's being ignored and treated as true in the calculations from what you said @springfall2008
> 
> Is it not marked as 'disabled"?

No, it's enabled and set to false.

iainfogg, 2023-12-09:
> I've made a new release which turns off the tweak calculation by default and should hopefully fix the issue you reported: https://github.com/springfall2008/batpred/releases/tag/v7.14.6

Thanks, I've got the new release now. It's been fine, although I ended up setting it to expert mode, as without it, it was discharging until midnight, where I prefer not to discharge during 23:30-05:30.

Now I've got it in expert mode, it's discharging at the right end of the day, but it's now stopping discharging at 23:00 rather than 23:30, which it's not done previously.

It's way closer now though, so thanks for changing it @springfall2008

springfall2008, 2023-12-10:
> Thanks, I've got the new release now. It's been fine, although I ended up setting it to expert mode, as without it, it was discharging until midnight, where I prefer not to discharge during [...]

It's probably as the slots are 1 hour normally so the 11pm one would overlap with the charge

iainfogg, 2023-12-10:
Is that planning in one hour blocks a new thing, as it used to handle charging to 2330 then switching to charging fine before?

springfall2008, 2023-12-10:
> Is that planning in one hour blocks a new thing, as it used to handle charging to 2330 then switching to charging fine before?

New(ish) it's been there for a little while, I changed it so that if the rates are the same (within 0.5p of each other) then they combine into 1 hour slots (bigger with combine enabled). It's a way to cope with bigger possible windows of charge/discharge without running too slowly, in theory I could make (yet another) option to split into 30 minutes instead...

iainfogg, 2023-12-11:
I'll leave it to you as to whether that's worth it or not, but I think a number of the overnight rates sit on a half hour boundary. But keeping Predbat simple is something I'm a fan of too 🙂

closed 2023-12-13 by springfall2008