springfall2008/batpred #442: Predbat still wants to charge at too high a price on Agile

fboundy, 2023-12-08:
Predbat still seems to want to charge at too high a price on Agile. See the plan below where it is starting charging at 7pm at 20p when all the charging needed could be done from midnight onwards at <14p.

[image]

I calculate the optimum plan to be this which I think is about £1 cheaper by 7pm tomorrow

[image]

I think it's because the Low Threshold is set to 25.8p which seems far too high. But then I don't really understand the need for Thresholds....

Redding43, 2023-12-08:
Did the view change as you approached as the prices that are 14 have a ‘?’ Meaning that at that point this was the potential price and not guaranteed

fboundy, 2023-12-08:
Can’t say for certain but I know what algorithm Trefor uses and I know they aren’t going to change very much

springfall2008, 2023-12-08:
A lot of this is down to losses, if you set charge, discharge, inverter losses and metric battery cycle to 0 then you will just get the cheaper slots. Running at 100% and importing means no losses

fboundy, 2023-12-08:
Not sure I get that. The same losses are involved to charge the battery at 20p as at 14p so why charge it earlier and then hold the charge at 100%
On 8 Dec 2023 at 22:19 +0000, springfall2008/batpred ***@***.***>, wrote:
>
> metric

springfall2008, 2023-12-08:
In the meantime you would have to supply the home from the battery so those losses are included.

fboundy, 2023-12-08:
If the battery is empty you are supplying the house from the grid so where are the losses?

springfall2008, 2023-12-09:
> If the battery is empty you are supplying the house from the grid so where are the losses?
> […](#)

Yes this is certainly true, but in this case it doesn't apply as plan shown here starts at 82% battery.

One thing Predbat could do better is to disable the battery use (charge freeze) but I don't know if a) it's enabled in this setup and b) many inverters don't actually hold at zero they leak around 200w.

springfall2008, 2023-12-09:
What version of Predbat?

gcoan, 2023-12-09:
> One thing Predbat could do better is to disable the battery use (charge freeze) but I don't know if in this setup it leaks or if it really holds at 0 discharge.

I wanted to also highlight #435 where predbat doesn't set the battery reserve so "hold charging" plans/statii actually result in the battery discharging.  This then caused oscillation of charging and discharging, swapping over every 5 minutes in a 30 minute slot

fboundy, 2023-12-09:
[image]

Much the same today. Predbat wants to charge much earlier. You can see what happens in the Calibration chart bottom right. When it gets past the peak it resets the thresholds and all makes sense.

springfall2008, 2023-12-09:
> Much the same today. Predbat wants to charge much earlier. You can see what happens in the Calibration chart bottom right. When it gets past the peak it resets the thresholds and all makes sense.

I did just make a new release which may or may not help, it would be good if you can try it: https://github.com/springfall2008/batpred/releases/tag/v7.14.6

I wouldn't read much into the threshold, if it's on automatic mode then the threshold value is representing the maximum charge price Predbat has used rather than impacting the end result.

fboundy, 2023-12-10:
Much better so far with 7.14.6

h12 and h8 are much more stable whereas previously they were flip-flopping. 

Now pretty much identical to my algorithm - I would expect mine to look marginally better on the calibration plot as it is controlling the inverter but the difference is minimal. I'm also just using the Solcast mid with none of the 10% which accounts for the Base difference

[image]

springfall2008, 2023-12-10:
> Much better so far with 7.14.6
>
> h12 and h8 are much more stable whereas previously they were flip-flopping.
>
> Now pretty much identical to my algorithm - I would expect mine to look marginally better [...]

Thanks for testing, that's good to know :)

The main thing I'm trying to achieve with Predbat is that the tuning is done to the inverter model rather than with assumed knowledge. It means things like clipping and export limits will be accounted for rather than just unit rates, of course it makes things more complication as a result also.

closed 2023-12-13 by springfall2008