springfall2008/batpred #449: Predbat's forward plan doesn't charge & discharge in optimal slots

gcoan, 2023-12-11:
I'm raising this as a separate issue, splitting it off from the latter points on #435 so as to keep the issues focussed.

It may be the same issue as reported on #442 by @fboundy 

> I think some of it comes down to losses, discharging effectively incurs a round trip loss (as you have to charge it back later as well). If you turn on plan debug you can see the rates after losses.
> 
> Tweak plan was removed although maybe I should keep it as an advanced feature for now to see if it helps or not...

Here's the plan from last night at 23:30
[image]

and logfile
[appdaemon.log.3.txt](https://github.com/springfall2008/batpred/files/13634177/appdaemon.log.3.txt)

I appreciate that discharging will cause a double loss as it has to be recharged later on,  but if I look in detail at the slots in the early morning:

01:00 (rate was 13.89p) and 02:00 (rate 14.11p) were both planned as discharge slots
but 23:00 (rate 14.33p) was planned to be a hold  
and 00:30 (rate 14.57p) was planned to be charge/retain soc at 100%

Predbat should be choosing the cheaper rates to charge in and more expensive to discharge, so 23:00 should have been a discharge as should 00:30

I don't know whether the plan then changed later on as I went to bed, but I have noticed that there's something about the way that predbat looks at future rates and what it adds or doesn't add to the weightings in a 'forward window' that causes the future plan to be 'not quite right'.  I have seen it correcting itself when it gets to execution time, but its still confusing to look at the plan. 

Here's another very clear example, the plan for today and to the evening agile peak:
[image]

The plan is fine up to 14:30, and then shows the soc being released (predbat presumably in idle/Eco mode) from 15:00 when the rates are 18.74, 22.49, and the battery being fully discharged by 17:00 when the rates are 39.45p, thus leaving import to occur in the 17:30, 18:00 and 18:30 when the rates are 39. 38 & 36p - all higher than at 15:00 and 15:30 when predbat plans to start releasing the battery soc.

Current logfile:
[appdaemon (1).log](https://github.com/springfall2008/batpred/files/13634272/appdaemon.1.log)

I'm fairly sure this behaviour is something introduced into predbat in the last month or so as I don't recall seeing this beforehand.

springfall2008, 2023-12-11:
Good spot, the reason the 15:00-16:00 isn't used is it's missing from the low import rate list. I think this maybe a bug

springfall2008, 2023-12-11:
Unrelated but you may have not enabled you next_day_dates event for Octopus integration as we are getting this warning:

2023-12-11 10:30:01.648506 INFO pred_bat: WARN: No Octopus data in sensor sensor.octopus_energy_electricity_22j0452464_1012444237400_next_rate attribute 'all_rates'

Which it's looking for as the event.octopus_energy_electricity_22j0452464_1012444237400_next_day_rates one doesn't have data.

springfall2008, 2023-12-11:
So I had a look and it's actually just down to lack of windows, those higher rate slots fell off the list (too many similar priced windows). In fact it won't have much if any impact as the next 8 windows are always included in the plan so when you get closer to the time they will get selected.

However I've made a change on 'main' you can try where there is a toggle called 'calculate_fast_plan' and if you turn it off then Predbat will see all windows, it will take a little longer but hopefully we can confirm it fixes the issue.

I've also cleaned up the HTML plan a little as I noticed one straddle bug and that Freeze Charge wasn't showing the real target %.

gcoan, 2023-12-11:
I've installed main (5d6e377)

Here's the plan with calculate_fast_plan enabled (the default):
[image]

And after turning calculate_fast_plan off:
[image]

I'd say this revised plan looks worse.  The charging from 4% to 23% from 21:30-23:00 looks unnecessary as there isn't any following higher rate slots that the charge is preventing what would have otherwise (in the original plan) being grid import,  and there's plenty of lower rate charge slots overnight at 11, 12, 13p that will enable the battery to be charged to 100% overnight.

But having said that, conversely, in the slots tomorrow morning, the plan looks better.   The original had the soc being preserved at 100% through 8:30, 9:00 and 10:30 slots when the rates goes up, but in the newer plan the battery is being more actively managed, releasing charge and replenishing as rates go up and down.

Looking in to tomorrow's peak, (with calculate_fast_plan off) its a similar story of the soc being released ahead of the peak rates:
[image]

JonathanLew1s, 2023-12-11:
> Good spot, the reason the 15:00-16:00 isn't used is it's missing from the low import rate list. I think this maybe a bug

I've noticed for the past few builds it's possible for a window to be flagged as both high and low rate. Maybe linked.

[image]

[image]

[image]

springfall2008, 2023-12-11:
> event.octopus_energy_electricity_22j0452464_1012444237400_next_day_rates

What is actually in this event, can you show it's contents?

gcoan, 2023-12-11:
I don't know. I can see the event is firing
[image]

Here's the last event in the state history.  Looks reasonable to me, showing tomorrow's prices up to 22:30
[image]

With _current_day_rates showing todays prices up to midnight (we're past the 4pm publication time)

Looking at the error message again:
2023-12-11 16:00:01.386850 INFO pred_bat: WARN: No Octopus data in sensor sensor.octopus_energy_electricity_22j0452464_1012444237400_next_rate attribute 'all_rates'

its looking for attribute 'all_rates' but my event has the attribute 'rates'

gcoan, 2023-12-11:
Back to the main ([5d6e377](https://github.com/springfall2008/batpred/commit/5d6e3772b26d0aff2d6059ef518497a2620c1073)) release with calculate_fast_plan disabled...

I'd have thought it would have been worth charging at 20:30 when the rate is 16.89p, then discharging at 21:00 when the rate rises to 20.68p.
4p difference is enough to be better than the 10% charging/discharging losses
[image]

And then later on, the choice of slots to charge the battery to 100% look mostly logical, except for 05:00 (18.22p) being chosen whereas 04:00 (15.43p) would have been cheaper
[image]

fboundy, 2023-12-11:
Yes - that's what my algorithm is saying is optimal 

[image]

springfall2008, 2023-12-11:
I don’t think 5 is really chosen as it’s at 99% at the start of the 5am slot?

gcoan, 2023-12-11:
Agree r/e 5:00, its 99% soc so its not really charging in that slot

Predbat is flip flopping again.  At 18:40, the 20:30 slot wasn't planned for charging but when I looked 5 minutes ago at about 20:10, the 20:30 was now planned to be a charge slot

At my PC and at 20:17 when I came to take the screen shot, it isn't !
[image]

Waited until 20:20 and the 20:30 slot has changed back to being charging again
[image]

Here's the logfile
[appdaemon (1).log](https://github.com/springfall2008/batpred/files/13640622/appdaemon.1.log)

switch.predbat_calculate_fast_plan is off throughout

gcoan, 2023-12-11:
20:30, predbat started charging with the plan changing to the second plan as above, target 36%

20:35, predbat changed to reserve and soc 4%, status is still charging but its of course discharging now.  The mobile notification I received said "charging target 0%" !
 
[image]

[image]

Here's the latest logfile.  I'm switching predbat to read only and will charge it manually up to the end of the slot
[appdaemon (2).log](https://github.com/springfall2008/batpred/files/13640745/appdaemon.2.log)

gcoan, 2023-12-11:
21:00 finished the manual charge and the plan is now showing that charge discharging, but then another charge from 22:00 - which looks quite marginal if not negative as to whether its worth charging and then discharging
[image]

21:05 the plan's reverted to not charging from 22:00
[image]

And if I compare the plans, at equivalent places,  at 2:00 soc=4%, plan with charging=£6.76, without charging=£6.73
At 05:00 soc=99%, plan with charging=£8.89, without charging=£8.83

So the two plans are close but the without charging is slightly more optimal.

I'm wondering if this is something to do with the plan only being recalculated every 10 minutes but predbat running every 5 minutes, so every other cycle has a different outcome?  This would explain the flip-flopping of plan changes I've seen, both here and in #435 (discharging not hold charging)??

springfall2008, 2023-12-11:
Do you still have tweak plan enabled, as that might cause a flip flop behaviour ?

gcoan, 2023-12-11:
yes I do 
[image]

springfall2008, 2023-12-11:
Can you try with it off?

gcoan, 2023-12-11:
I had turned tweak second pass off.  Plan at 21:20 (after the switch was changed) has charging over 4 slots in two planned charges from 21:30 (to 57%) and 22:30 (to 71%)
[image]

At 21:25, what were two separate planned charges from 21:30 and 22:30 are now combined into a single one with a 47% soc limit
[image]

I'm still not sure why predbat wants to charge here, with import losses the rates are 19.03p and 18.72p, and the subsequent slots from 23:30-01:00 when the soc is released are all priced at lower than this.  At the moment I've left predbat in read only as I don't see the point in this charge

21:30 no change, charging still planned covering 4 slots in a single combined charge to 47%:
[image]

21:35 plan still to charge, but now only for 2 slots from 21:30-22:30 to 56%, then hold the charge
[image]

21:40 charging now for all 4 slots in 2 bursts:
[image]

21:45 back to charging for an hour then hold charging for an hour
[image]

Its a different flip-flop, but its still coming up with a different plan every 5 minutes with calculate tweak second pass off (and full second pass also off)

gcoan, 2023-12-11:
> So I had a look and it's actually just down to lack of windows, those higher rate slots fell off the list (too many similar priced windows). In fact it won't have much if any impact as the next 8 [...]

I also don't think that predbat is seeing all windows, or if it is, something else is occurring because the plan ahead of tomorrow's peak only has 1 charge slot and the peak is planned to be entered with the battery exhausted:
[image]

Turning calculate_fast_plan back on, the plan holds the charge to the peak, but probably similar financially as there's the value of the battery charge to be included at say 15:00
[image]

springfall2008, 2023-12-12:
I made an update on main to try to stop any changes in the 5 minute period, it might have been a bug related to clipping and also to try to remove those warnings.

Can you retry and save a log if you see anything odd?

gcoan, 2023-12-12:
> I made an update on main to try to stop any changes in the 5 minute period, it might have been a bug related to clipping and also to try to remove those warnings.
> 
> Can you retry and save a log if you see anything odd?

Will do.

Here's the plan before the update:
[image]

And after installing the update to main:
[image]

At the moment it doesn't look like predbat is going to be doing much until this afternoon.

Hope it plans some more charging before the evening peak ...

springfall2008, 2023-12-12:
> Logfile: [appdaemon (3).log](https://github.com/springfall2008/batpred/files/13647416/appdaemon.3.log)

I just tried to look but the log is truncated, can you get a full run?

springfall2008, 2023-12-12:
Urgh, I got the switch the wrong way around so fast mode was slow mode and visa versa. I push a fixed to main now.

gcoan, 2023-12-12:
> Urgh, I got the switch the wrong way around so fast mode was slow mode and visa versa. I push a fixed to main now.

Doh indeed. 

I've installed the new version.  Left the switch unchanged ;off).
Got a couple of observations from the prior version I'll send when I get home

gcoan, 2023-12-12:
This is with the prior version of main (switch the wrong way round).

12:57, the plan now correctly holds the charge to 16:30 and then releases it from there. This is probably caused by time advancing into the "8 slots ahead" and the switch not looking at all slots.
[image]

13:30, plan says freeze charging 87% limit and the inverter is setup correctly to do this 
[image]
[image]
[image]

But the alert I got was "freeze charging target 4%" - probably just an error in the alert message?
[image]

gcoan, 2023-12-12:
After I'd installed the new main beta (18c2206) we got a surprise saving session.

I decided to leave predbat running and managing the saving session as it looked to have planned to start releasing the charge at 16:30 and would have plenty to force discharge from 17:00-18:00.

The discharging started OK at 17:00 with a full 1 hour Timed Export to 18:01.
I next saw at 17:36 that it had set the mode to Idle, at which point I intervened, turned predbat into read only and did the force discharge for the rest of the period myself.
But looking at the status log I can see that actually at 17:05 mode was changed back to Eco (stopping discharge), with Timed Export setup to resume at 17:15-18:01.  Then predbat stopped this discharge and swapped to Eco at 17:35 with the next discharge slot being planned for 17:45-18:01.
So rather than a full 1 hour of discharge planned I got/would have got 5+20+15=40 minutes.

Here's the status log:
[image]
[image]

And the logfile for this period:
[appdaemon.log.1.txt](https://github.com/springfall2008/batpred/files/13652674/appdaemon.log.1.txt)

I'm not seeing the "WARN: No Octopus data in sensor sensor.octopus_energy_electricity_xxxx_next_rate attribute 'all_rates'" message any more - assume this is fixed now?

In terms of whether the calculate_fast_plan switch (off) is making a difference, here's the plan for tomorrow:
[image]
[image]

In terms of tomorrow's peak, the plan is better in that it retains 100% soc all the way up to see me through the most expensive slots at 16:00 (34.44p) and 16:30 (37.24p).  This evening, 20:00-00:00 its charging in the cheaper periods, releasing soc when the rates go up, so looks OK.  

There could have been a bit of further refinement in the early hours of the morning, the battery charging could have been delayed a bit further, the 03:30 slot (13.91p) is the cheapest of the night, but its not utilised as the battery is already reaches 100% soc in the preceding 03:00 (14.73p) slot.

But overall its better, the soc is planned to be held for the peak that wasn't happening before (until it got a lot closer to the peak time).

gcoan, 2023-12-13:
The new calculate_fast_plan set to off does seem to be calculating the whole plan better and I'm not seeing the plan "evolve" as much as it did before.

One question: if this is set to off, and the plan calc takes longer, is there still enough charge windows (eg plan hours x 2) so I'm getting the optimal plan or is there a restriction on plan window numbers that might be constraining things ?

springfall2008, 2023-12-13:
> One question: if this is set to off, and the plan calc takes longer, is there still enough charge windows (eg plan hours x 2) so > I'm getting the optimal plan or is there a restriction on plan window numbers that might be constraining things ?

So with the fast plan off the maximum in plan hours x 2 so all windows are considered. When it's on you will get plan hours windows which is mostly enough but some high rate slots will be ignored.

gcoan, 2023-12-13:
That's great.  This is giving me the full plan which is what I wanted to be able to see, not as much "evolving" as time moves forward and more windows are brought into scope.

Couple of further observations:
1. Some of the slots chosen eg overnight seem to favour early charging rather than a bit later for more optimal charging.  This is with calculate fast plan off and combine charge slots on (it was previously off but the manual recommends on for Agile):

[image]

e.g. Charging at 21:30/22:30/23:30 (all 15.52p) to reach 100% soc when 02:30 (14.11p), 03:00 (14.73p), 03:30 (13.91p) are all cheaper 

About 10 minutes ago the plan was also including 22:00 (17.11p), but that's now not included so that's an improvement
[image]

2. I have also noticed what looks to be a bug in the window period.  These were taken at 11:22, plan hours=30

With calculate fast plan off, slots planned to 16:30
[image]

With calculate fast plan on, slots to 17:00

JonathanLew1s, 2023-12-14:
Still seeing some strange choices with the 7.14.10 release

I see no reason to hold charge in this slot. The following slot is cheaper, the battery SOC value is not at risk before a future charge slot.

[image]

JonathanLew1s, 2023-12-14:
Saturday plan in a 48 hour window also makes little sense in decision to hold/freeze rather than charge.

[image]

JonathanLew1s, 2023-12-14:
Interestingly, changing from 48 to 24 hour results in a less than desirable plan.

48 hour window.
[image]

24 hour window. Choosing hold vs charge resulting in insufficient battery at 11am.
[image]
[_appdaemon_logs-5.txt](https://github.com/springfall2008/batpred/files/13672823/_appdaemon_logs-5.txt)

JonathanLew1s, 2023-12-15:
Same today. Predbat determination on holdcharge/freezecharge vs charge or chosing to hold or freeze rather than discharge is dubious in latest releases.

I find myself having to switch between a 24 and 48 hour window to force the calculation into a sensible solution but still see daft things like freeze/hold charge now whilst at 91% and next slots are cheaper.

[image]

I've reset everything to stock settings.

[image]

gcoan, 2023-12-15:
@JonathanLew1s its worth turning on html debug so you can see the rates including import losses because when you include the losses in discharging the battery, replenishing it again later, and the inverter loss, what looks like a cheaper rate isn't always worthwhile predbat releasing the charge only to have to charge back up later on.

But I do agree though, some of the decisions in the latest release are maybe what you might call dubious 

Earlier this morning, charging up to 100% through 21, 17, 19p slots when there's 16p in the afternoon. 
[image]

I put predbat into read only, inverter to Eco and let the batteries discharge.

The plan for this afternoon now looks fine
[image]

JonathanLew1s, 2023-12-15:
@gcoan I'm comfortable with the decision to occasional use the grid and hold battery, however the choices of when this happens is not optimal in the short term window. 

I'm currently using the grid to avoid discharge at 17.28 rate where a hold in the 17.11 slot in 30 minutes has the same effect. Loads in the slots are near identical.

[image]

gcoan, 2023-12-15:
I agree @JonathanLew1s , why import at 17.28p and then release charge to prevent grid import 2 slots later at 17.11p?

I think predbat rounds rates to the nearest 0.1p, but this'd make 12:30 17.3p, 13:00 17.1p an 13:30 17.1p - so why different behaviour of the 13:00 and 13:30 slots?  You've got plenty of soc in the battery, enough to support the saving session export and the subsequent periods. I don't know.

Comparing your settings to mine, they are very similar.
I have my losses as 5% (charge), 5% (discharge), 0% (inverter). Yours are 3, 3, 4, so they total the same, just allocated differently.

I don't have a battery cycle cost included as I figure I have already paid for the battery so might as well use it, rather than artificially prevent from using the battery.  You have 1.5p
I don't have a future import offset set, you do,  and my in-day damping is 1 yours is 0.95.
I'm planning for 30 hours, you are for 48. In my case I have such a small battery I could plan for just 24 but I leave it on 30 to see the next day's rates.

What about switch.predbat_calculate_tweak_plan, I don't see that on your dashboard?  I have it turned off.

I have combine charge slots set to true as well

JonathanLew1s, 2023-12-16:
Noticed that the calculate fast pass was active, have disabled yet still see the dubious choices in the plan. 

7.14.13 determines discharge in a 4p slot to charge at 8p in the next slot. When that 4p slot also aligns with higher demand, a charge appears beneficial on both lower import cost and avoided DC to AC conversion to cover the house load.

@springfall2008 am I missing something?
[_appdaemon_logs-6.txt](https://github.com/springfall2008/batpred/files/13693353/_appdaemon_logs-6.txt)

[image]

[image]

gcoan, 2023-12-17:
In general predbat is doing OK for me at the moment. There's definitely periods where I think it can optimise a bit better than it has, but the shape is OK.  Today with import rates being less than export rates its been very busy, imported 23kW, exported 6kW, total cost for the day so far 45p!

I have just upgraded to 7.14.14.  It selected one less charge then discharge slot this afternoon than 7.14.13, and it could be doing "more" in the early afternoon.  e.g. why not charge and discharge in 13:00, 13:30, 14:00 slots when the rates are all lower than the chosen charge/discharge slot at 14:30?
[image]

This isn't the end of the world, but it is confusing.

But one thing I do see both on .14 and .13 is discharging is planned that isn't possible.

e.g. after the evening peak the battery is exhausted on 4%.  The evening slots 19:00-21:30, chosen to leave the house being supplied by the grid as there's no benefit to charging and discharging after losses taken into account (that's my reading of it)
[image]

But then from 22:00 predbat's plan is to discharge through to 02:00.  The battery is of course empty so none of these discharges can be executed.   Yet the import rates are all << export rate so could import and then export for a profit (metric battery cycle cost =0)

Discharge is impossible and looks like charge then discharge would be profitable?
[Uploading appdaemon (4).log…]()

gcoan, 2023-12-17:
Update 18:50.
The long period of impossible discharge has shortened.  Still trying to discharge for the first hour (22:00 and 22:30) slots, but then charge in the next two before further discharging through to 02:00
[image]

This is better.  Predbat could still charge in the 22:30 slot to make it even better, but it is an improvement.
The more I look at it, the more I see the actual profit potential (apart from the midnight slot at 10.8p) is miniscule.
Its typically about 3p charge/discharge difference (minus conversion losses),  but as the inverter limit is 2.6kW, that's 1.3kW in a half hour slot (best case) so there's only actually any export when the house load is < 1.3kW, and then profit is only of the limited excess discharge amount.
Its more a matter of charging and discharging to avoid higher rate peaks which (other than the 22:30 slot), predbat is doing.

If it could stop marking slots as discharge slot when there's no soc, that'd be even better
I'll open a new issue for predbat keeps on trying to hold the battery charge when its empty

gcoan, 2023-12-19:
I think I'm pretty happy with the Predbat plans its coming up with now.  Maybe a bit of further tweaking could have been done, but its pennies optimisation now - e.g. charge at the 11:30-13:30 slots tomorrow for circa 15p rather than 14:00-16:30 at 16p.
The calculate fast plan option is giving me a forward plan I am much more happy with
[image]
[image]