springfall2008/batpred #451: What does 'PV forecast... total sensors add up to, this is unexpected' warning mean?

gcoan, 2023-12-11:
Maybe something for the FAQ's, but I've noticed the following new warning appearing in my logfile and I don't know what it means:

```yaml
2023-12-11 16:00:02.010382 INFO pred_bat: PV Data for pv_forecast_today total 2.78 kWh
2023-12-11 16:00:02.027215 INFO pred_bat: PV Data for pv_forecast_tomorrow total 1.75 kWh
2023-12-11 16:00:02.060778 INFO pred_bat: PV Data for pv_forecast_d3 total 1.29 kWh
2023-12-11 16:00:02.072136 INFO pred_bat: PV Data for pv_forecast_d4 total 1.62 kWh
2023-12-11 16:00:02.077030 INFO pred_bat: WARN: PV Forecast data adds up to 14.89 kWh but total sensors add up to 7.4399999999999995 KWh, this is unexpected and hence data maybe misleading (factor 2.01)
```

The PV data for today, tomorrow, D3 and D4 match what I see in the solcast forecast entities, but I don't understand what is being added up to get to 7.43999 kWh, and what I should/could do about it?

Note as well, in the log message, missing rounding and should be 'kWh' not 'KWh'

Thanks

springfall2008, 2023-12-11:
It’s just a rounding error, you can ignore it but obviously my last fix didn’t clear this will try again

springfall2008, 2023-12-12:
Fixed in https://github.com/springfall2008/batpred/releases/tag/v7.14.8