springfall2008/batpred #454: Hold charges appearing in plan with losses and cycle set to 0

AGLAISTER, 2023-12-12:
I have adjusted my settings to avoid predbat causing a charge hold but I'm still getting charge holds even though as advised I've set 
Battery Loss Charge = 0
Battery Loss Discharge = 0
Inverter Loss = 0
Metric Battery Cycle cost = 0

The example below is not the best but illustrates it is happening - I have mainly been seeing it in a morning at breakfast time

Images and 

[image]

[image]

[image]

Logs attached

[appdaemon.log](https://github.com/springfall2008/batpred/files/13650690/appdaemon.log)

[appdaemon 1.log](https://github.com/springfall2008/batpred/files/13650692/appdaemon.1.log)

springfall2008, 2023-12-12:
This is probably due to the Best Soc keep at 1.5 meaning the hold is required to avoid getting below that. Try a keep of 0 or 0.5

AGLAISTER, 2023-12-12:
1.5/9.5 gives 15% so and when its holding charge the plan is showing 19%
I had best soc at 1.5 to keep enough charge going in for poor solar days as i was seeing it under plan for the next day.
I've reduced to 1 for now and that has removed that SoC keep
I'll monitor things and see what happens - I'll close this for now

Reducing in to 1 did remove the hold charge, I had it at 1.5 to ensure I was getting full charge overnight on poor solar days

springfall2008, 2023-12-12:
Soc keep is ignored for the next 4 hours which is why the slot disappears at you get closer, it maybe fine to leave it once you know why

AGLAISTER, 2023-12-14:
I had another occurrence of charge being held this morning even with all parameters set that I understand should be preventing charge hold

[appdaemon.log](https://github.com/springfall2008/batpred/files/13673733/appdaemon.log)
[appdaemon 1.log](https://github.com/springfall2008/batpred/files/13673734/appdaemon.1.log)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Note I'm still on v7.13.23 as I haven't had the right time/opportunity to update with the breaking change in v7.14

AGLAISTER, 2023-12-17:
got another occurrence today of charge holding.

[image]
[image]

settings are unchanged except that I pushed up load scaling yesterday to ensure I got full charge overnight due to low prices.

springfall2008, 2024-09-01:
Closing this as the model was re-written since. Please open a new ticket for any new issues

closed 2024-09-01 by springfall2008