springfall2008/batpred #456: See lots of errors with the newest release

Redding43, 2023-12-12:
Firstly I should say I am not ruling out it being a Octopus or GivTCP issue as my OctoAid app is erroring when trying to load rates currently

however, started seeing these errors after the update of Predbat today

[image]
[image]
[image]

springfall2008, 2023-12-12:
It looks like REST isn’t working, maybe try restarting GivTCP ?

Redding43, 2023-12-12:
Tried a restart. Seeing if it has fixed it. Spotted this just now in GivTCP logs though

[image]

gcoan, 2023-12-12:
@Redding43 I've seen something similar happen.  Restarting GivTCP in my case fixed it.

I did log #330 to see if any enhancements could be made to predbat's retry mechanism.

Also BTW I did add a couple of sample automations to the documentation https://springfall2008.github.io/batpred/output-data/ that monitor predbat and GivTCP for disconnects to give a better warning when one of these components stops working

Redding43, 2023-12-12:
Back up and working now, thanks all. As you say I think it was the GivTCP add-on

closed 2023-12-13 by springfall2008