springfall2008/batpred #457: Freeze charging target in alert msg sometimes wrong

gcoan, 2023-12-13:
Splitting this out from #449 as it's a separate issue,

Have seen twice now examples of the Freeze Charge alert notification having a 4% soc target, not the correct target per the plan and the inverter settings:

[image]
[image]
[image]

gcoan, 2023-12-13:
Yesterday's example of the same issue (moved from #449)

13:30, plan says freeze charging 87% limit and the inverter is setup correctly to do this 
[image]
[image]
[image]

But the alert I got was "freeze charging target 4%" - probably just an error in the alert message?
[image]

springfall2008, 2023-12-13:
Fixed in: https://github.com/springfall2008/batpred/commit/e53f8c7e59368aca97547da1ecff9212a3f6aa90

gcoan, 2023-12-13:
I downloaded the latest main, its d423cfe though?

springfall2008, 2023-12-15:
Did the fix work?

gcoan, 2023-12-15:
> Did the fix work?

I believe so.  I've received lots of freeze charging notifications and looking through them, none of them say 4%, they all say different soc target values; 98, 100, 51, etc

Thanks

closed 2023-12-15 by gcoan