springfall2008/batpred #458: Predbat didn't keep executing the plan for force discharge for whole of the saving session

gcoan, 2023-12-13:
Splitting this out from #449 as its a separate issue

After I'd installed the new main beta (18c2206) we got a surprise saving session.

I decided to leave predbat running and managing the saving session as it looked to have planned to start releasing the charge at 16:30 and would have plenty to force discharge from 17:00-18:00.

The discharging started OK at 17:00 with a full 1 hour Timed Export to 18:01.
I next saw at 17:36 that it had set the mode to Idle, at which point I intervened, turned predbat into read only and did the force discharge for the rest of the period myself.
But looking at the status log I can see that actually at 17:05 mode was changed back to Eco (stopping discharge), with Timed Export setup to resume at 17:15-18:01.  Then predbat stopped this discharge and swapped to Eco at 17:35 with the next discharge slot being planned for 17:45-18:01.
So rather than a full 1 hour of discharge planned I got/would have got 5+20+15=40 minutes.

Here's the status log:
[image]
[image]

And the logfile for this period:
[appdaemon.log.1.txt](https://github.com/springfall2008/batpred/files/13652674/appdaemon.log.1.txt)

springfall2008, 2023-12-13:
There's a fix on main that may help with this, I'd certainly value feedback.

The main change is to count metric_min_improvement to be for discharging or not rather than the specific time period or level picked. This will allow bigger slots to be picked when its marginal.

gcoan, 2023-12-13:
> There's a fix on main that may help with this, I'd certainly value feedback.
> 
> The main change is to count metric_min_improvement to be for discharging or not rather than the specific time period or level picked. This will allow bigger slots to be picked when its marginal.

OK I'll try it when we next get a saving session.  At the moment home demand>>battery capacity so no exporting going on under normal circumstances :-(

gcoan, 2023-12-13:
Turns out I had opportunity to see how forced discharging worked sooner than I expected.

Today we had an Octopus Power-Up event, 2 hours of free electricity so Predbat charged the battery up fully for free and I turned up the flow temperature on the ASHP to capitalise on the free electricity.

The power up event was 3-5pm, so that covered the first hour of the peak agile prices.  Then as the water in the heating pipes was hotter than normal the ASHP didn't call for any heat from 5-6pm, and so the battery was at 77% soc at 6pm.  Predbat thus planned and executed a force discharge for the 6-7pm period.

Predbat behaved very well.  Over the 1 hour period it stayed discharging for the entire time.  There were 4 different battery reserve figures set, progressively getting higher as time went on, but I don't think the changing reserve soc had any impact to the inverter activity.
There was a period from about 18:26 to 18:35 (and a second one from 18:49 to 18:52) where the inverter seemed to stop discharging the battery and exporting, but I can't see any interactions that predbat caused during this.  Probably just inverter weirdness.

Status:
[image]

Export and discharge power:
[image]

Battery reserve and soc:
[image]

logfile:
[appdaemon.log.1 (2).txt](https://github.com/springfall2008/batpred/files/13665359/appdaemon.log.1.2.txt)

This all looked good to me from a discharge activity.
One observation in the logfile, when the reserve was adjusted it looks like predbat changes the discharge times, even though there's actually no change.  e.g. 18:30:

2023-12-13 18:30:31.372285 INFO pred_bat: Inverter 0 Adjust force discharge to True, change times from 18:00:00 - 19:01:00 to 18:00:00 - 19:01:00

The GivTCP logs don't seem to include any details of discharge times being changed; only charge times, so I can't tell if the inverter times were actually changed or not
[log_inv_2.log](https://github.com/springfall2008/batpred/files/13665421/log_inv_2.log)

springfall2008, 2023-12-13:
Sounds good, I’ll release main shortly then

gcoan, 2023-12-15:
Update from today's saving session, I left predbat to manage the discharge entirely on its own (turning off my own over-ride discharge automation).

In short, worked perfectly, executed a forced discharge from 17:00 to 18:00, exporting a pretty constant 2kW throughout.

4 times predbat revised the reserve soc (from 11% at the start session) to 9%, 8%, 9% then 8%), at 20 then 10, 10 and 10 minutes between each of the changes (interestingly all were on the x5 minutes past the hour so when the plan wasn't re evaluated).

But other than a bit of inverter wobbling (can't see any reason for that, not related to battery or inverter temperature) it was solid discharge 
[image]
[image]

closed 2023-12-15 by gcoan