springfall2008/batpred #461: Not exported as much as expected

springfall2008, 2023-12-13:
Richard is finding export amounts have been lower than expected overnight

springfall2008, 2023-12-13:
[Uploading appdaemonlog.txt.log…]()

springfall2008, 2023-12-13:
[image]

springfall2008, 2023-12-13:
I think there are a couple of things you could try here:

1. Try setting metric_min_improvement_discharge to 0.1 to see what it does to then plan.
2. Maybe update to the current version on 'main' and see if it helps

gcoan, 2023-12-13:
Looks like its fully charging the battery first, then goes through a cycle of discharge for 30 mins, charge back up for 30 mins, etc. I notice each cycle of charge and discharge doesn't quite get the soc back to the same 100% position, presumably due to losses in the charge/discharge?

Not having seen the config settings, is charge & discharge loss %'s set the same?
Could potentially set 'combine charge slots" to false and calculate discharge on charge slots to true to encourage more intermediate charging.

But what expectation is there of extra export, it looks reasonable to me?

pottsrichard, 2023-12-13:
> I think there are a couple of things you could try here:
> 
> 1. Try setting metric_min_improvement_discharge to 0.1 to see what it does to then plan.
> 2. Maybe update to the current version on 'main' and see if it helps

Thanks Trefor. Metric Min Improvement and Metric Min Improvement Discharge are both set to zero. Happy to update, though also tempted to downgrade, as this has only been happening since Sunday night. 

[image]

pottsrichard, 2023-12-13:
> Looks like its fully charging the battery first, then goes through a cycle of discharge for 30 mins, charge back up for 30 mins, etc. I notice each cycle of charge and discharge doesn't quite get [...]

Thanks gcoan - is there an easy way to export the config settings for you other than several screenshots? The config's been adapted over several months with Trefor to maximise profit from export. Up until Sunday night, Batpred would typically export between 3.9KWh to 4.1KWh every night. Sunday night it exported 0KWh, Monday exported 1.3KWh and Tuesday exported 0.6KWh. Nothing's changed at this end other than incremental updates to Batpred. The house usage remains typical for us.

gcoan, 2023-12-13:
Hi Richard, no there isn't any way to export the config other than screen shots.

But I'm thinking if it's not discharging anything and the plan looks reasonable then it's not immediately likely to be a predbat issue.

You could look in the predbat and givtcp logs to see if there are any issues reported with setting the discharge, but to be honest I'd be resetting the inverter and restarting givtcp.
In the GivEnergy portal, do a reset to defaults and then a restart of the inverter. 
Then you could try in the GivEnergy app manually doing an export and see if it starts ok.

I've seen similar issues before where the inverter doesn't do what you want.  Might get an error in the givtcp logs or might not

pottsrichard, 2023-12-13:
> Hi Richard, no there isn't any way to export the config other than screen shots.
>
> But I'm thinking if it's not discharging anything and the plan looks reasonable then it's not immediately likely to [...]

Thanks Geoff. I've now reset to defaults, restarted the inverter, restarted GivTCP and also picked-up Trefor's latest Batpred update from earlier. The plan looks good, so fingers crossed for a return to normal service tonight. Cheers, Richard

springfall2008, 2023-12-13:
There is an updated related partly to export on main that have not released yet.

pottsrichard, 2023-12-14:
Morning Trefor/Geoff, I'm sorry to report there was no post charge export last night, despite the earlier plan looking great. It did, however export yesterday's remaining battery before charging from midnight, so it's certainly able to export. Note, it also exported fine during the Octopus saving session on Tuesday. Could it have some bad data in there? Is it worth me clearing out all my usage history? I'll add my config screenshots later when I get a minute. 

[image]
[image]

springfall2008, 2023-12-14:
Was your last run from the version on main or from the last release?

pottsrichard, 2023-12-14:
> Was your last run from the version on main or from the last release?

Thanks Trefor, it was the last release that HA picked-up yesterday afternoon, v 7.14.8. I've just been onto main and downloaded (showing as 841c953), so will see how that goes tonight, thank you. 

One other thing that's crossed my mind is that I sold my EV last week and won't have a replacement for a few weeks. Would Batpred see this as a reduction in usage somehow and it be affecting the exports?

springfall2008, 2023-12-14:
>v 7.14.8. I've just been onto main and downloaded (showing as https://github.com/springfall2008/batpred/commit/841c953442f47d727aff22dad02cb58cc9a9e812), so will see how that goes tonight, thank you.

I did make a release around 30 minutes ago, but taking main is also fine.

>One other thing that's crossed my mind is that I sold my EV last week and won't have a replacement for a few weeks. Would Batpred see this as a reduction in usage somehow and it be affecting the exports?

It won't be related.

gcoan, 2023-12-14:
Hi Richard

The overnight plan looked entirely reasonable to me, it just didn't execute it.
Could you share the history of what happened to predbat status overnight to see what predbat actually tried to execute.

Then maybe the graphs of battery soc, target soc, reserve soc overnight and battery mode.  Check what discharge rate and start/end charge and discharge times changed to.

Together these will help build a picture of what actually happened on the inverter.

You've checked the givtcp and app daemon logs for errors? 

Geoffrey

pottsrichard, 2023-12-14:
> Hi Richard
>
> The overnight plan looked entirely reasonable to me, it just didn't execute it. Could you share the history of what happened to predbat status overnight to see what predbat actually [...]

Thanks Geoffrey, see below for the GivTCP log. One thing to note, at 3am, my router does an auto-reboot - it's always done this, I'm more interested in '[ERROR   ] - Battery Object empty so skipping', what might cause this, or is it normal?

 Appdaemon log attached. I'll try and get the other things together this evening. Thank you so much for your help/support. 

Richard

[appdaemonlog 14th Dec.txt](https://github.com/springfall2008/batpred/files/13675385/appdaemonlog.14th.Dec.txt)

2023-12-14 01:31:16,807 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2023-12-14 01:31:22,928 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2023-12-14 02:10:57,028 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 3600 (50)
2023-12-14 02:11:00,971 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2023-12-14 02:11:07,576 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 02:00 - 02:50
2023-12-14 02:11:15,063 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2023-12-14 02:11:21,699 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 90
2023-12-14 02:11:27,955 - Inv1 - write       -  [INFO    ] - Setting Charge Target 90 was a success
2023-12-14 02:11:34,948 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 90
2023-12-14 02:11:40,711 - Inv1 - write       -  [INFO    ] - Setting shallow charge 90 was a success
2023-12-14 02:20:10,450 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 to: 02:50 - 03:01
2023-12-14 02:20:14,124 - Inv1 - write       -  [INFO    ] - Setting Discharge Slot 1 was a success
2023-12-14 02:21:08,251 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 89
2023-12-14 02:21:14,286 - Inv1 - write       -  [INFO    ] - Setting Charge Target 89 was a success
2023-12-14 02:21:20,918 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 89
2023-12-14 02:21:26,938 - Inv1 - write       -  [INFO    ] - Setting shallow charge 89 was a success
2023-12-14 02:30:56,941 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 88
2023-12-14 02:30:59,409 - Inv1 - write       -  [INFO    ] - Setting Charge Target 88 was a success
2023-12-14 02:31:06,099 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 88
2023-12-14 02:31:12,123 - Inv1 - write       -  [INFO    ] - Setting shallow charge 88 was a success
2023-12-14 02:40:55,360 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 05:00 - 05:30
2023-12-14 02:41:02,312 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2023-12-14 02:41:08,923 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2023-12-14 02:41:14,963 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2023-12-14 02:41:21,572 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2023-12-14 02:41:27,628 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2023-12-14 02:56:02,753 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2023-12-14 03:00:25,933 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:00:28,438 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:00:30,942 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:00:33,446 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:00:35,951 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:00:38,453 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:00:50,413 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:00:52,917 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:00:58,671 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.48, 8899) failed: timed out
2023-12-14 03:16:10,021 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping
2023-12-14 04:10:59,125 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 3600 (50)
2023-12-14 04:11:01,494 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 50 was a success
2023-12-14 04:11:07,995 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 to: 04:00 - 04:45
2023-12-14 04:11:15,228 - Inv1 - write       -  [INFO    ] - Setting Charge Slot 1 was a success
2023-12-14 04:11:21,838 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 100
2023-12-14 04:11:27,946 - Inv1 - write       -  [INFO    ] - Setting Charge Target 100 was a success
2023-12-14 04:11:33,278 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 100
2023-12-14 04:11:39,351 - Inv1 - write       -  [INFO    ] - Setting Charge Target 100 was a success
2023-12-14 04:11:45,943 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 98
2023-12-14 04:11:51,895 - Inv1 - write       -  [INFO    ] - Setting shallow charge 98 was a success
[... cut ...]
2023-12-14 04:41:05,465 - Inv1 - write       -  [INFO    ] - Setting battery charge rate to: 0 (0)
2023-12-14 04:41:11,328 - Inv1 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2023-12-14 04:41:18,317 - Inv1 - write       -  [INFO    ] - Setting battery reserve target to: 4
2023-12-14 04:41:23,869 - Inv1 - write       -  [INFO    ] - Setting shallow charge 4 was a success
2023-12-14 05:30:03,962 - Inv1 - read        -  [INFO    ] - Saving current energy stats at start of day rate tariff (Dynamic)

gcoan, 2023-12-14:
Hi Richard, I've not looked at the predbat log but the GivTCP log shows it didn't even try to do much discharging:

2:10 instruction to charge to 90% from 02:00-02:50

2:20 instruction to discharge from 02:50-03:01

2:41 battery reserve set to 4% (so discharge down to this level)

4:11 instruction to charge to 100% 04:00-04:45

4:21 instruction to discharge 04:50-05:01

4:40 instruction to charge 00:00-02:25

4:41 battery reserve set to 4%

So there would have only been 2 10 minute slots of discharging.
Nothing set between 3:00 and 4:00.  The router reboot may have impacted comms between GivTCP and the inverters, and if you can find out how to stop your router rebooting every night that would remove one potential source of things stopping working, but for some reason predbat didn't even try to programme much in the way of discharging

The battery object empty is a potential concern, its saying that it couldn't get the battery data, so givtcp will reuse the previous it had, but that only happened twice.

Do you have a copy of the predbat plan for last night?  Can work it out from the logfile but the plan is easier to read

pottsrichard, 2023-12-15:
> Hi Richard, I've not looked at the predbat log but the GivTCP log shows it didn't even try to do much discharging:
>
> 2:10 instruction to charge to 90% from 02:00-02:50
>
> 2:20 instruction to discharge [...]

Dear Geoffrey/Trefor,

In the words of Randolph Duke (in 'Coming to America'), "we're back!" 

[image]

I don't exactly know what did it, be-it the latest version of Batpred or hard re-booting the router (I'd soft re-booted on the previous days), re-booting HA, restarting GivTCP and restarting Appdaemon, but last night's export was excellent.  Fingers crossed for a continuation tonight!

Greg, point taken about the router re-booting at 3am. I have had it set to do this every other night for years and it's never previously caused an issue with Batpred or the Givenergy system. However, it certainly makes sense to change this - I've now set it to 5:35am to avoid any cheap rate charging etc. 

Thank so much for taking the time to go through the logs, Geoffrey. That gives me a really good idea of what I need to be looking at when this happens again. 

Many thanks both!
Cheers,
Richard

closed 2023-12-16 by springfall2008

gcoan, 2023-12-17:
Glad you got it sorted Richard.  I'm still leaning in the direction of there being an issue with the inverter not doing what it was told to do, which led to the lack of export, but a strange one.
Moving the router reboot outside of the time you want to be controlling the charge and discharge, definitely a good idea.

Fingers crossed !