springfall2008/batpred #462: not resetting discharge slot params

marcoaddario, 2023-12-13:
**Describe the bug**
Yesterday predbat managed to charge the batteries during the afternoon in preparation to the discharge them during the saving session slot.
It correctly set the discharge time (17:00 to 18:00) and the discharge current.
At the end of the session set the discharge current to 0A to stop the discharge and then set again to full rate after a while.
Yesterday night the discharge current has been set to 0 to avoid charging the car from batteries and at the end of the night tariff the discharge current has been set again to full.
As the discharge time has not been reset, the inverter discharged again today even if there was no reason for that.
I then set it back to 00:00 to 00:00 to stop it.

**Expected behavior**
at the end of the discharge session, the time should be reset or brought back to the previous value

**Predbat version**
7.14.8

**Environment details**

- Inverter and battery setup: Solis
- Standard HAOS installer

**Log file**
This is the extract with only the events related to the changes to the discharge_* params
```
┌2023-12-12 16:30:04.832200 INFO pred_bat: Inverter 0 Wrote 17:00:00 to discharge_start_time, successfully now 17:00:00                                                                           │
│2023-12-12 16:30:06.841428 INFO pred_bat: Inverter 0 Wrote 17 to discharge_start_hour, successfully now 17                                                                                       │
│2023-12-12 16:30:06.848147 INFO pred_bat: Inverter 0 No write needed for discharge_start_minute: 0 == 0                                                                                          │
│2023-12-12 16:30:08.876437 INFO pred_bat: Inverter 0 Wrote 18:01:00 to discharge_end_time, successfully now 18:01:00                                                                             │
│2023-12-12 16:30:10.882211 INFO pred_bat: Inverter 0 Wrote 18 to discharge_end_hour, successfully now 18                                                                                         │
│2023-12-12 16:30:12.891035 INFO pred_bat: Inverter 0 Wrote 1 to discharge_end_minute, successfully now 1                                                                                         │
│2023-12-12 16:30:14.897228 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0                                                            │
│2023-12-12 17:00:05.008846 INFO pred_bat: Inverter 0 Wrote 3600 to discharge_rate, successfully now 3600                                                                                         │
│2023-12-12 17:00:07.015521 INFO pred_bat: Inverter 0 Wrote 71.9 to timed_discharge_current, successfully now 71.9                                                                                │
│2023-12-12 17:30:17.227027 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0                                                            │
│2023-12-12 18:00:05.320005 INFO pred_bat: Inverter 0 Wrote 0 to discharge_rate, successfully now 0                                                                                               │
│2023-12-12 18:00:07.328616 INFO pred_bat: Inverter 0 Wrote 0.0 to timed_discharge_current, successfully now 0.0                                                                                  │
│2023-12-12 18:00:31.850206 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0                                                            │
│2023-12-12 18:00:33.891268 INFO pred_bat: Inverter 0 Wrote 3600 to discharge_rate, successfully now 3600                                                                                         │
│2023-12-12 18:00:35.901107 INFO pred_bat: Inverter 0 Wrote 76.6 to timed_discharge_current, successfully now 76.6                                                                                │
│2023-12-12 22:00:15.484945 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0                                                            │
│2023-12-12 22:30:05.557462 INFO pred_bat: Inverter 0 Wrote 0 to discharge_rate, successfully now 0                                                                                               │
│2023-12-12 22:30:07.565717 INFO pred_bat: Inverter 0 Wrote 0.0 to timed_discharge_current, successfully now 0.0                                                                                  │
│2023-12-12 23:45:11.646248 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0                                                            │
│2023-12-13 00:00:21.747942 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0                                                            │
│2023-12-13 08:00:04.237729 INFO pred_bat: Inverter 0 Wrote 3600 to discharge_rate, successfully now 3600                                                                                         │
│2023-12-13 08:00:06.247353 INFO pred_bat: Inverter 0 Wrote 72.6 to timed_discharge_current, successfully now 72.6                                                                                │
│2023-12-13 21:00:14.251288 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0                                                            │
│2023-12-13 21:30:04.310348 INFO pred_bat: Inverter 0 Wrote 0 to discharge_rate, successfully now 0                                                                                               │
│2023-12-13 21:30:06.316764 INFO pred_bat: Inverter 0 Wrote 0.0 to timed_discharge_current, successfully now 0.0                                                                                  │
│2023-12-13 22:19:45.936607 INFO pred_bat: Inverter 0 Wrote 0 to discharge_rate, successfully now 0                                                                                               │
│2023-12-13 22:19:45.943356 INFO pred_bat: Inverter 0 No write needed for timed_discharge_current: 0.0 == 0.0                                                                                     │
│2023-12-13 22:19:52.046867 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0
```

springfall2008, 2023-12-14:
The fix is in this release: https://github.com/springfall2008/batpred/releases/tag/v7.14.10

marcoaddario, 2023-12-14:
Thanks to the afternoon saving session I tested it and I'm not sure if the fix should have zeroed the time-window when disabled the discharge mode or if it should have reset it later.
This is the actual setup (after the end of the saving session):
[image]

This is the list of the predbat.status status changes (apologies for the reverse order... is the way the HA logbook shows them):
```
Status changed to Freeze charging
19:10:16 - 2 hours ago - Supervisor
Status changed to Inverter 0 set soc to 15
19:10:04 - 2 hours ago - Supervisor
Status changed to Freeze charging
19:05:04 - 2 hours ago - Supervisor
Status changed to Inverter 0 set soc to 18
19:05:04 - 2 hours ago - Supervisor
Status changed to Freeze charging
19:00:34 - 2 hours ago - Supervisor
Status changed to Inverter 0 set soc to 21
19:00:34 - 2 hours ago - Supervisor
Status changed to Inverter 0 Charge window change to: 18:00:00 - 05:30:00
19:00:32 - 2 hours ago - Supervisor
Status changed to Inverter 0 discharge rate changed to 0
19:00:18 - 2 hours ago - Supervisor
Status changed to Idle
18:40:14 - 2 hours ago - Supervisor
Status changed to ERROR: Exception raised 'PredBat' object has no attribute 'base'
18:35:00 - 3 hours ago - Supervisor
Status changed to Idle
18:00:04 - 3 hours ago - Supervisor
Status changed to Inverter 0 Set discharge mode to Eco
18:00:04 - 3 hours ago - Supervisor
Status changed to Discharging
17:05:52 - 4 hours ago - Supervisor
Status changed to Inverter 0 set discharge slot to 17:00:00 - 18:01:00
17:05:52 - 4 hours ago - Supervisor
Status changed to Inverter 0 Set discharge mode to Timed Export
17:05:04 - 4 hours ago - Supervisor
Status changed to Idle
17:00:06 - 4 hours ago - Supervisor
Status changed to Inverter 0 discharge rate changed to 3600
17:00:06 - 4 hours ago - Supervisor
Status changed to Freeze charging
16:40:04 - 4 hours ago - Supervisor
Status changed to Inverter 0 set soc to 87
16:40:04 - 4 hours ago - Supervisor
Status changed to Freeze charging
16:35:28 - 5 hours ago - Supervisor
Status changed to Inverter 0 set discharge slot to 17:05:00 - 18:01:00
16:35:16 - 5 hours ago - Supervisor
```

The exeption are a typo in the saving event joining notify 
``` self.base.call_notify("Predbat: Joined Octopus saving event: start {}, end {}, price per kWh {}".format(start, end, saving_rate))                                                            │
│    ^^^^^^^^^                                                                                                                                                                                    │
│AttributeError: 'PredBat' object has no attribute 'base'
```
[predbat_20221214.log](https://github.com/springfall2008/batpred/files/13678535/predbat_20221214.log)

gcoan, 2023-12-14:
The last error message was when predbat sent a mobile notification that it was joining the saving session. Bug on my part, it's fixed in the latest release. Only occurs at the moment of joining the saving session, it joins ok but then crashes when sending the mobile alert.
Next 5 minute run of predbat as you've already joined the saving session it runs ok

springfall2008, 2023-12-15:
@marcoaddario I think I found the issue, will release a fix soon

springfall2008, 2023-12-15:
Can you try the latest from 'main' please?

marcoaddario, 2023-12-15:
It raised an exception:
```
2023-12-15 10:30:19.811736 INFO pred_bat: Inverter 0 No write needed for charge_start_hour: 5 == 5
2023-12-15 10:30:19.817719 INFO pred_bat: Inverter 0 No write needed for charge_start_minute: 30 == 30
2023-12-15 10:30:21.839204 INFO pred_bat: Inverter 0 Wrote 17:00:00 to charge_end_time, successfully now 17:00:00
2023-12-15 10:30:21.845250 INFO pred_bat: Inverter 0 No write needed for charge_end_hour: 17 == 17
2023-12-15 10:30:21.851418 INFO pred_bat: Inverter 0 No write needed for charge_end_minute: 0 == 0
2023-12-15 10:30:23.856975 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0
2023-12-15 10:30:23.868673 INFO pred_bat: Inverter 0 Updated start and end charge window to 05:30:00 - 17:00:00 (old 23:59:00 - 23:59:00)
2023-12-15 10:30:23.870109 INFO pred_bat: Next discharge window will be: 2023-12-15 17:00:00+00:00 - 2023-12-15 18:01:00+00:00 at reserve 32
2023-12-15 10:30:23.871572 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 12-15 17:00:00 - 12-15 18:00:00
2023-12-15 10:25:42.406768 INFO pred_bat: ERROR: Exception raised 'Inverter' object has no attribute 'midnight_utc'
2023-12-15 10:25:42.414160 WARNING pred_bat: ------------------------------------------------------------
2023-12-15 10:25:42.414551 WARNING pred_bat: Unexpected error in worker for App pred_bat:
2023-12-15 10:25:42.415348 WARNING pred_bat: Worker Ags: {'id': '63940c01328647709794acf27d36ef1d', 'name': 'pred_bat', 'objectid': 'ebcbe17c69a047b395ad0d6d2fd48cec', 'type': 'scheduler', 'function': <bound method PredBat.update_time_loop of <predbat.PredBat object at 0x7fd649918250>>, 'pin_app': True, 'pin_thread': 1, 'kwargs': {'interval': 15, 'random_start': 0, 'random_end': 0, '__thread_id': 'thread-1'}}
2023-12-15 10:25:42.415710 WARNING pred_bat: ------------------------------------------------------------
2023-12-15 10:25:42.418200 WARNING pred_bat: Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/threading.py", line 1022, in worker
    funcref(self.AD.sched.sanitize_timer_kwargs(app, args["kwargs"]))
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9907, in update_time_loop
    raise e
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9902, in update_time_loop
    self.update_pred(scheduled=False)
  File "/usr/lib/python3.11/site-packages/appdaemon/adbase.py", line 35, in f_app_lock
    return f(*args, **kw)
           ^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 9265, in update_pred
    status, status_extra = self.execute_plan()
                           ^^^^^^^^^^^^^^^^^^^
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 8384, in execute_plan
    inverter.adjust_force_discharge(False)
  File "/homeassistant/appdaemon/apps/batpred/predbat.py", line 1552, in adjust_force_discharge
    new_start_time = self.midnight_utc
                     ^^^^^^^^^^^^^^^^^
AttributeError: 'Inverter' object has no attribute 'midnight_utc'

2023-12-15 10:25:42.418424 WARNING pred_bat: ------------------------------------------------------------
```

springfall2008, 2023-12-15:
Okay please update from main again, I fixed the typo

marcoaddario, 2023-12-15:
looks perfect now:
```
2023-12-15 11:15:15.003053 INFO pred_bat: Low import rate windows [0: 12-15 05:30:00 - 12-15 17:00:00 @ 29.58, 1: 12-15 18:00:00 - 12-15 23:30:00 @ 29.58, 2: 12-15 23:30:00 - 12-16 05:30:00 @ 7.5, 3: 12-16 05:30:00 - 12-16 23:30:00 @ 29.58, 4: 12-16 23:30:00 - 12-17 05:30:00 @ 7.5, 5: 12-17 05:30:00 - 12-17 23:15:00 @ 29.58]
2023-12-15 11:15:15.011603 INFO pred_bat: Charge window will be: 2023-12-15 05:30:00+00:00 - 2023-12-15 17:00:00+00:00 - current soc 99 target 15
2023-12-15 11:15:15.014920 INFO pred_bat: Configuring charge window now (now 12-15 11:15:00 target set_window_minutes 30 charge start time 12-15 05:30:00
2023-12-15 11:15:15.018410 INFO pred_bat: Next discharge window will be: 2023-12-15 17:00:00+00:00 - 2023-12-15 18:01:00+00:00 at reserve 32
2023-12-15 11:15:15.021058 INFO pred_bat: Setting ECO mode as we are not yet within the discharge window - next time is 12-15 17:00:00 - 12-15 18:00:00
2023-12-15 11:15:15.024298 INFO pred_bat: Inverter 0 Adjust force discharge to False, change times from 23:59:00 - 23:59:00 to 00:00:00 - 00:00:00
2023-12-15 11:15:15.026112 INFO pred_bat: Inverter 0 set new start time to 00:00:00
2023-12-15 11:15:17.037827 INFO pred_bat: Inverter 0 Wrote 00:00:00 to discharge_start_time, successfully now 00:00:00
2023-12-15 11:15:19.045530 INFO pred_bat: Inverter 0 Wrote 0 to discharge_start_hour, successfully now 0
2023-12-15 11:15:19.052107 INFO pred_bat: Inverter 0 No write needed for discharge_start_minute: 0 == 0
2023-12-15 11:15:19.057045 INFO pred_bat: Inverter 0 Set new end time to 00:00:00 was 23:59:00
2023-12-15 11:15:21.080021 INFO pred_bat: Inverter 0 Wrote 00:00:00 to discharge_end_time, successfully now 00:00:00
2023-12-15 11:15:23.102265 INFO pred_bat: Inverter 0 Wrote 0 to discharge_end_hour, successfully now 0
2023-12-15 11:15:25.112558 INFO pred_bat: Inverter 0 Wrote 0 to discharge_end_minute, successfully now 0
2023-12-15 11:15:27.119324 INFO pred_bat: Successfully pressed button button.solis_modbus_update_charge_discharge_times on Inverter 0
```
[image]

Thanks a lot again

closed 2023-12-15 by marcoaddario