springfall2008/batpred #475: Beta version on 'main' for testing

springfall2008, 2023-12-15:
I made too many changes on main so they really need testing.

- When the plan hours < 48 hours now (e.g. 24) the blackout period is calculated based on the next low rate charge slot and not any slots
- Added load_scaling10 to allow the worst case scenario to include higher load - default 1.1
  Here load_scaling10 will skew the house load in the 10% scenario to try to get Predbat to see that you might use more energy than expected as well as have a worse solar day.

* Fix charge freeze to disable the charge window when holding using reserve
* Put some more options into expert mode
* Try to stop charge window from being disabled when it doesn't need to be

gcoan, 2023-12-15:
I'm installing it now Trefor and will see how it goes.

Don't quite understand how the new load scaling works alongside the existing and why I might or might not want to change it.  Thanks

springfall2008, 2023-12-15:
> I'm installing it now Trefor and will see how it goes.
> 
> Don't quite understand how the new load scaling works alongside the existing and why I might or might not want to change it. Thanks

So it will scale the load used in the 10% calculation, it kind of makes the best10 into a worst case line

gcoan, 2023-12-15:
I installed the beta version just after 1pm and for the first part of the afternoon the plan was to charge the battery which occurred fine

Between 1:35 and 3:30 it swapped a couple of times between hold charge and freeze charge

The freeze charge I saw it doing was working fine 
[image]

The plan at 3:05 showed what appeared to be just normal Eco discharging in the saving session period from 5-6, which looked wrong
[image]

But by 4:30 the plan had changed to be a proper forced discharge (as expected) and as you can see the inverter was just changing to idle in the peak, ahead of the saving session 
[image]

At 5pm it started the force discharge correctly and held it force discharging to 6pm when it went into idle/Eco mode
[image]

So other than the plan being wrong initially for the saving session, but it corrected it, it worked fine for me

gcoan, 2023-12-15:
Are these changes in 7.14.13 or should I keep beta testing main?

springfall2008, 2023-12-15:
Yeh unfortunately I had an issue with the gap filling and so wanted a new release, so 7.4.13 is the one to try now

gcoan, 2023-12-15:
Ok I'll upgrade to 7.4.13

springfall2008, 2023-12-16:
Some of these issue might relate to metric keep, which I think I might have fixed now on main.

closed 2023-12-21 by gcoan