springfall2008/batpred #476: Precharge Battery before saver session config

Bcrutt, 2023-12-15:
Hi

I was wondering why my current setup seems not to charge the battery before a saver session, I have added some screenshots below showing my current setup when using Octopus Flux, any comments would be appreciated.

[image]
d
 
[image]

[image]

[image]

springfall2008, 2023-12-15:
Your rate low threshold is hard wired so it can't use the more expensive rate - if you set it to 0 then it's automatic and it can do

Bcrutt, 2023-12-15:
Changed Low Threshold to 0

[image]

Plan still not showing any sign of charging before saver session

[image]

Also charging on saturday am now looks like this

[image]

I have attached the appdaemon log
Thanks
[appdaemon.log](https://github.com/springfall2008/batpred/files/13685704/appdaemon.log)

Bcrutt, 2023-12-15:
After awhile it did start charging and then held the charge at 45%, it then changed again to this
[image]

[appdaemon.log](https://github.com/springfall2008/batpred/files/13686149/appdaemon.log)

Bcrutt, 2023-12-15:
Then changed to this
[image]

Bcrutt, 2023-12-15:
V 7.14.12

JonathanLew1s, 2023-12-15:
The hold charge from 2pm to 4pm looked like a good plan. No point in incurring losses to charge the battery to discharge later, hold the battery at 45% and run the house load from the grid. Going into the saving session with 40% battery.
No sure why the following runs changed from that plan.

Personally, I would set the high threshold rate to 0 and enable charge freeze switch.

Bcrutt, 2023-12-15:
I thought the plan looked ok, then on the next pass it dropped the hold charge and would have gone into the saver session with just 11% soc. So I decided to set to read only and charged from 15:00 to 16:00 and then set to charge \ discharge and it exported from 17:00 to 18:00.
The plan then showed it was going to charge next from 02:00 saturday am all the way to 16:00 Saturday pm, which I don't understand why it would do that.

[image]
[image]

springfall2008, 2023-12-15:
Maybe combine charge needs to be turned off?

gcoan, 2023-12-15:
@Bcrutt as you are on a 3 time of tariff, once you've charged the battery overnight (starting in the cheap period and finishing in the day rate), there's no point in predbat doing anything different with the battery until the evening peak import rate when the battery starts discharging to support your load.

So I think the plan is correct.  If you look at the soc column, once the battery reaches 100% at 5am it then stays there until 4pm discharge.  Maybe from 5am to 4pm it could be labelled 'hold charge' or 'charge freeze', but its the same as charge and maintain 100%

You could look at the config options for your tariff, as Trefor says, turn combine charge off, but it is recommended to be on for Cozy/Flux type of tariffs, and I think the plan is correct anyway
https://springfall2008.github.io/batpred/configuration-guide/

Unless you have a different view what should be planned?

closed 2024-02-03 by springfall2008