springfall2008/batpred #480: Metric keep doesn't really do what I want

springfall2008, 2023-12-15:
**Describe the bug**

Metric keep is too harsh and forces charging at peak rate just to stay within keep

**Expected behavior**

Somehow try to stay above keep but without doing it just for the sake of it

gcoan, 2023-12-15:
Agree.  I'd ideally like to keep a kW or so spare in the battery but because of just this problem, min keep causing charging in high rate periods, I ended up setting it to 0.1

mpartington, 2023-12-16:
Agree, should aim to keep above it when setting the plan, but not at the expense of charging at peak rate if usage increases from initial prediction when letting it run flat would be more cost effective.

springfall2008, 2023-12-16:
I think I have fixed this on main, please give it a spin when you get time

springfall2008, 2023-12-16:
Well, I thought i had - but 3 versions later it seems better.... still on main

springfall2008, 2023-12-17:
https://github.com/springfall2008/batpred/releases/tag/v7.14.14

closed 2023-12-22 by springfall2008